zwave-js/zwave-js #2370: Missing device config: 0x0068:0x0003:0x0008

AlCalzone, 2021-04-15:
No discovered info

marcus-j-davies, 2021-04-16:
I think we have this (in some form)

https://devices.zwave-js.io/?jumpTo=0x0118:0x0003:0x0008:0.0
http://manuals-backend.z-wave.info/make.php?lang=en&sku=TZ67CH&cert=ZC10-16075148&type=mini

cdnninja, 2021-05-21:
Do you know of any details on this? I couldn't find a manual or anything.

AlCalzone, 2021-05-21:
No, that came from telemetry. That device didn't have any auto-discoverable info that helps here.

cdnninja, 2021-05-29:
I found most of the manuals for the good way devices here: https://usermanual.wiki/Good-Way-Technology 

However these manuals don't mention device ID.   I believe someone who owns one will needs to comment on which one it is.

AlCalzone, 2021-05-29:
Didn't find anything on the Z-Wave Alliance Website either

wodan66, 2022-01-05:
I own the product. 
This is a USB Zwave Dongle as shown here:
[https://products.z-wavealliance.org/products/2118](url)
0x0068:0x0003:0x0008:6.4

It does have the Sigma Designs, Inc. Chip in it
Vendor 0658 and 0200 device ID, although this references to the Aeotec Stick zw090. The Good Way Stick does not have a battery.

Hope that helps.

wodan66, 2022-01-06:
[USB_ZWAVE_Dongle.txt](https://github.com/zwave-js/node-zwave-js/files/7821679/USB_ZWAVE_Dongle.txt)

Could this work as config file?

AlCalzone, 2022-01-06:
Looks good, I'd remove the `"reset": "n/a"` though. Can you post this as a pullrequest?

PR #5400 (2023-02-01): feat(config): add Good Way WD6051 controller
fixes: #2370
closed 2023-02-01 by AlCalzone