zwave-js/zwave-js #2385: Leviton DZ15S gets many timeouts during interview

frogbite, 2021-04-16:
**Checklist:**

-   [x] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
-   [x] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
-   [x] I have checked the troubleshooting section and my problem is **not** described there.
-   [x] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

I migrated from OZW which had been working well for several years to Z Wave JS when I saw the deprecated next to the integration.

I first went to Z Wave JS using the official add-on. All my devices were found, which is a very small mesh of 4 devices, two of which are batter powered. After implementing this one of my Leviton switches worked but the second immediately showed as a dead node. After a while an automation with the working switch stopped working (about 2 weeks). The Leviton switch in that configuration now also showed as dead.

So to help with troubleshooting I removed the integration and the official add-on and went to Z-Wave JS to MQTT hoping this would give me better control and help me resolve the issues. Both my battery powered motion sensor and my Kwikset lock are seen and configured into the network without issues. I am not trying to add in a "Leviton - DZ15S - Scene Capable Push On/Off" back in but although it is seen I get many timeout messages during the interview stage.

2021-04-15T22:42:27.306Z CNTRLR [Node 004] Interviewing Association...
2021-04-15T22:42:27.308Z CNTRLR » [Node 004] querying number of association groups...
2021-04-15T22:42:37.437Z CNTRLR [Node 004] Timed out while waiting for a response from the node
2021-04-15T22:42:37.444Z CNTRLR [Node 004] Querying association groups timed out, skipping interview...
2021-04-15T22:42:37.466Z CNTRLR [Node 004] Interviewing Association Group Information...
2021-04-15 18:42:37.470 INFO ZWAVE: Node 4: interview stage COMMANDCLASSES completed
2021-04-15T22:42:37.473Z CNTRLR [Node 004] Interview stage completed: CommandClasses
2021-04-15 18:42:37.475 INFO ZWAVE: Node 4: interview stage OVERWRITECONFIG completed
2021-04-15T22:42:37.477Z CNTRLR [Node 004] Interview stage completed: OverwriteConfig
2021-04-15T22:42:37.479Z CNTRLR » [Node 004] requesting node neighbors...
2021-04-15T22:42:37.521Z CNTRLR « [Node 004] node neighbors received: 1
2021-04-15 18:42:37.522 INFO ZWAVE: Node 4: interview stage NEIGHBORS completed
2021-04-15T22:42:37.524Z CNTRLR [Node 004] Interview stage completed: Neighbors
2021-04-15 18:42:37.526 INFO ZWAVE: Node 4: interview stage COMPLETE completed
2021-04-15T22:42:37.528Z CNTRLR [Node 004] Interview completed
2021-04-15 18:42:37.541 INFO ZWAVE: Node 4: value added 4-37-0-currentValue => undefined
2021-04-15 18:42:37.543 INFO ZWAVE: Node 4: value added 4-37-0-targetValue => undefined
2021-04-15 18:42:37.544 INFO ZWAVE: Node 4: value added 4-134-0-libraryType => undefined
2021-04-15 18:42:37.546 INFO ZWAVE: Node 4: value added 4-134-0-protocolVersion => undefined
2021-04-15 18:42:37.547 INFO ZWAVE: Node 4: value added 4-134-0-firmwareVersions => undefined

Thinking that maybe it was the distance from my z-stick I moved the z-stick within 10 feet of the switch and went through a remove of the add-on and integration to find I have the same issue. Still getting timeouts. My motion sensor which is in the same room is able to be added and seen effortlessly and the switch is about 20 feet from the z-stick when it is in its native location.

From the add-on:

[image]**Device information**

From the integration:

[image]

I opened a git issue for the add-on and frenck thinks this is an upstream issue.

Wondering if I should go back to OZW or allow the community the opportunity to use my system to debug?

Which device(s) is/are affected (make/model)?  
What are the node IDs?

Leviton - DZ15S - Scene Capable Push On/Off
Leviton - DZ6H - Dimmer

Currently the node is 4 and it is the DZ15S.   

**Did you change anything?**

-   [X] **Yes**: _(please describe)_. 
Went from OZW to Z Wave JS with the official plugin.  Saw all my devices but the Leviton switch and dimmer both showed as dead shortly after adding.  With the official plugin, I did get heuristics back from the interview though.   Decided to install the ZWaveJS2MQTT as it offered better debugging via the console.  With the MQTT add-on the Leviton devices don't go dead after a day or two but I always see timeouts during the interview phase.  So it never completes.  The only other two devices this network are a Dome Motion sensor which I configured back in and a Kwikset lock that was configured in but I have not added it back as I have removed the integration and add-on trying to debug the issue with help from others on #zwave. 
-   [ ] **No**

**Did this use to work before?**

-   [ ] **Don't know**, this is a new device
-   [ ] **No**, it never worked anywhere
-   [x] **Yes**, in: _(specify application with versions)_ Worked with OZW for 2 years.  Was working before moving to Z Wave JS and has not worked since for these Leviton switches. 

**How are you using `node-zwave-js`**

-   [x] `zwavejs2mqtt` (latest) docker image Current version: 0.11.0
-   [ ] `zwavejs2mqtt` (dev) docker image
-   [ ] `zwavejs2mqtt` Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] ioBroker.zwave2 adapter
-   [x] `HomeAssistant` version 5.13
-   [ ] Pkg
-   [ ] Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] `node-red-contrib-zwave-js`
    -   Module version (double click node): 
-   [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github))
-   [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:

1. Install zwavejs2mqtt
2. Configure zwave in zwavejs2mqtt
3. Add Z Wave JS integration and unselect the box for using the official add-on
4. Exclude the. Leviton switch with AEON z-stick in manual mode
5. Include the Leviton switch with the AEON x-stick 
6. Reboot the rasperry PI
7. Log above from Interview after reboot done explicitly by me

**Additional context**

As stated started debugging this with the official Z Wave JS add-on and member of the Discord #zwave suggest using zwavejs2mqtt.

**Logfile:**

[zwave_js_dump_16_APR_2021.json.txt](https://github.com/zwave-js/node-zwave-js/files/6327955/zwave_js_dump_16_APR_2021.json.txt)
[zwavejs_358.log](https://github.com/zwave-js/node-zwave-js/files/6327977/zwavejs_358.log)

frogbite, 2021-04-16:
https://github.com/hassio-addons/addon-zwavejs2mqtt/issues/119#event-4601613411 for the add-on issue
https://github.com/home-assistant/core/issues/49279 for the core issue

blhoward2, 2021-04-16:
Try healing your nodes. You may have a bad route setup. Two repeaters makes for a very weak mesh.

AlCalzone, 2021-04-16:
It might also be that https://github.com/zwave-js/node-zwave-js/pull/2271 improves the situation, but HA hasn't caught up to that version yet.

frogbite, 2021-04-16:
Iv'e tried healing and re-interviews.   So, I don't think that will help in my case.  Reading #2271

AlCalzone, 2021-04-16:
It is just a hunch, but maybe the device doesn't get a valid route and can't answer because of that.
Not sure if reading the PR will help you, but once you're on a version that includes it, you can try again.

frogbite, 2021-04-16:
Well, the network is very simple.  It being the only node that is powered it is connected directly to z-stick. 

[image]

AlCalzone, 2021-04-17:
Is it possible that the device was originally included securely in OZW but now you're trying to use it without encryption? Some devices don't reset the network key even after exclusion, so you might have to reset it. Or configure the network key in zwave-js and try that first.

frogbite, 2021-04-17:
I did a factory reset of the z-stick at one point, so even if it was secure, it would not be after that.  I could do a factory reset of the switch though, in case it still thinks it is a secure device.

blhoward2, 2021-04-17:
Resetting the stick wouldn't address what @AlCalzone is describing.

AlCalzone, 2021-04-17:
The stick doesn't know if a device is secure. However, the device doesn't even report "Security CC" as supported, so that's probably not it. My guess is that you need #2271, so at least `node-zwave-js` version 7.2.2.

frogbite, 2021-04-17:
Well, what can I say.  @AlCalzone is genius!.  Factory reset the switch, did an include and it is working.  @blhoward2, I see, makes sense now.  Going to close this out.

frogbite, 2021-04-17:
Maybe it wasn't secure, but the reset did the trick.

closed 2021-04-17 by AlCalzone

frogbite, 2021-04-17:
[image]