zwave-js/zwave-js #2388: Sensative guard cannot inclusing

tomcsi, 2021-04-17:
Hello,

I try to inclusing the Sensative Guard Strip. but state of gui is ProtocolInfo, and everything else unknown.

### Version

**Checklist:**

- [x] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is **not** described there.
- [x] I have read the changelog problem was **not** mentioned there.

Build/Run method

- [x] Docker
- [ ] PKG
- [ ] Snap package
- [ ] Manually built (git clone - npm install - npm run build )

zwavejs2mqtt version: 3.3.0
zwave-js version: 7.2.2

### Describe the bug

Try to add Sensative Guard Stip after reset the device, but stuck on ProtocolInfo state after add device. The device is "dead".

### To Reproduce

Steps to reproduce the behavior:

1. Reset the device
2. Start inclusing
3. Add device with magnet

### Expected behavior

Add device correctly.

### Additional context

Debug log:
```
Apr 16 23:49:13 raspberrypi docker[2672]: 2021-04-16T22:49:13.287Z CNTRLR   starting inclusion process...
Apr 16 23:49:13 raspberrypi docker[2672]: 2021-04-16T22:49:13.323Z SERIAL » 0x0105004ac10372                                                     (7 bytes)
Apr 16 23:49:13 raspberrypi docker[2672]: 2021-04-16T22:49:13.325Z DRIVER » [REQ] [AddNodeToNetwork]
Apr 16 23:49:13 raspberrypi docker[2672]:                                     payload: 0xc103
Apr 16 23:49:13 raspberrypi docker[2672]: 2021-04-16T22:49:13.331Z SERIAL « [ACK]                                                                   (0x06)
Apr 16 23:49:13 raspberrypi docker[2672]: 2021-04-16T22:49:13.340Z SERIAL « 0x0107004a03010000b0                                                 (9 bytes)
Apr 16 23:49:13 raspberrypi docker[2672]: 2021-04-16T22:49:13.341Z SERIAL » [ACK]                                                                   (0x06)
Apr 16 23:49:13 raspberrypi docker[2672]: 2021-04-16T22:49:13.344Z DRIVER « [REQ] [AddNodeToNetwork]
Apr 16 23:49:13 raspberrypi docker[2672]:                                     payload: 0x03010000
Apr 16 23:49:13 raspberrypi docker[2672]: 2021-04-16T22:49:13.347Z DRIVER   handling request AddNodeToNetwork (74)
Apr 16 23:49:13 raspberrypi docker[2672]: 2021-04-16T22:49:13.348Z DRIVER     1 handler registered!
Apr 16 23:49:13 raspberrypi docker[2672]: 2021-04-16T22:49:13.348Z DRIVER     invoking handler #0
Apr 16 23:49:13 raspberrypi docker[2672]: 2021-04-16T22:49:13.350Z CNTRLR   handling add node request (status = Ready)
Apr 16 23:49:13 raspberrypi docker[2672]: 2021-04-16T22:49:13.351Z CNTRLR     the controller is now ready to add nodes
Apr 16 23:49:13 raspberrypi docker[2672]: 2021-04-16 22:49:13.352 INFO ZWAVE: Controller status: Non-secure inclusion started
Apr 16 23:49:13 raspberrypi docker[2672]: 2021-04-16T22:49:13.355Z DRIVER       the message was handled
Apr 16 23:49:13 raspberrypi docker[2672]: 2021-04-16 22:49:13.358 INFO ZWAVE: Success zwave api call startInclusion { success: true, message: 'Success zwave api call', result: true }
Apr 16 23:50:28 raspberrypi docker[2672]: 2021-04-16 22:50:28.532 INFO ZWAVE: Calling api stopInclusion with args: [ [length]: 0 ]
Apr 16 23:50:28 raspberrypi docker[2672]: 2021-04-16T22:50:28.536Z CNTRLR   stopping inclusion process...
Apr 16 23:50:28 raspberrypi docker[2672]: 2021-04-16T22:50:28.552Z SERIAL » 0x0105004ac50471                                                     (7 bytes)
Apr 16 23:50:28 raspberrypi docker[2672]: 2021-04-16T22:50:28.553Z DRIVER » [REQ] [AddNodeToNetwork]
Apr 16 23:50:28 raspberrypi docker[2672]:                                     payload: 0xc504
Apr 16 23:50:28 raspberrypi docker[2672]: 2021-04-16T22:50:28.560Z SERIAL « [ACK]                                                                   (0x06)
Apr 16 23:50:28 raspberrypi docker[2672]: 2021-04-16T22:50:28.612Z SERIAL « 0x0107004a04060600b6                                                 (9 bytes)
Apr 16 23:50:28 raspberrypi docker[2672]: 2021-04-16T22:50:28.614Z SERIAL » [ACK]                                                                   (0x06)
Apr 16 23:50:28 raspberrypi docker[2672]: 2021-04-16T22:50:28.618Z DRIVER « [REQ] [AddNodeToNetwork]
Apr 16 23:50:28 raspberrypi docker[2672]:                                     payload: 0x04060600
Apr 16 23:50:28 raspberrypi docker[2672]: 2021-04-16T22:50:28.623Z DRIVER   handling request AddNodeToNetwork (74)
[... 261 lines cut ...]
Apr 16 23:54:35 raspberrypi docker[2672]: 2021-04-16 22:54:35.528 INFO ZWAVE: Node 7: interview stage PROTOCOLINFO completed
Apr 16 23:54:35 raspberrypi docker[2672]: 2021-04-16T22:54:35.538Z CNTRLR   [Node 007] Interview stage completed: ProtocolInfo
Apr 16 23:54:35 raspberrypi docker[2672]: 2021-04-16T22:54:35.540Z CNTRLR » [Node 007] querying node info...
Apr 16 23:54:35 raspberrypi docker[2672]: 2021-04-16T22:54:35.541Z CNTRLR » [Node 007] pinging the node...
Apr 16 23:54:35 raspberrypi docker[2672]: 2021-04-16T22:54:35.549Z SERIAL » 0x01080013070100250ccb                                              (10 bytes)
Apr 16 23:54:35 raspberrypi docker[2672]: 2021-04-16T22:54:35.550Z DRIVER » [Node 007] [REQ] [SendData]
Apr 16 23:54:35 raspberrypi docker[2672]:                                   │ transmit options: 0x25
Apr 16 23:54:35 raspberrypi docker[2672]:                                   │ callback id:      12
Apr 16 23:54:35 raspberrypi docker[2672]:                                   └─[NoOperationCC]
Apr 16 23:54:35 raspberrypi docker[2672]: 2021-04-16T22:54:35.554Z SERIAL « [ACK]                                                                   (0x06)
Apr 16 23:54:35 raspberrypi docker[2672]: 2021-04-16T22:54:35.559Z SERIAL « 0x0104011301e8                                                       (6 bytes)
Apr 16 23:54:35 raspberrypi docker[2672]: 2021-04-16T22:54:35.561Z SERIAL » [ACK]                                                                   (0x06)
Apr 16 23:54:35 raspberrypi docker[2672]: 2021-04-16T22:54:35.562Z DRIVER « [RES] [SendData]
Apr 16 23:54:35 raspberrypi docker[2672]:                                     was sent: true
Apr 16 23:54:39 raspberrypi docker[2672]: 2021-04-16T22:54:39.675Z SERIAL « 0x011800130c01019b007f7f7f7f7f01010700000000020400001d              (26 bytes)
Apr 16 23:54:39 raspberrypi docker[2672]: 2021-04-16T22:54:39.677Z SERIAL » [ACK]                                                                   (0x06)
Apr 16 23:54:39 raspberrypi docker[2672]: 2021-04-16T22:54:39.685Z DRIVER « [REQ] [SendData]
Apr 16 23:54:39 raspberrypi docker[2672]:                                     callback id:     12
Apr 16 23:54:39 raspberrypi docker[2672]:                                     transmit status: NoAck
Apr 16 23:54:39 raspberrypi docker[2672]: 2021-04-16T22:54:39.695Z CNTRLR   [Node 007] The node did not respond after 1 attempts.
Apr 16 23:54:39 raspberrypi docker[2672]:                                   It is probably asleep, moving its messages to the wakeup queue.
Apr 16 23:54:39 raspberrypi docker[2672]: 2021-04-16T22:54:39.698Z CNTRLR   [Node 007] ping failed: The node is asleep
Apr 16 23:54:39 raspberrypi docker[2672]: 2021-04-16T22:54:39.699Z CNTRLR   [Node 007] Interview attempt (1/5) failed, node is dead.
Apr 16 23:54:39 raspberrypi docker[2672]: 2021-04-16 22:54:39.701 ERROR ZWAVE: Interview of node 7 has failed: The node is dead
```

AlCalzone, 2021-04-17:
> NoAck

The device does not acknowledge the commands it gets sent, so from the driver POV, it is in fact dead.

tomcsi, 2021-04-17:
Have any workaround or possible implement the driver an exception for this devices of manufaturer?

AlCalzone, 2021-04-18:
We have several users where this device is working. You could try bringing it closer to the controller to include and then later perform a heal on it once it is in its final location.

tomcsi, 2021-04-19:
I tried to including again to old zipato system, and it's success. It means that zipato including device with other method than this implementation. Is it possible that Zipato not interested ack from device?

AlCalzone, 2021-04-19:
> Is it possible that Zipato not interested ack from device?

That would be weird, since then it could never know if a device actually received a command. But you never know - unless you have a Zniffer to figure out what's actually happening on RF.

Have you tried manually waking up the device a couple of times after including?

tomcsi, 2021-04-19:
Yes, I tried to waking up the device. The device has reporting if I open or close the window, but the state of device never go the next step.

tomcsi, 2021-04-19:
Can I debug the RF from raspberry pi 4 ? Can you help me that I help to you solve the issue?

AlCalzone, 2021-04-19:
> Yes, I tried to waking up the device. The device has reporting if I open or close the window, but the state of device never go the next step.

That it is reporting does not mean that it is awake. Check the manual how to wake it up. Most devices require you to press the action button or something on the device.

> Can I debug the RF from raspberry pi 4 ? Can you help me that I help to you solve the issue?

No, you need a Zniffer (another Z-Wave stick with special firmware) and the Zniffer software from Silabs to capture the RF traffic. That could help narrow it down.

AlCalzone, 2021-04-20:
There is a bit of chaos going on, but it looks like the device is awake only for a **very short** time.
I suggest you let it all settle, then wake up the device (no re-inclusion or anything, just wake it) a couple more times.

tomcsi, 2021-04-20:
I order yesterday a Aeotec Z-Stick Gen5, because possible Razberry2 device is s*xx.

tomcsi, 2021-04-25:
I changed the Razberry2 device to Aeotec Z-Stick Gen5 and all are fine. Thanks for the many help.

closed 2021-04-25 by tomcsi