zwave-js/zwave-js #2393: Strange behaviour with configured minimum dimmer limit on Qubino dimmers

MarkGodwin, 2021-04-18:
**Checklist:**

-   [x] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
-   [x] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
-   [x] I have checked the troubleshooting section and my problem is **not** described there.
-   [x] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

With the Qubino DIN Dimmer - ZMNHSDx - you can set configuration values for maximum and minimum dimming. Some lights don't glow at all below around 20% dimming, so I want to use the minimum value to cut this dead-zone.

Setting a target value respects this minimum value, by re-mapping the target value to the appropriate value within the available range. However, reading back the current value reads a value adjusted by the range, and not the original value as set.

**Device information**

Which device(s) is/are affected (make/model)?  Qubino ZMNHSDx
What are the node IDs? 54

**Did you change anything?**

-   [ ] **Yes**: _(please describe)_
-   [x] **No** - only the minimum dimming value configuration.

**Did this use to work before?**

-   [ ] **Don't know**, this is a new device
-   [ ] **No**, it never worked anywhere
-   [x] **Yes**, in: It seems to work correctly in Hubitat C7's Hub, if that helps.

**How are you using `node-zwave-js`**

-   [ ] `zwavejs2mqtt` (latest) docker image
-   [ ] `zwavejs2mqtt` (dev) docker image
-   [ ] `zwavejs2mqtt` Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] ioBroker.zwave2 adapter
-   [x] `HomeAssistant` version core-2021.4.4 - with zwavejs2mqtt 0.12.0
-   [ ] Pkg
-   [ ] Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] `node-red-contrib-zwave-js`
    -   Module version (double click node): 
-   [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github))
-   [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:

1. Set config value "Minimum dimming value" (112-0-60) to 22.
2. Set the "Target value" (38-0-targetValue) to 30.
3. Read "Current value" (38-0-currentValue). This reports 45.

Note that the dimmer is probably doing something like (30 / 100) * 78 + 22 = 45.4

Zwave JS knows about the min and max values that have been configured, so could do the job of mapping back to the original target value instead. Does that seem like it would be possible?

**Additional context**

The refresh that happens a few seconds after a target value is set causes the reported dimmer value in HA to jump away from the chosen target value, making the feature unusable.

**Logfile:**

[zwavejs_359.log](https://github.com/zwave-js/node-zwave-js/files/6331739/zwavejs_359.log)

blhoward2, 2021-04-19:
The difficulty would be that this isn't universal and is a device bug. My aeotec nano dimmers allow the same setting but they report back the correct value. Generally we have to trust the level the device tells us its at as something could have happened locally or there could be something on the device we don't know about like auto-off, ramp rates, etc.

MarkGodwin, 2021-04-19:
Yes, it would definitely have to be a device or vendor specific setting, if that's possible.
I have not investigated the code yet, but would you be willing to accept a PR along these lines?

blhoward2, 2021-04-19:
That's up to @AlCalzone but I doubt it. We try to avoid device specific workarounds in the code and here you could just not set the range and not dim low. Even my hardware dimmers cut out at some point.

AlCalzone, 2021-04-19:
To be honest, I don't think it is worth the complexity to relate configuration params and dimmer levels. You'd have to think about remapping the value when the config param changes, etc...

https://github.com/zwave-js/node-zwave-js/issues/1532 would s

AlCalzone, 2021-04-19:
To be honest, I don't think it is worth the complexity to relate configuration params and dimmer levels. You'd have to think about remapping the value when the config param changes, etc...

The device does report (almost) the correct value on its own:
```
2021-04-18T16:35:54.266Z DRIVER » [Node 054] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      32
                                  └─[MultilevelSwitchCCSet]
                                      target value: 30
...
2021-04-18T16:35:54.377Z DRIVER « [Node 054] [REQ] [ApplicationCommand]
                                  └─[MultilevelSwitchCCReport]
                                      current value: 29
```
Therefore I agree with @blhoward2's assessment that this is a device bug. You need https://github.com/zwave-js/node-zwave-js/issues/1532, which would avoid the verfication completely.

AlCalzone, 2021-04-19:
Duplicate of #1532

MarkGodwin, 2021-04-20:
I agree that #1532 would resolve this adequately. That would also fix a second problem I have with the Qubino where the transition value is polled back during a transition, instead of the target value - e.g. if I set one target value, and then set a second target value a few seconds later, the scheduled poll kicked off by the first set reads back an intermediate value from the second transition.

Feel free to close this.

closed 2021-04-21 by AlCalzone