zwave-js/zwave-js #2398: Bali/Somfy Button

jesse-schein, 2021-04-18:
For some reason when I pair my bali blind simple up/down remote to ZwaveJS it can find the correct device information, but because the remote is battery it is only limited for how long it can pair, and the ZwaveJS interview kind of freaks out by this.

At one point no ACK happens which is because the remote has stopped being in pair mode.

17:42:30.859 DRIVER 1 handler registered!
17:42:30.860 DRIVER invoking handler #0
17:42:30.861 CNTRLR handling add node request (status = Done)
17:42:30.862 CNTRLR done called for 0
17:42:30.864 CNTRLR finished adding node 22:
basic device class: Controller
generic device class: Remote Controller
specific device class: Portable Remote Controller
supported CCs:
· Z-Wave Plus Info (0x5e)
· Version (0x86)
· Manufacturer Specific (0x72)
· Device Reset Locally (0x5a)
· Powerlevel (0x73)
· Association (0x85)
· Association Group Information (0x59)
· Battery (0x80)
· Central Scene (0x5b)
· Wake Up (0x84)
controlled CCs:
· Basic (0x20)
· Central Scene (0x5b)
· Wake Up (0x84)
· Multilevel Switch (0x26)
17:42:30.868 CNTRLR » [Node 022] Assigning SUC return route...
17:42:30.877 DRIVER » [Node 022] [REQ] [AssignSUCReturnRoute]
payload: 0x1606
17:42:30.878 CNTRLR the inclusion process was stopped
2021-04-18 17:42:30.879 INFO ZWAVE: Controller status: Inclusion stopped
17:42:30.894 DRIVER « [RES] [AssignSUCReturnRoute]
was executed: true
17:42:31.630 DRIVER « [REQ] [AssignSUCReturnRoute]
callback id: 6
transmit status: NoAck
17:42:31.648 CNTRLR [Node 022] Configuring Z-Wave+ Lifeline association...
17:42:31.745 DRIVER » [Node 022] [REQ] [SendData]
│ transmit options: 0x25
│ callback id: 7
└─[AssociationCCSet]
group id: 1
node ids: 1
17:42:31.758 DRIVER « [RES] [SendData]
was sent: true
17:42:38.652 DRIVER « [REQ] [SendData]
callback id: 7
transmit status: NoAck
17:42:38.663 CNTRLR [Node 022] did not respond after 1/3 attempts. Scheduling next try in 500 ms.
17:42:39.176 DRIVER » [Node 022] [REQ] [SendData]
│ transmit options: 0x25
│ callback id: 7
└─[AssociationCCSet]
group id: 1
node ids: 1
17:42:39.189 DRIVER « [RES] [SendData]
was sent: true
17:42:44.206 DRIVER « [REQ] [SendData]
callback id: 7
transmit status: NoAck
17:42:44.211 CNTRLR [Node 022] did not respond after 2/3 attempts. Scheduling next try in 500 ms.
17:42:44.721 DRIVER » [Node 022] [REQ] [SendData]
│ transmit options: 0x25
│ callback id: 7
└─[AssociationCCSet]
group id: 1
node ids: 1
17:42:44.737 DRIVER « [RES] [SendData]
was sent: true
17:42:50.228 DRIVER « [REQ] [SendData]
callback id: 7
transmit status: NoAck
17:42:50.239 CNTRLR [Node 022] Node 22 did not respond after 3 attempts, it is presumed dead
17:42:50.249 CNTRLR [Node 022] Failed to configure Z-Wave+ Lifeline association: Failed to send th
e command after 3 attempts (Status NoAck)
17:42:50.254 CNTRLR » [Node 022] pinging the node...
17:42:50.297 DRIVER » [Node 022] [REQ] [SendData]
│ transmit options: 0x25
│ callback id: 8
└─[NoOperationCC]
17:42:50.310 DRIVER « [RES] [SendData]
was sent: true
17:42:57.050 DRIVER « [REQ] [SendData]
callback id: 8
transmit status: NoAck
17:42:57.054 CNTRLR [Node 022] Node 22 did not respond after 1 attempts, it is presumed dead
17:42:57.057 CNTRLR [Node 022] ping failed: Failed to send the command after 1 attempts (Status No
Ack)
17:42:57.058 CNTRLR [Node 022] Cannot configure wakeup destination: The message cannot be sent bec
ause node 22 is dead
17:42:57.061 CNTRLR [Node 022] Beginning interview - last completed stage: None
17:42:57.061 CNTRLR [Node 022] new node, doing a full interview...
17:42:57.063 CNTRLR » [Node 022] querying protocol info...
17:42:57.075 DRIVER » [REQ] [GetNodeProtocolInfo]
payload: 0x16
2021-04-18 17:42:57.076 INFO ZWAVE: Node 22: added
17:42:57.084 DRIVER the message was handled
17:42:57.094 DRIVER « [RES] [GetNodeProtocolInfo]
payload: 0x000000030000
17:42:57.100 CNTRLR « [Node 022] received response for protocol info:
basic device class: Slave
generic device class: UNKNOWN (0x00)
specific device class: Unused
node type: Controller
is always listening: false
is frequent listening: false
can route messages: false
supports security: false
supports beaming: false
maximum data rate: 9600 kbps
protocol version: 0
2021-04-18 17:42:57.102 INFO ZWAVE: Node 22: interview stage PROTOCOLINFO completed
17:42:57.111 CNTRLR [Node 022] Interview stage completed: ProtocolInfo
17:42:57.113 CNTRLR » [Node 022] querying node info...
17:42:57.114 CNTRLR » [Node 022] pinging the node...
17:42:57.127 DRIVER » [Node 022] [REQ] [SendData]
│ transmit options: 0x25
│ callback id: 9
└─[NoOperationCC]
17:42:57.138 DRIVER « [RES] [SendData]
was sent: true
17:43:02.955 DRIVER « [REQ] [SendData]
callback id: 9
transmit status: NoAck
17:43:02.962 CNTRLR [Node 022] The node did not respond after 1 attempts.
It is probably asleep, moving its messages to the wakeup queue.
17:43:02.964 CNTRLR [Node 022] ping failed: The node is asleep
17:43:02.965 CNTRLR [Node 022] Interview attempt (1/5) failed, node is dead.
2021-04-18 17:43:02.966 ERROR ZWAVE: Interview of node 22 has failed: The node is dead
2021-04-18 17:43:24.801 INFO ZWAVE: Calling api refreshInfo with args: [ 22, [length]: 1 ]
17:43:24.814 CNTRLR [Node 022] Beginning interview - last completed stage: None
17:43:24.817 CNTRLR [Node 022] new node, doing a full interview...
2021-04-18 17:43:24.818 INFO ZWAVE: Node 22: interview started
17:43:24.820 CNTRLR » [Node 022] querying protocol info...
17:43:24.840 DRIVER » [REQ] [GetNodeProtocolInfo]
payload: 0x16
2021-04-18 17:43:24.842 INFO ZWAVE: Success zwave api call refreshInfo { success: true, message: 'Success zwave api call', result: undefined }
17:43:24.853 DRIVER « [RES] [GetNodeProtocolInfo]
payload: 0x000000030000
17:43:24.861 CNTRLR « [Node 022] received response for protocol info:
basic device class: Slave
generic device class: UNKNOWN (0x00)
specific device class: Unused
node type: Controller
is always listening: false
is frequent listening: false
can route messages: false
supports security: false
supports beaming: false
maximum data rate: 9600 kbps
protocol version: 0
17:43:24.863 CNTRLR [Node 022] The node is now asleep.
2021-04-18 17:43:24.866 INFO ZWAVE: Node 22 is now asleep
2021-04-18 17:43:24.869 INFO ZWAVE: Node 22: interview stage PROTOCOLINFO completed
17:43:24.871 CNTRLR [Node 022] Interview stage completed: ProtocolInfo
17:43:24.872 CNTRLR » [Node 022] querying node info...
2021-04-18 17:43:35.942 INFO ZWAVE: Calling api refreshInfo with args: [ 22, [length]: 1 ]
17:43:35.949 CNTRLR [Node 022] Beginning interview - last completed stage: None
17:43:35.951 CNTRLR [Node 022] new node, doing a full interview...
2021-04-18 17:43:35.952 INFO ZWAVE: Node 22: interview started
17:43:35.953 CNTRLR » [Node 022] querying protocol info...
17:43:35.967 DRIVER » [REQ] [GetNodeProtocolInfo]
payload: 0x16
2021-04-18 17:43:35.970 INFO ZWAVE: Success zwave api call refreshInfo { success: true, message: 'Success zwave api call', result: undefined }
17:43:35.979 DRIVER « [RES] [GetNodeProtocolInfo]
payload: 0x000000030000
17:43:35.984 CNTRLR « [Node 022] received response for protocol info:
basic device class: Slave
generic device class: UNKNOWN (0x00)
specific device class: Unused
node type: Controller
is always listening: false
is frequent listening: false
can route messages: false
supports security: false
supports beaming: false
maximum data rate: 9600 kbps
protocol version: 0
2021-04-18 17:43:35.986 INFO ZWAVE: Node 22: interview stage PROTOCOLINFO completed

blhoward2, 2021-04-19:
This frequently happens with batter devices. Just wake up the device a few times until it is complete.

jesse-schein, 2021-04-19:
I tried but unfortunately it's a simple up down remote so waking it up is only for a few seconds. I ended up having to pair it manually with my aeotec stick in order to manually associate it to my shade. Now at least I got the shade in zwave and the remote associated to the shade for manual control. 

Does this sound right?

blhoward2, 2021-04-19:
That sounds extreme. Though an interview usually doesn't take that long even at a few seconds a piece. I have an old water sensor that takes 7-8 wake ups.

I'm less clear why we're repeating the protocol info stage. Did you hit reinterview device in-between those?

jesse-schein, 2021-04-19:
No, I did not reinterview between. Which is sorta why I posted the issue because it looks like it grabbed the info correctly and then it looks like it grabbed protocolinfo again but because its a battery device it was a sleep so couldnt be contacted, so it sorta reset the info to nothing. 

I couldnt do anything timing wise to get the device to stay on long enough for it to figure things out.

AlCalzone, 2021-04-19:
> No, I did not reinterview between

Looks like you did, twice:

> 2021-04-18 17:43:24.801 INFO ZWAVE: Calling api refreshInfo with args: [ 22, [length]: 1 ]
> 2021-04-18 17:43:35.942 INFO ZWAVE: Calling api refreshInfo with args: [ 22, [length]: 1 ]

Nonetheless, the device didn't respond at all after it was included. I'm not sure why that is and why it works with the stick. 
That causes a bunch of critical interview queries to fail, so the interview can't proceed until the device wakes up. Unfortunately, even the first couple of commands that set the wakeup destination to the controller fail, so probably the device doesn't even know who to tell that it woke up.

AlCalzone, 2021-04-19:
Just to be sure - (how) does the device work with a different driver? Is it expected that there are some values? This looks like it is a device that doesn't want to talk to central controller at all.

jesse-schein, 2021-04-19:
> Looks like you did, twice:
>
>
> Nonetheless, the device didn't respond at all after it was included. I'm not sure why that is and why it works with the stick.
> That causes a bunch of critical interview [...]

I dont know why those show, I only hit the inclusion button once I specifically cleared the logs before running inclusion, for some reason it did things again.

> Just to be sure - (how) does the device work with a different driver? Is it expected that there are some values? This looks like it is a device that doesn't want to talk to central controller at all.

That may be it but it idenfitied the device. Basically what I did now is its paired the Aeotec stick manually and then I was able to associate the button to the shade directly using the bali/somfy instructions which is to press the button on the back of the remote once, then click and hold the shade for 3 seconds. Then it gets associated.

AlCalzone, 2021-04-19:
Just to be clear: `zwave-js` also paired the device. It just couldn't interview it afterwards because there was no response.

> I was able to associate the button to the shade directly using the bali/somfy instructions which is to press the button on the back of the remote once, then click and hold the shade for 3 seconds. Then it gets associated.

Sounds to me that doesn't have to do with the central controller though.

Do you have a manual PDF I could take a look at?

jesse-schein, 2021-04-19:
> Just to be clear: `zwave-js` also paired the device. It just couldn't interview it afterwards because there was no response.
>
>
> Sounds to me that doesn't have to do with the central controller [...]

Right the manual operations I did with the pair button and procedures didnt have anything to do with the central controller. Although I couldn't get it to work until it was sucessfully paired to the same network as the shade which to me indicates zwave-js wasnt adding it to the network properly and something wrong was happening.

I mean to be honest, I dont really need the controller to be in zwave-js or homeassistant since all the button should ever do is raise or lower the shade.

I'd be happy to debug and run the project locally if that could help, I am a software developer myself and I am not unfamiliar with node or typescript.

Here is the guide that I used that came printed with my shade.

https://member.baliblinds.com/globalassets/3.-documents/installation-instructions/bali-motorization-owners-manual-autoview.pdf

blhoward2, 2021-04-19:
^ Although I couldn't get it to work until it was sucessfully paired to the same network as the shade which to me indicates zwave-js wasnt adding it to the network properly and something wrong was happening.

I read that manual as the remote acts as it's own controller. Does it show up as a node now?

jesse-schein, 2021-04-19:
> > Although I couldn't get it to work until it was sucessfully paired to the same network as the shade which to me indicates zwave-js wasnt adding it to the network properly and something wrong was happening.
> 
> I read that manual as the remote acts as it's own controller. Does it show up as a node now?

Everytime I added it, it to would add it to zwave-js as a node but all the information on it is defaulted. Despite it showing the proper capabilities initally during inclusion.

Currently, it shows as a node in zwave-js although I included it with my Aeotec stick manually by unplugging the USB and adding it manually since the stick has a battery and pairing capabilities while not connected.

[image]

blhoward2, 2021-04-19:
It's still not interviewed though. If it doesn't respond to pings it'll get marked as dead eventually.

jesse-schein, 2021-04-19:
> It's still not interviewed though. If it doesn't respond to pings it'll get marked as dead eventually.

How long is that supposed to take? Its been sitting like this since yesterday I believe.

blhoward2, 2021-04-19:
It won’t progress unless and until the device wakes up and starts sending
info.

AlCalzone, 2021-04-19:
> Although I couldn't get it to work until it was sucessfully paired to the same network as the shade

This is expected. Z-Wave uses a so-called home ID to identify which devices can communicate with each other. While the shades are in one network and the remote is not, it is expected that the remote cannot control the shade.

> which to me indicates zwave-js wasnt adding it to the network properly and something wrong was happening.

Your log above shows that the remote was **successfully paired** using zwave-js. This means it is now known to the Z-Wave stick and part of the network:
```
17:42:30.864 CNTRLR finished adding node 22:
basic device class: Controller
generic device class: Remote Controller
specific device class: Portable Remote Controller
supported CCs:
· Z-Wave Plus Info (0x5e)
· Version (0x86)
· Manufacturer Specific (0x72)
· Device Reset Locally (0x5a)
· Powerlevel (0x73)
· Association (0x85)
· Association Group Information (0x59)
· Battery (0x80)
· Central Scene (0x5b)
· Wake Up (0x84)
controlled CCs:
· Basic (0x20)
· Central Scene (0x5b)
· Wake Up (0x84)
· Multilevel Switch (0x26)
```
You end up in the same situation when you pair it with the button on the stick directly. What zwave-js now does is an interview where it queries the supported CCs for more details. The stick doesn't do that.

> Everytime I added it, it to would add it to zwave-js as a node but all the information on it is defaulted. Despite it showing the proper capabilities initally during inclusion.

Yes, we know which CCs are supported. But we don't know which versions and which capabilities of these CCs the device supports and so on. Discovering that requires the device to actually respond to commands. As long as that doesn't happen, the interview cannot progress.
The manual isn't terribly helpful in this regard, but it also doesn't mention anything regarding wakeup etc. The remote probably isn't meant to be a "real" member of a Z-Wave network and that the inclusion is just a crutch to make it possible to control the shades.

> How long is that supposed to take? Its been sitting like this since yesterday I believe.

Until the device tells the central controller that it is now awake. My guess is on never.

jesse-schein, 2021-04-19:
> > Although I couldn't get it to work until it was sucessfully paired to the same network as the shade
> 
> This is expected. Z-Wave uses a so-called home ID to identify which devices can communicate with each other. While the shades are in one network and the remote is not, it is expected that the remote cannot control the shade.
> 
> > which to me indicates zwave-js wasnt adding it to the network properly and something wrong was happening.
> 
> Your log above shows that the remote was **successfully paired** using zwave-js. This means it is now known to the Z-Wave stick and part of the network:
> 
> ```
> 17:42:30.864 CNTRLR finished adding node 22:
> basic device class: Controller
> generic device class: Remote Controller
> specific device class: Portable Remote Controller
> supported CCs:
> · Z-Wave Plus Info (0x5e)
> · Version (0x86)
> · Manufacturer Specific (0x72)
> · Device Reset Locally (0x5a)
> · Powerlevel (0x73)
> · Association (0x85)
> · Association Group Information (0x59)
> · Battery (0x80)
> · Central Scene (0x5b)
> · Wake Up (0x84)
> controlled CCs:
> · Basic (0x20)
> · Central Scene (0x5b)
> · Wake Up (0x84)
> · Multilevel Switch (0x26)
> ```
> 
> You end up in the same situation when you pair it with the button on the stick directly. The info you're seeing is stored on the stick for each included node. What zwave-js now does is an interview where it queries the supported CCs for more details.
> 
> > Everytime I added it, it to would add it to zwave-js as a node but all the information on it is defaulted. Despite it showing the proper capabilities initally during inclusion.
> 
> Yes, we know which CCs are supported. But we don't know which versions and which capabilities of these CCs the device supports and so on. Discovering that requires the device to actually respond to commands. As long as that doesn't happen, the interview cannot progress.
> The manual isn't terribly helpful in this regard, but it also doesn't mention anything regarding wakeup etc. The remote probably isn't meant to be a "real" member of a Z-Wave network and that the inclusion is just a crutch to make it possible to control the shades.
> 
> > How long is that supposed to take? Its been sitting like this since yesterday I believe.
> 
> Until the device tells the central controller that it is now awake. My guess is on never.

Ok, this was my theory more or less as well. So good on zwave-js for discovering the capabilities but the device I dont think actually responds to commands so thats probably why the interview failed.

Unfortunately, because the interview failed, it did NOT add it properly to the network. I was not able to continue on with the manual in order to associate the remote to the shade.

The only way I got things to work were as follows, and I list these steps in order for anyone in the future to find this issue and learn what to do:

1. Follow bali manual and remove remote from shade by holding the shade pair button for 7 seconds
2. follow bali manual and clear remote associations by holding remote pair button for 15 seconds and red flashing stops
3. Start inclusion in zwave-js
4. Hold pair button on the shade for 3 seconds
5. Shade will jog when the inclusion detected the shade
6. let zwave-js do its thing
7. stop zwave-js container/server
8. in my specific case, remove the Aeotec Stick from the pi/computer
9. start inclusion on the stick
10. hold the pair button on remote for 3 seconds
11. should be paired very quickly
12. reinsert usb & start zwave-js
13. press and release button on remote (as indicated in manual)
14. press and hold shade pair button for 3 seconds until the shade flashes
15. shade should jog and remote should be associated to it
16. shade will still function in zwave-js

AlCalzone, 2021-04-19:
> Unfortunately, because the interview failed, it did NOT add it properly to the network. I was not able to continue on with the manual in order to associate the remote to the shade.

Then something else went wrong. Interviewing does not have anything to do with pairing/inclusion into a network. It just allows zwave-js to know what it can do with the device.
The inclusion part is handled entirely by the Z-Wave protocol which is baked into the stick's firmware.

You don't have a Zniffer by any chance so you can capture RF logs of both the zwave-js inclusion and the stick inclusion method?

closed 2021-05-03 by zwave-js-assistant[bot]