zwave-js/zwave-js #2400: Fibaro Outlet FGWPG-121

siwilson2000, 2021-04-19:
Hi,

Just added a Fibaro outlet FWPG-121 (UK variant) device to my Z-wave mesh in Home Assistant and it comes up as unknown manufacturer and model. Looking at the database it seems only the FWPB-121 US version is there today. I have a read through the manual, and created a FWPG-121.json file using the FWPB-121.json file as a template. However I am unsure how to get this into the database Can someone help here?

Thanks.

[fgwpg-121.json.zip](https://github.com/zwave-js/node-zwave-js/files/6333799/fgwpg-121.json.zip)

AlCalzone, 2021-04-19:
There are two options:
1. Fork the repo, clone your fork, make a branch, add the file there, push it back to your fork, make a PR. This allows you to make multi-file changes at once
2. Navigate to the folder https://github.com/zwave-js/node-zwave-js/tree/master/packages/config/config/devices/0x010f and press "Upload file" there. This should create a PR with the file for you.

In any case it makes sense if you add the changes. This will allow you to make edits more easily when the change is reviewed.

siwilson2000, 2021-04-20:
I have never forked a repo so I went with option 2. It would not allow me to upload a file but would allow me to create one so I did that and made the PR. It looks to have passed all the checks  but is now blocked as a review is required. Hopefully someone can do that for me.

Thanks.

AlCalzone, 2021-04-20:
I've asked @blhoward2 to do it (or I'll do it later or tomorrow). Thanks so far!

siwilson2000, 2021-04-21:
Any update? I'd like to give the code a test.

Thanks

PR #2404 (2021-04-24): feat(config): add Fibaro outlet FWPG-121 (UK version)
fixes: #2400
closed 2021-04-24 by AlCalzone