zwave-js/zwave-js #2401: [question] Re-Interview controller node results in node to become Unknown

bushvin, 2021-04-19:
I'm setting up a new z-wave network, and I'm trying to do it right (or as right as possible) this time.
I bought a Silicon Labs SLUSB7000A UZB-7, and after the initial Firmware issues, I got the controller working correctly.

When I re-interview the controller node, It becomes "unknown".
Controller node debug info correct:
```json
{
  "id": 1,
  "deviceId": "0-4-4",
  "manufacturer": "Silicon Labs",
  "manufacturerId": 0,
  "productType": 4,
  "productId": 4,
  "name": "silabs_zst10_700_00",
  "loc": "",
  "values": [],
  "groups": [],
  "neighbors": [
    2,
    3
  ],
  "ready": true,
  "available": true,
  "hassDevices": {},
  "failed": false,
  "lastActive": 1618829957449,
  "supportsBeaming": true,
  "supportsSecurity": false,
  "isSecure": false,
  "keepAwake": false,
  "maxBaudRate": null,
  "isRouting": true,
  "isFrequentListening": false,
  "isListening": true,
  "inited": true,
  "hexId": "0x0000-0x0004-0x0004",
  "dbLink": "https://devices.zwave-js.io/?jumpTo=0x0000:0x0004:0x0004:0.0",
  "productLabel": "ZST10-700",
  "productDescription": "700 Series-based Controller",
  "protocolVersion": 3,
  "nodeType": 0,
  "endpointsCount": 0,
  "isControllerNode": true,
  "dataRate": 100000,
  "deviceClass": {
    "basic": 2,
    "generic": 2,
    "specific": 1
  },
  "status": "Alive",
  "interviewStage": "Complete",
  "_name": "silabs_zst10_700_00"
}
```

Controller node debug info after re-interview:
```json
{
  "id": 1,
  "deviceId": "NaN-NaN-NaN",
  "manufacturer": "Unknown manufacturer 0xXXXX",
  "name": "silabs_zst10_700_00",
  "loc": "",
  "values": [],
  "groups": [],
  "neighbors": [
    2,
    3
  ],
  "ready": true,
  "available": true,
  "hassDevices": {},
  "failed": false,
  "lastActive": 1618829587402,
  "supportsBeaming": true,
  "supportsSecurity": false,
  "isSecure": false,
  "keepAwake": false,
  "maxBaudRate": null,
  "isRouting": true,
  "isFrequentListening": false,
  "isListening": true,
  "inited": true,
  "hexId": "0xXXXX-0xXXXX-0xXXXX",
  "dbLink": "https://devices.zwave-js.io/?jumpTo=0xXXXX:0xXXXX:0xXXXX:0.0",
  "productLabel": "Unknown product 0xXXXX",
  "productDescription": "0xXXXX",
  "protocolVersion": 3,
  "nodeType": 0,
  "endpointsCount": 0,
  "isControllerNode": true,
  "dataRate": 100000,
  "deviceClass": {
    "basic": 2,
    "generic": 2,
    "specific": 1
  },
  "status": "Alive",
  "interviewStage": "Complete",
  "_name": "silabs_zst10_700_00"
}
```

Here's the logs I captured:
[zwavejs_1.log.gz](https://github.com/zwave-js/zwavejs2mqtt/files/6335538/zwavejs_1.log.gz)

The only way to go back to a 'sane' situation is to stop the container, plug out the USB stick, wait for a moment, plug it back in and start the container.

blhoward2, 2021-04-19:
What version of zwave-js are you running?

AlCalzone, 2021-04-19:
The manufacturer info for the stick is queried differently, which is not repeated with a normal "interview".
Basically, re-interviewing a controller should neither be necessary nor possible.

bushvin, 2021-04-19:
@blhoward2 :
zwavejs2mqtt: 3.3.0
zwave-js: 7.2.2

@AlCalzone do you want me to open a new issue on zwave-js regarding this?

AlCalzone, 2021-04-20:
No, it's okay - we've got this one.

PR #2422 (2021-04-24): fix(node): do not re-interview the controller
Re-interviewing the controller essentially only deletes its values from the value DB. Therefore we no longer allow doing that.

fixes: #2401
closed 2021-04-24 by AlCalzone