zwave-js/zwave-js #2405: Popp Solar Siren 2 v2.6

dijbi, 2021-04-20:
**Which device is missing?**
Popp Solar Siren 2 **v2.6**

**What are the IDs?**
Manufacturer ID: 0x0154 
Product type: 0x0004  
Product ID: 0x0002
Firmware : **2.6**  

**Did you check if we really don't have it?**

- [x] It is not in the [configuration DB](https://devices.zwave-js.io/) because max version is set to **2.5**
- [ ] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)
- [ ] It is not in the [TODO list](https://github.com/zwave-js/node-zwave-js/issues/1600)

PDF manual
[Same PDF](https://aeotec.freshdesk.com/support/solutions/articles/6000219167-solar-outdoor-siren-2-user-guide-) file (no firmware version is specified so it should work for any version ??)
If so, it helps in creating the config file.

Config parameters are therefore not accessible inside my HA setup with - Zwave integration & Zwave2Mqtt
How can i help ?

Thx

AlCalzone, 2021-04-20:
The only similar device with a different firmware version seems to be the [first revision](https://devices.zwave-js.io/?jumpTo=0x0154:0x0004:0x0002:1.1) of this device. I think it makes sense to be a bit more optimistic regarding the versions and change
https://github.com/zwave-js/node-zwave-js/blob/41f095b0ee304cf09b05161c0d8f1965428a1493/packages/config/config/devices/0x0154/700854_2.4-2.5.json#L15-L16
to be 2.0 - 255.255 until there's a new revision.

Do you want to provide a PR?

dijbi, 2021-04-20:
Made the PR  (First PR ever ! - hope i didn't mess up)
#2408

AlCalzone, 2021-04-20:
Perfect! I've edited the commit message but the rest was good

closed 2021-04-20 by AlCalzone