zwave-js/zwave-js #2409: [bug] Notifications no longer updating for Yale Conexis L1 Door Lock

lolouk44, 2021-04-20:
### Version
zwavejs2mqtt: 3.2.0
zwave-js: 7.1.1
home id: 3731321360
 home hex:0xde677210

**Checklist:**

- [ ] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is **not** described there.
- [x] I have read the changelog problem was **not** mentioned there.

Build/Run method

- [x] Docker
- [ ] PKG
- [ ] Snap package
- [ ] Manually built (git clone - npm install - npm run build )

zwavejs2mqtt version: 3.2.0
zwave-js version: 7.1.1

### Describe the bug

Notifications no longer updating for Yale Conexis L1 Door Lock:
- [5-113-0-alarmType] Alarm Type
- [5-113-0-alarmLevel] Alarm Level
This used to work and started to fail to refesh from late March, around 22nd

### To Reproduce

Steps to reproduce the behavior:

1. Open Zwave JS Server UI URL
2. Expand Yale node
3. Expand the Notification Tab
4. Nothing changes

### Expected behavior

Notifications values should change

### Additional context

I do use Home Assistant, and eventually hope/expect that resolution on the JS Server will trickle down to HA, but since the issue is directly seen on the JS Server I'm creating an issue here. I hope this is ok.

lolouk44, 2021-04-20:
part of the issue indeed is the lack of standardization, hence the other issue that we/you all worked on.
Here is anyway a list of states that work for a Conexis L1. They probably won't apply to other locks.

alarmtype:   9 => Jammed
alarmtype: 16 =>   Unlocked by Yale App
alarmtype: 21 =>   Locked manually
alarmtype: 22 =>   Unlocked manually
alarmtype: 23 =>   Jammed by RF
alarmtype: 24 =>   Locked by RF
alarmtype: 25 =>   Unlocked by RF
alarmtype: 26 =>   Jammed by Auto relock
alarmtype: 27 =>   Auto relock
alarmtype: 144 =>   Unlocked by RFID tag (alarmlevel gives the tag ID)
alarmtype: 167 =>   Battery Low
alarmtype: 168 =>   Battery Critical
alarmtype: 169 =>   Battery Too Low To Operate Lock

current issues (once I've created the template sensor that catches zwave events):
- unlock by RF doesn't seem to trigger an event
- jammed lock doesn't seem to trigger an event
What would be good to have is the notification exposed as attributes in one of the HA sensors, rather than the _dirty_ workaround I've had to create
One thing that would make things easier is to update the alarmtype and alarmlevel regardless of whether a "standard" notification has been caught / guessed...

AlCalzone, 2021-04-20:
> One thing that would make things easier is to update the alarmtype and alarmlevel regardless of whether a "standard" notification has been caught / guessed...

Funny, I just had the same discussion with the HA devs. I had a couple of concerns but ultimately this seems to make sense because we won't be breaking automations every time we add a mapping.

captaindaveman, 2021-05-18:
Is there any solution for this challenge? It's a real shame that one of the best features of this lock and it's HA integration was being able to tell who unlocked a door and how it was unlocked , key fob, through HA or manually, it only seems to work when a key fob is used now

AlCalzone, 2021-05-18:
It's done in 7.5.0 which I released yesterday. We overlooked this issue and forgot to close it.