zwave-js/zwave-js #2415: ZEN34 Device config doesn't match

hoopsta1423, 2021-04-21:
**Which device is missing?**
Zooz ZEN34

**What are the IDs?**

Manufacturer ID: 0x027a
Product type: 0x0004  
Product ID: 0xf001  

**Did you check if we really don't have it?**
There is already another ZEN34 config file where product type: 0x7000 but I have 3 switches and they all show up with product type: 0x0004. 

I created another .json config file placed in /usr/src/app/node_modules/@zwave-js/config/config/devices/0x027a with my reported ID/type and also edited the index.json file to include this information. However, after excluding/re-including, the remote switches still show up as unknown manufacturer 0x027a, product 0x0004, unknown product 0xf001

Any help would be excellent!

- [ ] It is not in the [configuration DB](https://devices.zwave-js.io/)
- [ ] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)
- [ ] It is not in the [TODO list](https://github.com/zwave-js/node-zwave-js/issues/1600)

**Do you have a link to the PDF manual?**

If so, it helps in creating the config file.

AlCalzone, 2021-04-21:
> also edited the index.json file to include this information

There's a reason the index file starts with this comment:
```
// This file is auto-generated. DO NOT edit it by hand if you don't know what you're doing!"
```

Changes to config files get detected automatically **if** you don't edit the index file by hand. Your best bet is to delete the index file altogether and restart, so it gets regenerated.

hoopsta1423, 2021-04-21:
> > also edited the index.json file to include this information
> 
> There's a reason the index file starts with this comment:
> 
> ```
> // This file is auto-generated. DO NOT edit it by hand if you don't know what you're doing!"
> ```
> 
> Changes to config files get detected automatically **if** you don't edit the index file by hand. Your best bet is to delete the index file altogether and restart, so it gets regenerated.
> 
> Also if it is the same device, you should just add the additional device ID to the existing config file.

Thanks, the reason I thought index.json needed to be edited was a similar post here about modifying device config files https://github.com/zwave-js/zwavejs2mqtt/discussions/582

What is the proper way to add additional device id to the existing config? Is it as simple as adding another "productType": "0x0004" line within "devices"?

hoopsta1423, 2021-04-21:
> As @AlCalzone Mentioned, `index.json` is an internal index, and really shouldn't be edited by hand.
> 
> > What is the proper way to add additional device id to the existing config?
> 
> Add the additional entry in the config (and delete `index.json` for good measure - since you have made changes, that may have broken it)
> 
> ```
> "devices": [
>   {
>     "productType": "0x7000",
>     "productId": "0xf001"
>   },
>   {
>     "productType": "0x0000", <----- HERE
>     "productId": "0x0000" <------ and HERE
>   }
> ]   
> ```

Worked great, thanks!

closed 2021-04-21 by hoopsta1423

AlCalzone, 2021-04-21:
@hoopsta1423 is your device detected correctly with the additional ID?

hoopsta1423, 2021-04-21:
> @hoopsta1423 is your device detected correctly with the additional ID?

Yep, it is!

marcus-j-davies, 2021-04-21:
@AlCalzone,

Whilst I am here - you want me to PR?

AlCalzone, 2021-04-21:
Sure