zwave-js/zwave-js #2417: Fibaro FGK101 not reporting sensor IN1 status correctly [bug]

gbillyb, 2021-04-21:
### Version
zwavejs2mqtt: 3.2.1.c4123a4
zwave-js: 7.2.2
home id: 23188717
 home hex:0x161d4ed

**Checklist:**

- [x] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is **not** described there.
- [x] I have read the changelog problem was **not** mentioned there.

Build/Run method

- [ ] Docker
- [ ] PKG
- [ ] Snap package
- [x] Manually built (git clone - npm install - npm run build )

zwavejs2mqtt: 3.2.1.c4123a4
zwave-js: 7.2.2

### Describe the bug
After discovery and interview of a v2.1 Fibaro FGK101 Door/Window Sensor, the opening/closing of the door and status of the sensor (IN1) is not reported to zwavejs2mqtt correctly.

### To Reproduce

1. Include FGK101 v2.1 and wait for interview to complete
2. Close/Open door
3. In zwavejs2mqtt.log (with logging set to silly) the following is received when the door is opened/closed:
2021-04-17 13:41:43.623 INFO ZWAVE: Node 45: value updated: 32-0-currentValue 255 => 0
2021-04-17 13:41:46.263 INFO ZWAVE: Node 45: value updated: 32-0-currentValue 0 => 255

### Expected behavior
the following is received on a different node (44) when the door is opened/closed
2021-04-17 13:32:19.088 INFO ZWAVE: Node 44: metadata updated: 48-0-Any
2021-04-17 13:32:19.089 INFO ZWAVE: Node 44: value updated: 48-0-Any false => true
2021-04-17 13:32:21.456 INFO ZWAVE: Node 44: metadata updated: 48-0-Any
2021-04-17 13:32:21.457 INFO ZWAVE: Node 44: value updated: 48-0-Any true => false

### Additional context
Node 44 is the same device but at v2.5.

Also, it appears that each time a Refresh is actioned, the `groups` attribute is appended and contains duplicates.

[Node_045_Interview.log](https://github.com/zwave-js/zwavejs2mqtt/files/6328596/Node_045_Interview.log)
[Node_045_Debug_after_Refresh.log](https://github.com/zwave-js/zwavejs2mqtt/files/6328597/Node_045_Debug_after_Refresh.log)
[Node_044_Interview.log](https://github.com/zwave-js/zwavejs2mqtt/files/6328598/Node_044_Interview.log)
[Node_044_Debug_after_Refresh.log](https://github.com/zwave-js/zwavejs2mqtt/files/6328599/Node_044_Debug_after_Refresh.log)

AlCalzone, 2021-04-21:
> Is there any device db mapping we can update to fix this ?

Yep, we basically need exactly https://github.com/zwave-js/node-zwave-js/pull/2386 for this device too. It looks like it should be enough for the 2.1-2.3 firmware range. Accepting PRs.

closed 2021-04-22 by AlCalzone

gbillyb, 2021-04-22:
Confirming this is now fixed.
Many thanks.

PR #2421 (2021-04-22): fix(config): map Basic Set to Binary Sensor Reports for Fibaro FGK101
Fibaro FGK101 with firmware version lower than 2.3 does not report properly state change and uses `BasicCC:Set` to do so.

As per discussed in #2417, enabling compat flag to treat the basic set report as a status change (similar to what has been done on #2386).

After applying the patch, I have got the following in logs, confirmed working on two FGK101-2.1 devices
```
2021-04-22T06:31:25.722Z CNTRLR [Node 024] treating BasicCC::Set as a report
2021-04-22 08:31:25.736 INFO ZWAVE: Node 24: value updated: 48-0-Any false => true [... cut]