zwave-js/zwave-js #2454: MCO home MH-C421 multilevel switch is not good

S1M8N, 2021-04-24:
**Checklist:**

-   [ ] I am using HomeAssistant.
-   [ ] I am using ZWaveJS2MQTT.
-   [ ] I have checked the troubleshooting section and my problem is **not** described there. Yes and No
-   [ ] I have read the changelog and my problem was **not** mentioned there. Yes and No...

**Describe the bug**

Hello,

I use home assistant OS without any add-on except zwavejs2mqtt which is in the latest version and the z-wave js integration that goes with it. However my C421 home MCO switches are not considered a multilevel switch (I think). I can still move them up and down I don't really know how but it works. On the other hand, it is impossible for me to do the STOP / PAUSE action ...

If you need a log please let me know where they are so I can send them to you. If I need an additional add on to retrieve these logs let me know

This is an issue that had already been addressed and corrected by adding a force multilevel switch for my device. When I did not yet have the add-on version it was good the force multievel switch worked but there it does not seem to work or badly ...
I noticed that if the module was in an unknown state everything was working in the management of the multilevel switch however when it is added correctly this is no longer the case and the stop button no longer works ...

I won't go into details because I don't really know how it works.

Hope we can solve this problem.

Thank you in advance

**Device information**

MCO home C421
What are the node IDs?
6 and 8

**Did you change anything?**

-   [ ] **No**

**Did this use to work before?**

-   [ ] **Don't know**

**How are you using `node-zwave-js`**
-   [ ] `HomeAssistant` version 0.13.0

**To Reproduce**

Steps to reproduce the behavior:

1. Go to 'Zwave js Integration'
2. Click on 'Stop button'
4. See error (doesn't work)
[zwave_js_dump.zip](https://github.com/zwave-js/node-zwave-js/files/6370474/zwave_js_dump.zip)

blhoward2, 2021-04-26:
What do you mean the stop/pause action? Can you go into the zwavejs2mqtt UI, go to that node, and see if everything works when controlling it under the Multilevel sub-tab?

AlCalzone, 2021-04-26:
Also, I'm probably going to need a zwave-js logfile where you control the device.

S1M8N, 2021-04-26:
> Also, I'm probably going to need a zwave-js logfile where you control the device.

[faf58650.zip](https://github.com/zwave-js/node-zwave-js/files/6378748/faf58650.zip)

blhoward2, 2021-04-27:
The log is helpful but can you also answer my questions? We need to understand the issue.

AlCalzone, 2021-04-27:
@S1M8N that's no logfile :(

S1M8N, 2021-04-27:
Yes sorry :

AlCalzone, 2021-04-27:
Thanks, but we still need a bit of explanation what the problem is. In your log I see that you control the device using `Multilevel Switch CC` and it changes the value.

S1M8N, 2021-04-27:
Yes, however the pause button does not work : 
[image]

AlCalzone, 2021-04-27:
That's in Home Assistant though. I'd need to see a log where you push that, but I don't know how to get a log out of HA.

S1M8N, 2021-04-27:
I understand but if I don't go through HA I don't have the pose button in the add on zwavejs2mqtt

AlCalzone, 2021-04-27:
@S1M8N your cache file above contains `version: 0` for `Multilevel Switch` on both nodes. Either the node didn't respond during the interview, or your cache file got corrupted.
Please re-interview both nodes in `zwavejs2mqtt` (under the "advanced" node options) and post a log of that. Maybe that fixes the problem - if not, I'll at least know why.

S1M8N, 2021-04-27:
it doesn't seem to have helped. Even worse ...
However once I added the module and I had an unknown state and everything worked ... But when I have everything correct at Device ID level the problem is the one mentioned above .... 
[Re-interview_log.zip](https://github.com/zwave-js/node-zwave-js/files/6387493/Re-interview_log.zip)

AlCalzone, 2021-04-27:
Okay, the node is contradicting itself. First it lists Multilevel Switch as supported, then it claims it does not support Multilevel Switch:
```
2021-04-27T20:24:05.603Z CNTRLR « [Node 006] node info received
                                  supported CCs:
                                  · [...]
                                  · Multilevel Switch
...
2021-04-27T20:24:05.967Z DRIVER « [Node 006] [REQ] [ApplicationCommand]
                                  └─[VersionCCCommandClassReport]
                                      CC:      Multilevel Switch
                                      version: 0
2021-04-27T20:24:05.974Z CNTRLR   [Node 006]   does NOT support CC Multilevel Switch (0x26)
```

We can force-override this information via the config files, but I'm not sure which version it actually supports. Going with 1 probably doesn't hurt.

AlCalzone, 2021-04-27:
Ok, it turns out we already forced it to support `Multilevel Switch`, but that was not enough.

S1M8N, 2021-05-01:
Your changes had no effect

I have included the module again and check that the config file has been changed...

AlCalzone, 2021-05-05:
@S1M8N you have to re-interview it for the changes to be applied.

S1M8N, 2021-05-05:
> @S1M8N you have to re-interview it for the changes to be applied.

Yes

S1M8N, 2021-05-12:
Hello @AlCalzone, 

I'm sorry but I don't have an answer, were you able to find something or do you need something more ??

thank you for your work

AlCalzone, 2021-05-12:
Please re-interview the device and share the zwave-js log and the cache files `<homeid>.json` and `<homeid>.values.json` afterwards.

S1M8N, 2021-05-12:
> Please re-interview the device and share the zwave-js log and the cache files `<homeid>.json` and `<homeid>.values.json` afterwards.
Ok, I have re-interview the devices. It's the node 10
[Logs.zip](https://github.com/zwave-js/node-zwave-js/files/6467802/Logs.zip)

S1M8N, 2021-05-15:
@AlCalzone,

The logs is ok for you ?
[Logs (1).zip](https://github.com/zwave-js/node-zwave-js/files/6487550/Logs.1.zip)

AlCalzone, 2021-05-16:
Ok I think I found it. The change does get applied and you can even see in the cache now that the version is forced to be `1`:
```json
                "0x26": {
                    "name": "Multilevel Switch",
                    "endpoints": {
                        "0": {
                            "isSupported": true,
                            "isControlled": false,
                            "secure": false,
                            "version": 1
                        }
                    }
                },
```

However, the interview for that CC is still not done, which is why you're not seeing those values.

S1M8N, 2021-05-16:
ok can you do something about it?

AlCalzone, 2021-05-16:
Yeah

AlCalzone, 2021-05-16:
Duplicate of #2273

Please follow that issue for an update

closed 2021-05-16 by AlCalzone

S1M8N, 2021-05-28:
@AlCalzone
Hello, I still cannot use the stop button on my shutter. I have version 7.7.0 of z-wavejs and the latest version of the plugin. I re included the device. Its node is now 12.
[Log 28.02.2021.zip](https://github.com/zwave-js/node-zwave-js/files/6562912/Log.28.02.2021.zip)

AlCalzone, 2021-05-28:
The issue I linked to isn't done yet.

S1M8N, 2021-05-29:
> The issue I linked to isn't done yet.

Ok when are you planning to do this?

S1M8N, 2021-06-01:
> The issue I linked to isn't done yet.

@AlCalzone
Ok when are you planning to do this?

AlCalzone, 2021-06-01:
I can't make any promises right now. Will do it soon, but that's about as concrete as I can be at the moment.

S1M8N, 2021-06-01:
> I can't make any promises right now. Will do it soon, but that's about as concrete as I can be at the moment.

I hope you will have time before the end of June it would be super cool

PR #2471 (2021-04-27): fix(config): override reported Multilevel Switch version for MH-C421
fixes: #2454