zwave-js/zwave-js #2462: [bug] Siren not working and no config menu for Popp Solar Outdoor Siren (005107), interview OK and completed

davidwaze, 2021-04-26:
Build/Run method

- [ X ] Manually built (git clone - npm install - npm run build )

zwavejs2mqtt version: 3.5
zwave-js version: 7.2.4

### Describe the bug

Interview seems to be OK, but no config menu and impossible to set on/off the horn/light

Device : (0x0154-0x0004-0x0002)

Tried : Update topics / Refresh Values / Heal Node / Re-interview Node / Remove store - rediscover store

2021-04-25 11:09:57.922 INFO ZWAVE: Node 118: interview stage NODEINFO completed
2021-04-25 10:55:52.511 INFO ZWAVE: Node 118: interview stage COMMANDCLASSES completed
2021-04-25 10:55:52.512 INFO ZWAVE: Node 118: interview stage OVERWRITECONFIG completed
2021-04-25 10:55:52.523 INFO ZWAVE: Node 118: interview stage NEIGHBORS completed
2021-04-25 10:55:52.524 INFO ZWAVE: Node 118: interview stage COMPLETE completed
2021-04-25 10:55:52.527 INFO ZWAVE: Node 118: value added 118-37-0-currentValue => undefined
2021-04-25 10:55:52.527 INFO ZWAVE: Node 118: value added 118-37-0-targetValue => undefined
2021-04-25 10:55:52.528 INFO ZWAVE: Node 118: value added 118-128-0-level => 99
2021-04-25 10:55:52.528 INFO ZWAVE: Node 118: value added 118-128-0-isLow => false
2021-04-25 10:55:52.528 INFO ZWAVE: Node 118: value added 118-114-0-manufacturerId => 340
2021-04-25 10:55:52.528 INFO ZWAVE: Node 118: value added 118-114-0-productType => 4
2021-04-25 10:55:52.528 INFO ZWAVE: Node 118: value added 118-114-0-productId => 2
2021-04-25 10:55:52.528 INFO ZWAVE: Node 118: value added 118-134-0-libraryType => 3
2021-04-25 10:55:52.528 INFO ZWAVE: Node 118: value added 118-134-0-protocolVersion => 4.5
2021-04-25 10:55:52.528 INFO ZWAVE: Node 118: value added 118-134-0-firmwareVersions => 1.4
2021-04-25 10:55:52.529 INFO ZWAVE: Node 118: value added 118-134-0-hardwareVersion => undefined
2021-04-25 10:55:52.529 INFO ZWAVE: Node 118 ready: Popp & Co - 005107 (Solar Powered Outdoor Siren)
2021-04-25 10:55:52.529 INFO ZWAVE: Node 118: interview COMPLETED, all values are updated

Any idea ?

Another information, my device is working correctly as i'm currently runing with Jeedom/openzwave and siren is working fine, that's my last device i'm trying to migrate.

Thanks

AlCalzone, 2021-04-27:
@davidwaze The siren is included securely, but you haven't configured the Z-Wave network key in zwavejs2mqtt.

davidwaze, 2021-04-28:
Hello,

You were right sorry for the disturbance (my siren was very high on my outdorr wall to get it that's why i wanted to troubleshoot first).

With my previous zwave solution (jeedom/openzwave), secure connection was not really functionnal, so it was not really setted into secure mode.

I had to climb to my outdoor wall :-), removed/re-added to the zwave network (via zwavejs which is simply amazing) and all is fine.

Thank you again for your help and congrats for this project !

Bye bye.

David

closed 2021-04-28 by davidwaze