zwave-js/zwave-js #2464: Homeseer WD-100 device control failing on "NoSupport" response.

millercentral, 2021-04-27:
**Checklist:**

-   [ x ] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
-   [ x ] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
-   [ x ] I have checked the troubleshooting section and my problem is **not** described there.
-   [ x ] I have read the changelog and my problem was **not** mentioned there.

[a home assistant developer sent me here]

**Describe the bug**

Issue controlling a Homeseer WD-100 dimmer switch from Zwave JS. Issue appears with both Home Assistant Zwave JS Supervisor and ZwaveJS2MQTT server. Originally referenced in [this Home Assistant bug](https://github.com/home-assistant/core/issues/49397), but triage suggests its an upstream Node Zwave JS issue.

Updates for the device are failing when response status of a SupervisionCCReport is "NoSupport", causing device remote control to be non functional.

**Device information**

Which device(s) is/are affected (make/model)?  Homeseer WD-100
What are the node IDs? 35, 10, 11, 29, 33, 38, 39, 40

**Did you change anything?**

-   [ ] **Yes**: _(please describe)_
-   [ x ] **No**

**Did this use to work before?**

-   [ ] **Don't know**, this is a new device
-   [ ] **No**, it never worked anywhere
-   [ x ] **Yes**, in: other controllers: SmartThings, Hubitat. Never in Zwave JS

**How are you using `node-zwave-js`**

-   [ ] `zwavejs2mqtt` (latest) docker image
-   [ ] `zwavejs2mqtt` (dev) docker image
-   [ ] `zwavejs2mqtt` Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] ioBroker.zwave2 adapter
-   [ x ] `HomeAssistant` version 2021.4.5 w/ ZwaveJS2MQTT addin version 0.13.0
-   [ ] Pkg
-   [ ] Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] `node-red-contrib-zwave-js`
    -   Module version (double click node): 
-   [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github))
-   [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:

1. Add a Homeseer WD-100 switch to Zwave JS network
2. Attempt to turn switch on and off via Zwave JS command
3. See error

**Additional context**

Another homeseer dimmer switch, the HS-WD-200+ has SupervisionCC disabled in the device config and works correctly with Zwave JS.  This iss was originally raised as a [Home Assistant bug here](https://github.com/home-assistant/core/issues/49397), but the developers triaged in as an upstream Zwave JS issue. 

**Logfile:**

Node Dump JSON:
[nodes_dump.json.txt](https://github.com/zwave-js/node-zwave-js/files/6381054/nodes_dump.json.txt)

Debug log of ZwaveJS2MQTT session attempting to control device (captured from debug window in addin control panel):
[zwavejs2mqtt debug log.txt](https://github.com/zwave-js/node-zwave-js/files/6381059/zwavejs2mqtt.debug.log.txt)

Let me know what else I can do to help. Thanks!

blhoward2, 2021-04-27:
I can add the flag to remove supervision but you're not going to be able to test it with the addon at the moment. @AlCalzone do you want me to add this blindly?

millercentral, 2021-04-27:
So I was able to test this by exec-ing a bash shell into to the zwavejs2mqtt addin docker container and modifying the file `/opt/node_modules/@zwave-js/config/config/devices/0x000c/hs-wd100.json` with the following addition just after the `paramInfo` section:
```
        "compat": {
                "commandClasses": {
                        "remove": {
                                "0x6c": {
                                        // The device reports that it supports Supervision, but it does not support all
                                        // commands supervision-encapsulated
                                        "endpoints": "*"
                                }
                        }
                }
        },
```
... and then re-interviewing one of the WD-100 switches.  

Once I did this, the dimmer control worked normally in Home Assistant. Here is the updated debug log file from the addon UI (the node in question is id: 35):
[debug_log_compat_enabled.txt](https://github.com/zwave-js/node-zwave-js/files/6381609/debug_log_compat_enabled.txt)

Of course this is a very temporary scenario for me, for if I restart the addon it will rebuild the container, but it does appear that compat section will solve the issue.

One other extremely minor note, the Zwave JS Config DB calls these dimmers "WD-100", but I believe the official model number is actually "HS-WD100+". Every other homeseer switch model number start with "HS-", and [this is the manual](https://homeseer.com/wp-content/uploads/2020/09/HS-WD100-Manual-7.pdf) for the device.

blhoward2, 2021-04-27:
Ok. I'm sold.

PR #2465 (2021-04-27): fix(config): add compat flag to remove supervision from homeseer hs-wd100+
Closes #2464
closed 2021-04-27 by AlCalzone