zwave-js/zwave-js #2467: Missing Parameter 5 "RGB" for "AEON Labs - ZW130 - WallMote Quad"

motom001, 2021-04-27:
**Which device is missing?**

AEON Labs - ZW130 - WallMote Quad

**What are the IDs?**
manufacturer: AEON Labs
manufacturerId: 0x0086
label: ZW130
productType: 0x0002, 0x0102, 0x0202
productId: 0x0082

Link to existing json-file:
https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x0086/zw130.json

- [X ] It is not in the [configuration DB](https://devices.zwave-js.io/)
- [X ] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)
- [ X] It is not in the [TODO list](https://github.com/zwave-js/node-zwave-js/issues/1600)

You can see an additional parameter 5 for this devices to change the color and brightness in another project:
https://github.com/kentaclub/homey.aeotec/blob/master/drivers/ZW130/device.js#L86

I tested this parameter with my homey-setup and this devices and could change the color.

Parameter 5 has to be an "buffer" of 4 float-numbers - examples are here:
https://github.com/kentaclub/homey.aeotec/blob/master/drivers/ZW130/driver.settings.compose.json#L92

Maybe @kentaclub can help to update the missing parameter?

AlCalzone, 2021-04-27:
That one was removed a while back because I couldn't find any firmware version where this was actually supported:
https://github.com/zwave-js/node-zwave-js/commit/e64b7bb808d49c0ad9d61520edd5d1a65f1c8dbb#diff-848f557e15bdea16fe7d21a3294df5b292172f5d22fd3eb3695c5df6b7fb7262

Quote:
> 		// My device with did not have it in v1.5 and v1.8, which is currently the latest version
>		// It is also not present in v2.3 (new hardware revision)

It is also not included in @blhoward2's PR that revamps all Aeotec config files based on information he got from Aeotec directly, so I'm guessing it wasn't documented in any of the manuals he received:
https://github.com/zwave-js/node-zwave-js/pull/2459/files#diff-848f557e15bdea16fe7d21a3294df5b292172f5d22fd3eb3695c5df6b7fb7262

So at the very least we need to know which firmware version(s) this parameter exists for, because having unsupported config parameters in the device files causes ugly delays during the initial interview.

motom001, 2021-04-27:
I bought five devices in 2019 (in germany) with firmware 2.3

Informations from the device backside:

`Aeotec Ltd.
WallMote Quad
Model: ZW130-C (868.42MHz)
`

AlCalzone, 2021-04-27:
Can you share your cache files (`<homeid>.json` and `<homeid>.values.jsonl`) or a zwavejs2mqtt node dump so I can verify the device IDs listed above?

motom001, 2021-04-27:
I use home assistent with add-on "Z-Wave JS" (https://www.home-assistant.io/integrations/zwave_js/) and so i have no plan where I can found your requested files :(
I found also no way to attach to the docker and access the files directly

There is an option to dump all z-wave config, maybe you search that (changed homeId to 0000000000)?

`
[
  {
    "type": "version",
    "driverVersion": "7.2.4",
    "serverVersion": "1.5.0",
    "homeId": 0000000000,
    "minSchemaVersion": 0,
    "maxSchemaVersion": 4
  },
  {
    "type": "result",
    "success": true,
    "messageId": "api-schema-id",
    "result": {}
  },
  {
    "type": "result",
    "success": true,
    "messageId": "listen-id",
    "result": {
      "state": {
        "driver": {
          "logConfig": {
            "enabled": true,
            "level": "info",
            "logToFile": false,
            "filename": "/usr/src/node_modules/@zwave-js/server/dist/bin/zwave-240.log",
            "forceConsole": true
          },
          "statisticsEnabled": false
        },
        "controller": {
          "libraryVersion": "Z-Wave 4.54",
          "type": 1,
          "homeId": 0000000000,
          "ownNodeId": 1,
          "isSecondary": false,
          "isUsingHomeIdFromOtherNetwork": false,
          "isSISPresent": true,
          "wasRealPrimary": true,
          "isStaticUpdateController": true,
          "isSlave": false,
          "serialApiVersion": "1.1",
          "manufacturerId": 134,
          "productType": 1,
          "productId": 90,
          "supportedFunctionTypes": [
            2,
            3,
            4,
            5,
            6,
            7,
            8,
            9,
            16,
            17,
            18,
            19,
            20,
            21,
            22,
            23,
            24,
[... 843 lines cut ...]
                "id": 134,
                "name": "Version",
                "version": 2,
                "isSecure": false
              },
              {
                "id": 142,
                "name": "Multi Channel Association",
                "version": 3,
                "isSecure": false
              },
              {
                "id": 152,
                "name": "Security",
                "version": 1,
                "isSecure": true
              }
            ]
          }
        ]
      }
    }
  }
]
`

AlCalzone, 2021-04-27:
Could you use the zwavejs2mqtt addon for a test? It has the ability to read and write custom config parameters. I'd be interested in seeing a log of where you do that.

closed 2021-05-08 by zwave-js-assistant[bot]