zwave-js/zwave-js #2469: [bug]Qubino ZMNHCD Flush Shutter Problems

febalci, 2021-04-27:
### Version

Build/Run method

- [ ] Docker

zwavejs2mqtt version: 3.5.0
zwave-js version: 7.2.4

Using with HA WS Server, MQTT Gateway disabled.
### Describe the bug

Using 12 Qubino ZMNCHD Flush Shutters at home, 3 of them has Temperature Sensor ZMNHEA1 connected to the flush shutters. Have 2 problems:
1. The temperature sensors shows 49 degrees C, whenever i manually refresh them they show the correct value for a while but when it refreshes automatically it goes back to 49 degrees C, 2 seconds after the correct value.
```
2021-04-25T20:33:56.927Z CNTRLR   [Node 004] [Multilevel Sensor] Air temperature: metadata updated  [Endpoint 0]
2021-04-25T20:33:56.927Z CNTRLR   [Node 004] [~] [Multilevel Sensor] Air temperature: 22.1 => 22    [Endpoint 0]
2021-04-25T20:33:56.927Z SERIAL » [ACK]                                                                   (0x06)
2021-04-25T20:33:56.928Z DRIVER « [Node 004] [REQ] [ApplicationCommand]
                                  └─[MultilevelSensorCCReport]
                                      type:  Air temperature
                                      scale: Celsius
                                      value: 22
2021-04-25T20:33:56.929Z CNTRLR   [Node 004] Mapping unsolicited report from root device to first supporting end
                                  point zwave-js/zwavejs2mqtt#2
2021-04-25T20:33:56.929Z CNTRLR   [Node 004] [Multilevel Sensor] Air temperature: metadata updated  [Endpoint 2]
2021-04-25T20:33:56.930Z CNTRLR   [Node 004] [~] [Multilevel Sensor] Air temperature: 22.1 => 22    [Endpoint 2]
2021-04-25T20:33:58.052Z SERIAL « 0x01140004001d0e3202a16400035f4000050003605320                      (22 bytes)
2021-04-25T20:33:58.053Z SERIAL » [ACK]                                                                   (0x06)
2021-04-25T20:33:58.243Z SERIAL « 0x01140004001d0e3202a16c0000330200050000330402                      (22 bytes)
2021-04-25T20:33:58.244Z SERIAL » [ACK]                                                                   (0x06)
2021-04-25T20:33:58.396Z SERIAL « 0x0110000400040a600d02003105010131058a                              (18 bytes)
2021-04-25T20:33:58.397Z CNTRLR   [Node 004] [Multilevel Sensor] Air temperature: metadata updated  [Endpoint 2]
2021-04-25T20:33:58.398Z CNTRLR   [Node 004] [~] [Multilevel Sensor] Air temperature: 22 => 49      [Endpoint 2]
2021-04-25T20:33:58.399Z SERIAL » [ACK]                                                                   (0x06)
2021-04-25T20:33:58.401Z DRIVER « [Node 004] [REQ] [ApplicationCommand]
                                  └─[MultiChannelCCCommandEncapsulation]
                                    │ source:      2
                                    │ destination: 0
                                    └─[MultilevelSensorCCReport]
                                        type:  Air temperature
                                        scale: Celsius
                                        value: 49
```
2. When i press close shutter from HA, the shutter closes but level is stuck at 23%. Also when opening, it opens but the level shown  it is stuck at 76%. This happens on the temperature sensor connected ones all the time.

```
2021-04-25T13:32:51.760Z CNTRLR   [Node 004] [~] [Multilevel Switch] currentValue: 99 => 23         [Endpoint 1]
2021-04-25T13:32:51.761Z SERIAL » [ACK]                                                                   (0x06)
2021-04-25T13:32:51.763Z DRIVER « [Node 004] [REQ] [ApplicationCommand]
                                  └─[MultiChannelCCCommandEncapsulation]
                                    │ source:      1
                                    │ destination: 0
                                    └─[MultilevelSwitchCCReport]
                                        current value: 23
```

### To Reproduce

When i press open or close button in HA, the temperature sensor connected ones always have this 49 degree and half closed/opened problem. The others lose level only seldomly.

### Expected behavior

A clear and concise description of what you expected to happen.

AlCalzone, 2021-04-27:
This **might** be a duplicate of #2286, at least partially. 

But honestly, I think this is a case for the Qubino support. `zwave-js` works with what the device tells it, and the 49 °C is decoded correctly. I'm not sure why it sends that, but it does.

closed 2021-05-24 by AlCalzone

febalci, 2021-05-30:
I think i solved the problem at last:

The second problem (wrong level for temperature sensor connected ones) is solved like this:
As default, zwavejs2mqtt makes a group assignment of Endpoint 0 lifeline to Endpoint 0 of NodeID_1. I deleted this and recreated a new group assignment as Endpoint 0 lifeline connection to Endpoint None of Node 1. It solved the problem.

For the temperature problem, I created a new group assignment Endpoint 2 TemperatureReport to Endpoint 0 of Node 1.  This solved the temperature problem too...

Thanks for the support.

AlCalzone, 2021-05-30:
We should be able to configure that in the config files. Just to be sure, can you tell me which group numbers that are?

febalci, 2021-05-31:
1. When i am using Flush Shutter ZMNHCD without any temperature sensor, the groups came out as follows; no problem here, the default in the config file should remain the same.
[image]

2. When i use temperature sensor connected; then after re-interview, it creates Multi Channel and Multi Channel Sensor additionally and creates 3 endpoints, namely Endpoint 0,1 and 2. But the default comes up wrong. The following is the correct groups setting that works. 

[image]

AlCalzone, 2021-05-31:
We should be able to add the default assignment to endpoint 2. If that endpoint doesn't exist, no assignment will be done.

febalci, 2021-05-31:
Additionally if Endpoint 2 exists, the default comes up as 0,lifeline,NodeId_1,target endpoint 0 whereas it should also be 0,lifeline,NodeId_1,target endpoint None.

AlCalzone, 2021-05-31:
So, the lifeline should just be addressed to **Node 1, no endpoint** always. Correct?

febalci, 2021-05-31:
Yes exactly. Why did i try to make it so complicated , lol :) ...

febalci, 2021-07-23:
The problem is back again in a different form in 8.5.0  zwsavejs2mqtt 5.4.2. Whether the module has temperature sensor or not, it only has Endpoint 0. Back in zwavejs2mqtt version 4.x i used to have Endpoint 0,1 and 2 in the modules that has temp sensor attached and only Endpoint 0 in the modules that do not have temp sensor. 

So, the ones that has the temp sensors attached has both temperature update and wrong switch level problems...

AlCalzone, 2021-07-25:
Z-Wave JS 8.x includes some changes to endpoints/associations which might require compat flags for some devices. Please share a driver log of a re-interview, so I can use that to figure out which changes are necessary here.