zwave-js/zwave-js #2474: Add Zooz ZSE11

blhoward2, 2021-04-28:
Requested by Zooz: https://products.z-wavealliance.org/products/4152?selectedFrequencyId=-1

The sensor reports binary sensor for motion by default instead of notification because we discovered some issues on other platforms so Parameter 16 is set to value 1 by default which means send binary sensor report (don't send notification). If parameter 16 is set to value 0, the sensor won't send binary sensor reports and will send notification reports instead. I mention that in case you'd like to force reconfiguration for the device upon inclusion if you think notification will work better for your platform.

AlCalzone, 2021-04-28:
@zwave-js-bot import config 4152 from zwa

closed 2021-05-03 by AlCalzone

paulcusick3, 2021-05-05:
Seeing that motion is being received as the tamper sensor, not the motion sensor.  Am I alone?

blhoward2, 2021-05-05:
No, they sent me one of these to test. I suspect we need to undo the mapping. Can you get a silly log of the motion tripped and post it?

paulcusick3, 2021-05-05:
> No, they sent me one of these to test and this was just step one. I

Here is the log set to debug - let me know if you need verbose.   Device is
Node 046

paulcusick3, 2021-05-05:
[zwavejs_log.txt](https://github.com/zwave-js/node-zwave-js/files/6430629/zwavejs_log.txt)

AlCalzone, 2021-05-06:
```
2021-05-05T22:28:02.307Z CNTRLR   [Node 046] [~] [Binary Sensor] Motion: true => false              [Endpoint 0]
...
2021-05-05T22:28:07.623Z CNTRLR   [Node 046] [~] [Notification] Home Security[Cover status]: 0 => 3 [Endpoint 0]
```
I see both motion and cover status being reported by the device.

blhoward2, 2021-05-06:
Yes, if you go to the other issue I confirmed it. I've let Zooz know.

PR #2478 (2021-05-03): fix(config): update ZSE11 with ZWA import
fixes: #2474

**TODO:**
- [x]	Change PR title to be more specific