zwave-js/zwave-js #2475: Missing device configuration 0x0346:0x0101:0x0401:0.0

xhemp, 2021-04-28:
Ring Keypad v2

Manufacturer ID: 0x0346
Product type: 0x0101 
Product ID: 0x0401 (the one from ZWave Aliance says 0x0301, probably it's the difference between the US and the EU one (the one I have)).

zwavejs node dump:
[ring_keypad_v2.txt](https://github.com/zwave-js/node-zwave-js/files/6392645/ring_keypad_v2.txt)

PDF manual: https://products.z-wavealliance.org/ProductManual/File?folder=&filename=product_documents/3870/Z-WaveUserManual_Keypad_700.pdf

AlCalzone, 2021-04-28:
@zwave-js-bot import config 3870 from zwa

AlCalzone, 2021-04-28:
Let's see if this can get you started.

AlCalzone, 2021-04-28:
And again:
@zwave-js-bot import config 3870 from zwa

blhoward2, 2021-05-02:
On a different note, are you using this keypad with HA? How's it working? Can you set the code on it through HA? What does it send when activated/deactivated?

xhemp, 2021-05-02:
> On a different note, are you using this keypad with HA? How's it working? Can you set the code on it through HA? Does it jsut send the numbers pressed through to HA? What does it send when activated/deactivated?

Was trying to but it doesn't do anything really,only thing that actually show in the logs is the proximity control .

closed 2021-05-02 by AlCalzone

blhoward2, 2021-05-02:
Can you grab a silly level log from zwave-js while using it so we can see what it does?

xhemp, 2021-05-02:
> Can you grab a silly level log from zwave-js while using it so we can see what it does?

I would but any key I type doesn't really send anything (nothing show up in the logs), only thing I can see is the proximity sensor, I am close to return it as it's useless, unfortunately:

Here's a full inclusion log at silly level if it makes any difference.:

[ring_keypadv2-eu_inclusion_silly_level.txt](https://github.com/zwave-js/node-zwave-js/files/6411611/ring_keypadv2-eu_inclusion_silly_level.txt)

mattstocum, 2021-07-07:
I'm having the same problem, and after some testing with PC Z-Wave Controller 5, as best as I can tell, the keypad will only send entry control notifications when paired with S2 security. If it is paired with either S0 or no security it will send the very first button press, then nothing else until it is unpaired and re-paired. All other notifications from the keypad work just fine, and all commands sent to the keypad work. This behavior was consistent in both Z-Wave JS and PC Z-Wave Controller 5. I can try to grab additional logs if needed, but when I was watching at debug level, there was zero traffic being shown, not even raw serial commands coming in from the controller.

xhemp, 2021-07-07:
@mattstocum I got a v1 and it's working well as it doesn't need S2 to work.

IvanBrazza, 2021-08-16:
@mattstocum with the recent release of zwave-js supporting S2 I updated zwavejs2mqtt locally to use the new S2 functionality. I can include the ring keypad v2 with S2 enabled but seems the same situation still occurs. Only this time whenever there's a motion sensor event zwave-js throws a deserialisation error. It's almost as if the keypad is expecting something else in addition to S2. That or the recent update is missing something - unfortunately I don't have another S2 device I can check with...

AlCalzone, 2021-08-16:
Are you guys all using the node-red plugin or how are you including devices with S2 already? None of the other UIs I know have added support yet.

IvanBrazza, 2021-08-16:
@AlCalzone I made some tweaks to zwavejs2mqtt to start supporting S2 (https://github.com/IvanBrazza/zwavejs2mqtt/commit/3eb0cda05f174fff17d38f593f10a128ba469427)

Raised an issue in that repo with more details and silly logs, but I now realise that probably would have been better off in this repo! https://github.com/zwave-js/zwavejs2mqtt/issues/1572

AlCalzone, 2021-08-16:
I see... I've moved the issue to this repo. I figured it out with @marcus-j-davies already - the driver does not support responding to supervision commands yet. Just need the time to fix it.

IvanBrazza, 2021-08-16:
That's awesome thank you! I eagerly await your changes :)

Was definitely planning on raising a PR, this is my first foray into Node and all its accompanying treats so I'm slowly figuring it out as I go along.

PR #2477 (2021-05-02): feat(config): add Ring Keypad v2
fixes: #2475

**TODO:**
- [x]	Change PR title to be more specific