zwave-js/zwave-js #2480: Inovelli LZW36 Light turns on when Fan commanded on

kreene1987, 2021-04-28:
Since updating to 0.1.20, when my LZW Fan is commanded on, the light is also turning on. Here are the debug logs.

`2021-04-28T16:32:59.816Z CNTRLR   [Node 075] Mapping unsolicited report from root device to first supporting end
                                  point #1
2021-04-28T16:32:59.949Z CNTRLR   [Node 075] Mapping unsolicited report from root device to first supporting end
                                  point #1
2021-04-28T16:32:59.956Z CNTRLR   [Node 075] Mapping unsolicited report from root device to first supporting end
                                  point #1
2021-04-28T16:33:03.000Z CNTRLR   [Node 075] Mapping unsolicited report from root device to first supporting end
                                  point #1
2021-04-28T16:33:03.227Z CNTRLR   [Node 075] Mapping unsolicited report from root device to first supporting end
                                  point #1
2021-04-28T16:33:03.753Z CNTRLR   [Node 075] Mapping unsolicited report from root device to first supporting end
                                  point #1
2021-04-28T16:33:03.850Z CNTRLR   [Node 075] Mapping unsolicited report from root device to first supporting end
                                  point #1
2021-04-28T16:34:28.157Z CNTRLR   [Node 075] Mapping unsolicited report from root device to first supporting end
                                  point #1
2021-04-28T16:35:58.098Z CNTRLR   [Node 075] Mapping unsolicited report from root device to first supporting end
                                  point #1
2021-04-28T16:35:58.247Z CNTRLR   [Node 075] Mapping unsolicited report from root device to first supporting end
                                  point #1`

Probably related to #2159 and/or #2154.

kreene1987, 2021-04-28:
I'll get way better logs, these are useless...

blhoward2, 2021-04-28:
Don't worry about it, we know the issue we just missed readding a compat flag here as we work around it. I've pushed a PR fixing this.

blhoward2, 2021-04-28:
Actually, if memory serves this one acts a bit differently. Try removing the controller from associations group 1 (the lifeline) and re-adding it. OZW setup weird associations sometimes and it breaks this one.

@AlCalzone I think we should still add the compat flag until you fix #2257

kreene1987, 2021-04-28:
Can this be done from within ZJS, ZJS2M, or HA integration?

blhoward2, 2021-04-28:
Can what be done? The fix will roll out once merged in the next release. In the interim if you can get access to the device files inside the container you can change it on your end. How difficult that is depends on how you're running it and your level of familiarity.

kreene1987, 2021-04-28:
Sorry I wasn't clear. I was speaking to this:

> Try removing the controller from associations group 1 (the lifeline) and re-adding it

If ZJS2M can do it I can make it happen. No issues on the additional entity, I can handle that!

blhoward2, 2021-04-28:
Ah, yes. Go to the node in zjs2mqtt. Then go to the groups tab.

PR #2481 (2021-04-28): fix: add compat flag to lzw36
Closes #2480. After reverting #2160, this switch was missing the compat flag workaround we need until #2286 is solved. This PR re-adds it.
closed 2021-04-28 by AlCalzone