zwave-js/zwave-js #2482: Not able to add node

RobbyRash, 2021-04-28:
**Checklist:**

-   [X ] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
-   [X ] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
-   [X] I have checked the troubleshooting section and my problem is **not** described there.
-   [X ] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

Node can't be added in it's home location ~ 8 feet away from the nearest powered node.  It CAN be added next to the hub.

**Device information**

Which device(s) is/are affected (make/model)?  NAS-WR01ZE Switch/HUSBZB-1 Hub
What are the node IDs? 20/1

**Did you change anything?**

-   [ ] **Yes**: _(please describe)_
-   [X ] **No**

**Did this use to work before?**

-   [ ] **Don't know**, this is a new device
-   [X ] **No**, it never worked anywhere
-   [ ] **Yes**, in: _(specify application with versions)_

**How are you using `node-zwave-js`**

-   [ ] `zwavejs2mqtt` (latest) docker image
-   [ ] `zwavejs2mqtt` (dev) docker image
-   [ ] `zwavejs2mqtt` Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] ioBroker.zwave2 adapter
-   [X ] `HomeAssistant` version XYZ
-   [ ] Pkg
-   [ ] Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] `node-red-contrib-zwave-js`
    -   Module version (double click node): 
-   [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github))
-   [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:

1. Go to 'Configure'
2. Click on 'Add node'
3. Scroll down to 'Devices'
4. See error

**Additional context**

Steps taken in order:

1. Plugged switch in near hub (not home location) and added it.  Oops
2. Plugged switch in home location and it didn't work
3. Removed (exclusion process) from HA
4. Factory reset device
5. Restarted HA
6. Plugged switch in home location
7. Tried to add to HA and only get generic info and no response to switch commands
[zwave_js_dump (2).zip](https://github.com/zwave-js/node-zwave-js/files/6394749/zwave_js_dump.2.zip)

**Logfile:**

AlCalzone, 2021-04-28:
Please make a zwave-js log, loglevel `debug` or `silly` and attach it here as a file.
I can't do much without that.

RobbyRash, 2021-04-28:
transmit status: OK
2021-04-28T21:06:32.926Z SERIAL » 0x010800130301002502c1                                              (10 bytes)
2021-04-28T21:06:32.926Z DRIVER » [Node 003] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      2
                                  └─[NoOperationCC]
2021-04-28T21:06:32.928Z CNTRLR   [Node 002] The node is alive.
2021-04-28T21:06:32.928Z CNTRLR   [Node 002] The node is ready to be used
2021-04-28T21:06:32.929Z CNTRLR « [Node 002] ping successful
2021-04-28T21:06:32.933Z SERIAL « [ACK]                                                                   (0x06)
2021-04-28T21:06:32.937Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2021-04-28T21:06:32.938Z SERIAL » [ACK]                                                                   (0x06)
2021-04-28T21:06:32.939Z DRIVER « [RES] [SendData]
                                    was sent: true
2021-04-28T21:06:32.957Z SERIAL « 0x010500130200eb                                                     (7 bytes)
2021-04-28T21:06:32.957Z SERIAL » [ACK]                                                                   (0x06)
2021-04-28T21:06:32.958Z DRIVER « [REQ] [SendData]
                                    callback id:     2
                                    transmit status: OK
2021-04-28T21:06:32.968Z SERIAL » 0x010800130401002503c7                                              (10 bytes)
2021-04-28T21:06:32.972Z DRIVER » [Node 004] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      3
                                  └─[NoOperationCC]
2021-04-28T21:06:32.973Z CNTRLR   [Node 003] The node is alive.
2021-04-28T21:06:32.973Z CNTRLR   [Node 003] The node is ready to be used
2021-04-28T21:06:32.974Z CNTRLR « [Node 003] ping successful
2021-04-28T21:06:32.977Z SERIAL « [ACK]                                                                   (0x06)
2021-04-28T21:06:32.981Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2021-04-28T21:06:32.981Z SERIAL » [ACK]                                                                   (0x06)
2021-04-28T21:06:32.982Z DRIVER « [RES] [SendData]
                                    was sent: true
2021-04-28T21:06:33.010Z SERIAL « 0x010500130300ea                                                     (7 bytes)
2021-04-28T21:06:33.011Z SERIAL » [ACK]                                                                   (0x06)
2021-04-28T21:06:33.014Z DRIVER « [REQ] [SendData]
                                    callback id:     3
                                    transmit status: OK
2021-04-28T21:06:33.048Z SERIAL » 0x010800130601002504c2                                              (10 bytes)
2021-04-28T21:06:33.050Z DRIVER » [Node 006] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      4
                                  └─[NoOperationCC]
2021-04-28T21:06:33.051Z CNTRLR   [Node 004] The node is alive.
2021-04-28T21:06:33.052Z CNTRLR   [Node 004] The node is ready to be used
2021-04-28T21:06:33.052Z CNTRLR « [Node 004] ping successful
2021-04-28T21:06:33.074Z SERIAL « [ACK]                                                                   (0x06)
2021-04-28T21:06:33.075Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2021-04-28T21:06:33.076Z SERIAL » [ACK]                                                                   (0x06)
2021-04-28T21:06:33.077Z DRIVER « [RES] [SendData]
                                    was sent: true
2021-04-28T21:06:33.080Z SERIAL « 0x010500130400ed                                                     (7 bytes)
2021-04-28T21:06:33.081Z SERIAL » [ACK]                                                                   (0x06)
2021-04-28T21:06:33.082Z DRIVER « [REQ] [SendData]
                                    callback id:     4
                                    transmit status: OK
2021-04-28T21:06:33.087Z SERIAL » 0x010800131401002505d1                                              (10 bytes)
2021-04-28T21:06:33.088Z DRIVER » [Node 020] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      5
                                  └─[NoOperationCC]
2021-04-28T21:06:33.089Z CNTRLR   [Node 006] The node is alive.
2021-04-28T21:06:33.090Z CNTRLR   [Node 006] The node is ready to be used
2021-04-28T21:06:33.090Z CNTRLR « [Node 006] ping successful
2021-04-28T21:06:33.093Z SERIAL « [ACK]                                                                   (0x06)
2021-04-28T21:06:33.099Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2021-04-28T21:06:33.100Z SERIAL » [ACK]                                                                   (0x06)
2021-04-28T21:06:33.101Z DRIVER « [RES] [SendData]
                                    was sent: true
2021-04-28T21:06:37.196Z SERIAL « 0x010500130501ed                                                     (7 bytes)
2021-04-28T21:06:37.197Z SERIAL » [ACK]                                                                   (0x06)
[... 5 lines cut ...]
2021-04-28T21:06:37.204Z CNTRLR   All nodes are ready to be used
[... cut ...]
2021-04-28T21:06:41.307Z DRIVER « [REQ] [SendData]
                                    callback id:     6
                                    transmit status: NoAck
2021-04-28T21:06:41.310Z CNTRLR   [Node 020] Node 20 did not respond after 1 attempts, it is presumed dead
2021-04-28T21:06:41.311Z CNTRLR   [Node 020] ping failed: Failed to send the command after 1 attempts (Status No
                                  Ack)
2021-04-28T21:06:41.311Z CNTRLR   [Node 020] Interview attempt (1/5) failed, node is dead.
New client

AlCalzone, 2021-04-28:
> transmit status: NoAck
> 2021-04-28T21:06:41.310Z CNTRLR [Node 020] Node 20 did not respond after 1 attempts, it is presumed dead

The node does not respond or its responses don't reach the controller, so zwave-js cannot communicate with it. You could try healing the network so the device can learn how to reach the controller. Other than that, I don't see many options, except moving the device or adding another repeater between it and the controller.

RobbyRash, 2021-04-28:
First question.  What is the preferred method to heal the network with ZWave JS in Home Assistant?
Second question. The device is within 10 -15 feet of the nearest zwave node (plugged in).  Since ZWave is a mesh network, isn’t this OK?  Why does it have to be in close proximity to the controller?

AlCalzone, 2021-04-28:
> What is the preferred method to heal the network with ZWave JS in Home Assistant?

I'm not familiar with HA, you're better off asking on the HA discord. If you have zwavejs2mqtt, there are options to do it.

> The device is within 10 -15 feet of the nearest zwave node (plugged in).  Since ZWave is a mesh network, isn’t this OK?  Why does it have to be in close proximity to the controller?

To be honest, I don't know. Z-Wave is a mesh, but routing is handled by the chip firmwares. Maybe something is weird there, maybe your stick is in a bad location, maybe the desired location for the device is bad. Maybe you have interference somewhere.

blhoward2, 2021-04-29:
You can't heal in HA yet. You have to use zjs2mqtt.

ivdimitro, 2021-04-30:
On all zwave devices I have used it explicitly states to make the inclusion within the range of the controller. 
Also I think the routing is based on ids which are provided by the controller so I am not sure how a new devices can be routed to the controller it is is not already part of the network. 
Some controller have batteries so you can bring them close to the new device that needs inclusion. You can try with such controller.

AlCalzone, 2021-04-30:
> I am not sure how a new devices can be routed to the controller it is is not already part of the network.

There is a thing called network-wide inclusion, but the device to include has to support it. Sounds like yours don't.

> Some controller have batteries so you can bring them close to the new device that needs inclusion. You can try with such controller.

No, don't do that. https://github.com/zwave-js/node-zwave-js/issues/1995 for the reason why.

closed 2021-05-03 by zwave-js-assistant[bot]