zwave-js/zwave-js #2493: [bug] Unknown manufacturer/Unknown product after power loss

ldymek, 2021-04-30:
### Version

**Checklist:**

- [x] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [ ] I have checked the troubleshooting section and my problem is **not** described there.
- [ ] I have read the changelog problem was **not** mentioned there.

Build/Run method

- [x] Docker
- [ ] PKG
- [ ] Snap package
- [ ] Manually built (git clone - npm install - npm run build )

zwavejs2mqtt: 3.5.0
zwave-js: 7.2.4

### Describe the bug

After power loss all devices are not working properly, including zwave controller on usb, displaying Unknown manufacturer/Unknown product. The controller is showing after delete and pull docker image (it can be the same version) - devices has to be intervieved again (where went option to interview all devices at once, when I have to interviev 20+ devices it can make me crazy?).

### To Reproduce

Steps to reproduce the behavior:

Pull the power plug from RPi or Server with zwavejs2mqtt and power it again (like enviroment power loss)

### Expected behavior

Devices loaded correctly OR option to automatically interview them when all/part off them become Unknown

[nodes_dump_28.04.2021.json.zip](https://github.com/zwave-js/zwavejs2mqtt/files/6394197/nodes_dump_28.04.2021.json.zip)
[zwavejs2mqtt_28.04.2021.zip](https://github.com/zwave-js/zwavejs2mqtt/files/6394198/zwavejs2mqtt_28.04.2021.zip)
[zwavejs_28.04.2021.zip](https://github.com/zwave-js/zwavejs2mqtt/files/6394199/zwavejs_28.04.2021.zip)

beerygaz, 2021-05-01:
I can confirm I experienced the same issue.  All devices reverted to "Unknown manufacturer 0xXXXX" after a power loss.  Unfortunately I began the re-interview process before I found this thread so don't have the cache files in the state needed.  A "re-interview all" option would be a huge help in this scenario.

ldymek, 2021-05-04:
> They should be in the same folder as the logfiles. And to avoid any confusion: I need to see the status of the cache files after power loss but **before** zwave-js is started again.

so I have to disable docker autostart to achieve this... I had enviroment power loss few minutes ago and have the issue again.

kayintveen, 2021-05-16:
Had these issues as well as powering down the house to do some electrical work.

AlCalzone, 2021-05-16:
Any chance you could provide the files I requested above?

ldymek, 2021-05-16:
> Any chance you could provide the files I requested above?

Sorry I will prepare this in one, two days.

larstobi, 2021-05-25:
I have been hit by this same issue. I have 54 devices, so a Re-interview all button would be great.

robertsLando, 2021-05-25:
> I have been hit by this same issue. I have 54 devices, so a Re-interview all button would be great.

You could use a driver function

```js
for (const [, node] of driver.controller.nodes) {
      await node.refreshInfo()
}
```

AlCalzone, 2021-05-25:
> I have been hit by this same issue

Any chance you can provide the files I requested?

larstobi, 2021-05-25:
[cf48071b.zip](https://github.com/zwave-js/node-zwave-js/files/6539084/cf48071b.zip)

larstobi, 2021-05-25:
@AlCalzone I hope they don't contain sensitive information.

Unfortunately, I have already tried to re-interview the nodes, however, most of the battery ones have not already replied, so I still have lots of unknown in this.

larstobi, 2021-05-25:
I have seen corrupt cache files in other software caused by incorrect shutdowns, typically during abrupt power cuts.

Would it be too much to use an ACID-kind of cache storage system for this? I don't know if such systems exist for JavaScript/TypeScript.

Another solution could be to use a distributed network cache, like for example Redis with on-disk persistence, which would be power-cut crash-resistant.

AlCalzone, 2021-05-25:
Thanks, I've downloaded and deleted the file to be sure. The main problem here is to figure out which part of the cache breaks and how. I haven't been able to do this yet without seeing the files in their "broken" state.

We currently have two different systems in place, one is a JSON file that gets written as a whole, one is a mostly appendonly JSONL storage (or rather two), which takes quite a few ideas from redis. Both could be the reason, although I hope it is not the JSONL part.

robertsLando, 2021-05-25:
@AlCalzone How does JSONL handels writes on disk?

AlCalzone, 2021-05-25:
It keeps a file descriptor open, append linewise to this file immediately (or throttled based on the configuration). Additions or updates are written with a value, deletions without a value (just the key). When certain conditions are met, a compressed copy is written, the original gets closed, the compressed copy gets renamed to the original file (overwriting it) and the new original gets opened. Anything that should have been written in the meantime gets appended thereafter.

robertsLando, 2021-05-25:
It seems like https://github.com/louischatriot/nedb

robertsLando, 2021-05-25:
IMO the bug is in the JSONL, what could happen if the power loss happens while it is compressing the json file?

larstobi, 2021-05-25:
Another way to ensure writes are complete is to just use the atomic-rename-trick.
* Create new temp-file
* Write complete new JSONL structure to temp-file
* fsync() (if that's available in JavaScript)
* rename temp-file to the target filename (atomic operation)

AlCalzone, 2021-05-25:
> what could happen if the power loss happens while it is compressing the json file?

Then the original file does not get overwritten.

> Another way to ensure writes are complete is to just use the atomic-rename-trick.

That's what I do: https://github.com/AlCalzone/jsonl-db/blob/8c330850ab5be88864f69e91da73e4bd3c62fb58/src/lib/db.ts#L571-L576

larstobi, 2021-05-25:
In that case, I'm not convinced that incomplete file writes are the cause of the issue.

AlCalzone, 2021-05-25:
I just verified with someone who has this problem on every restart. The cache files are virtually identical before, after shutdown and after the startup. The problem is the communication with the stick that gets tripped up to the point it does not respond any more.

This seems to be caused by zwavejs2mqtt querying the neighbors for all nodes on startup (that one is on me) to draw the network map.

robertsLando, 2021-05-26:
Submitted a fix for this

closed 2021-05-26 by robertsLando