zwave-js/zwave-js #2497: MaxListenersExceededWarning - Since 4.0.0

declankenny, 2021-04-30:
I am using Home Assistant, but @robertsLando told me to post the bug here. 

After an upgrade to the 4.0.0 docker image, my container will not start. Attached is the log

```
/bin/sh: git: not found
2021-04-30 16:59:09.267 INFO APP: Version: 4.0.0
2021-04-30 16:59:09.272 INFO APP: Application path:/usr/src/app
(node:1) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 uncaughtException listeners added to [process]. Use emitter.setMaxListeners() to increase limit
(Use `node --trace-warnings ...` to show where the warning was created)
(node:1) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 SIGINT listeners added to [process]. Use emitter.setMaxListeners() to increase limit
(node:1) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 exit listeners added to [process]. Use emitter.setMaxListeners() to increase limit
```

I have since, reverted back to the 3.5.0 image and everything is working as expected.

AlCalzone, 2021-04-30:
The event listeners are added when the driver is created and removed when it is destroyed. This sounds like multiple instances of the driver are created without destroying them.

@robertsLando?

robertsLando, 2021-05-01:
@AlCalzone I create driver [here](https://github.com/zwave-js/zwavejs2mqtt/blob/master/lib/ZwaveClient.js#L1884) and when this fails for some reason I destroy it. Also I destroy it when ZwaveClient is closed so sincerly I have no idea what's going on here

AlCalzone, 2021-05-02:
@declankenny is there any more logging output or anything that could help us track this down?

declankenny, 2021-05-03:
I just recreated the docker container with the latest image, set the log level to silly. Attached in the start of zwavejs2mqtt.log

After this snippet all I get are repeats of `Controller status: Driver: this.container.getLogger is not a function` and `Retry connection in 3 seconds...`

```
2021-05-03 13:18:57.891 INFO ZWAVE: Controller status: Driver: this.container.getLogger is not a function
2021-05-03 13:18:57.880 INFO MQTT: MQTT is disabled
2021-05-03 13:18:57.900 INFO APP: Listening on port 8091 host 0.0.0.0 protocol HTTP
2021-05-03 13:18:57.893 WARN ZWAVE: Retry connection in 3 seconds...
2021-05-03 13:18:58.736 DEBUG SOCKET: New connection fMN1pzqmx7yp6bBxAAAB
2021-05-03 13:19:00.899 INFO ZWAVE: Controller status: Driver: this.container.getLogger is not a function
2021-05-03 13:19:00.900 WARN ZWAVE: Retry connection in 3 seconds...
2021-05-03 13:19:03.904 INFO ZWAVE: Controller status: Driver: this.container.getLogger is not a function
2021-05-03 13:19:03.905 WARN ZWAVE: Retry connection in 3 seconds...
2021-05-03 13:19:06.909 INFO ZWAVE: Controller status: Driver: this.container.getLogger is not a function
2021-05-03 13:19:06.910 WARN ZWAVE: Retry connection in 3 seconds...
2021-05-03 13:19:09.916 INFO ZWAVE: Controller status: Driver: this.container.getLogger is not a function
2021-05-03 13:19:09.917 WARN ZWAVE: Retry connection in 3 seconds...
2021-05-03 13:19:11.149 DEBUG SOCKET: User disconnected fMN1pzqmx7yp6bBxAAAB
2021-05-03 13:19:11.200 INFO APP: GET /settings 200 43.036 ms - 1371
2021-05-03 13:19:11.316 INFO APP: GET /static/js/app.a3320bb046b712ca125a.js 200 24.054 ms - 1750048
2021-05-03 13:19:11.345 INFO APP: GET /static/css/app.ef50254efc76b722abda.css 200 14.668 ms - 641511
2021-05-03 13:19:11.883 INFO APP: GET /api/auth-enabled 304 7.702 ms - -
2021-05-03 13:19:11.924 INFO APP: GET /static/favicons/favicon-32x32.png 200 4.975 ms - 1831
2021-05-03 13:19:12.199 INFO APP: GET /static/fonts/MaterialIcons-Regular.fa3334f.woff2 200 8.875 ms - 82492
2021-05-03 13:19:12.201 INFO APP: GET /static/logo.png 200 4.015 ms - 10014
2021-05-03 13:19:12.224 DEBUG SOCKET: New connection SzCIUhBSuxoG6GjYAAAD
2021-05-03 13:19:12.227 INFO APP: GET /static/favicons/favicon-32x32.png 304 4.327 ms - -
2021-05-03 13:19:12.247 DEBUG SOCKET: Event INITED emitted to SzCIUhBSuxoG6GjYAAAD
2021-05-03 13:19:12.335 INFO APP: GET /api/settings 200 131.109 ms - 1184
2021-05-03 13:19:12.395 INFO APP: GET /api/auth-enabled 304 3.029 ms - -
2021-05-03 13:19:12.399 INFO APP: GET /api/settings 304 46.751 ms - -
2021-05-03 13:19:12.436 INFO APP: GET /api/auth-enabled 304 2.032 ms - -
2021-05-03 13:19:12.922 INFO ZWAVE: Controller status: Driver: this.container.getLogger is not a function
2021-05-03 13:19:12.922 WARN ZWAVE: Retry connection in 3 seconds...
```

declankenny, 2021-05-03:
For reference, here is the services entry from my docker-compose file.

```
  zwavejs2mqtt:
    container_name: zwavejs2mqtt
    image: zwavejs/zwavejs2mqtt:3.5.0
    restart: ${RESTART_MODE}
    tty: true
    stop_signal: SIGINT
    devices:
      - '/dev/ttyACM0:/dev/ttyACM0'
    volumes:
      - ${DOCKERDIR}/zwavejs2mqtt/store:/usr/src/app/store
      - ${DOCKERDIR}/zwavejs2mqtt/config:/usr/src/app/node_modules/@zwave-js/config
    networks:
      macvlan:
        ipv4_address: 192.168.0.61
```

AlCalzone, 2021-05-03:
> Driver: this.container.getLogger is not a function

Now we're getting somewhere. And I think this may be the problem:

> - ${DOCKERDIR}/zwavejs2mqtt/config:/usr/src/app/node_modules/@zwave-js/config

You're mounting an incompatible version of a package.

declankenny, 2021-05-03:
@AlCalzone 
OK, that was it. I removed that volume and 4.0.0. is now working correctly.

closed 2021-05-03 by declankenny