zwave-js/zwave-js #2514: [bug] Custom log transport for Home Assistant is causing the zwjs2mqtt container to crash and restart

rccoleman, 2021-05-01:
### Version

**Checklist:**

- [x] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is **not** described there.
- [x] I have read the changelog problem was **not** mentioned there.

Build/Run method

- [x] Docker
- [ ] PKG
- [ ] Snap package
- [ ] Manually built (git clone - npm install - npm run build )

zwavejs2mqtt version: 4.0.0
zwave-js version: 7.3.0

### Describe the bug

I'm using the new custom log transport functionality with HA 2021.5.0b3 and as soon as I switch to the new logging tab in HA, HA sends the logging configuration to the server and the zwjs2mqtt container crashes and restarts.  I can reproduce this with the "latest" tag of the Docker hub zwavejs2mqtt container, my own container that I've built from the latest node-zwave-js and zwavejs2mqtt repos, and with the container that firstof9 is using successfully.

The only interesting log messages that I see when this happens are these from zwavejs2mqtt right at the point of failure, after which the container restarts:

```
This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason:
Error: write after end
    at writeAfterEnd (/usr/src/app/node_modules/readable-stream/lib/_stream_writable.js:261:12)
    at PassThrough.Writable.write (/usr/src/app/node_modules/readable-stream/lib/_stream_writable.js:305:21)
    at File.log (/usr/src/app/node_modules/winston/lib/winston/transports/file.js:185:34)
    at File._write (/usr/src/app/node_modules/winston-transport/index.js:103:17)
    at doWrite (/usr/src/app/node_modules/winston-transport/node_modules/readable-stream/lib/_stream_writable.js:428:64)
    at writeOrBuffer (/usr/src/app/node_modules/winston-transport/node_modules/readable-stream/lib/_stream_writable.js:417:5)
    at File.Writable.write (/usr/src/app/node_modules/winston-transport/node_modules/readable-stream/lib/_stream_writable.js:334:11)
    at DerivedLogger.ondata (/usr/src/app/node_modules/readable-stream/lib/_stream_readable.js:681:20)
    at DerivedLogger.emit (node:events:390:22)
    at DerivedLogger.EventEmitter.emit (node:domain:470:12)
```

I only see normal ZWave traffic prior to that, and then that, and then the container restarts.

The HA logs look like this:

```
2021-04-30 12:08:16 DEBUG (MainThread) [zwave_js_server] Received message:
WSMessage(type=<WSMsgType.TEXT: 1>, data='{"type":"result","success":true,"messageId":"cbf8b366f65b4fdc81e33a905044536c","result":{"config":{"enabled":true,"level":"debug","logToFile":true,"filename":"/usr/src/app/store/zwavejs_1.log","forceConsole":true}}}', extra='')
2021-04-30 12:08:40 DEBUG (MainThread) [zwave_js_server] Received message:
WSMessage(type=<WSMsgType.TEXT: 1>, data='{"type":"event","event":{"source":"driver","event":"logging","formattedMessage":"2021-04-30T19:08:40.568Z DRIVER   destroying driver instance...","level":"verbose","message":"destroying driver instance...","direction":"  ","label":"DRIVER","timestamp":"2021-04-30T19:08:40.568Z","multiline":false,"secondaryTagPadding":-1}}', extra='')
2021-04-30 12:08:40 DEBUG (MainThread) [zwave_js_server] Listen completed. Cleaning up
2021-04-30 12:08:40 INFO (MainThread) [homeassistant.components.zwave_js] Disconnected from server. Reloading integration
```

From that, it appears that HA attempted to set the logging parameters, and 24s later it received a message about the driver instance being destroyed.

The log window in the zwavejs2mqtt web UI correctly shows the logs and the log level adjusts based on what I've configured in the web UI, so it seems to only affect interaction with HA over the custom log transport.

It seems like I may be the only one currently seeing this issue, as others on Discord do not.  You can see the full discussion here: https://discord.com/channels/330944238910963714/427516175237382144/837746193193762866.

I suspect that it has something to do with the files in my Store, but I wanted to get some guidance on where to look before going scorched earth on my working, production config.

### To Reproduce

Steps to reproduce the behavior:

Using Home Assistant 2021.5.0b3 and a properly configured zwavejs2mqtt container connected through the websocket server, go to Configuration->Integrations->Zwave_js->Configure->Logs.  Note that the only message that appears in the log window is:

`2021-04-30T21:36:42.095Z DRIVER   destroying driver instance...`

This message indicates that the zwavejs2mqtt container crashed, and will repeat if I refresh the page, or the page loses focus and regains it.  Logging in the zwavejs2mqtt log window appears normally.

### Expected behavior

I expect the same logging that shows up in the zwavejs2mqtt logger UI to show up in HA, and for the zwavejs2mqtt container to not crash and restart.

### Additional context

Add any other context about the problem here.

rccoleman, 2021-05-05:
I suspect that it's just another innocent victim of a crash that brings down the container, but I have both node-zwave-js and zwavejs2mqtt debug on "silly" and I don't see anything else.  I'll try to get some more information about what's going on.

rccoleman, 2021-05-17:
I just pulled the latest node-zwave-js and zwavejs2mqtt and built my own container with the following versions:

```
zwavejs2mqtt: 4.2.1+045889b.045889b
zwave-js: 7.4.0-501238756b

Home Assistant 2021.5.4
```

Now, pulling up the logs in Home Assistant works the first time I navigate to that tab, and I see the logs as expected without the container crashing.  Unfortunately, navigating away from the tab and going back in or changing the debug level while staying on the tab seems to cause the zwavejs2mqtt container to hang.  After that, I get no logs at all from node-zwave-js or zwavejs2mqtt and my zwave network is non-functional.  Restarting the container brings it back, and I can repeat the behavior at will.  The only logs that I see in HA are those related to restarting the container.

I'll keep poking around, but the behavior has changed in the latest builds.

AlCalzone, 2023-02-01:
Since this is the only time I've ever seen this issue reported and no way to reproduce, I'm afraid I'll have to close this.

closed 2023-02-01 by AlCalzone