zwave-js/zwave-js #2516: Feature Request: Show firmware versions of all MCUs if device has more than 1

jtronicus, 2021-05-01:
**Is your feature request related to a problem? Please describe.**
Some devices (such as the Inovelli LZW31-SN Red Series Dimmer) have multiple microcontroller units, each with its own firmware (target 0, target 1, etc).  Right now, I can update the firmware of both MCUs, but can only see the current firmware version for Target 0.

**Describe the solution you'd like**
It would be nice if the firmware version for all MCUs is captured.  I can see that the device is reporting both firmwares using a zwave packet sniffer, but I am not sure if this information is being exposed to zwavejs2mqtt in the first place.

**Additional context**
[image]

```
2021-05-01T15:01:41.012Z SERIAL « 0x0111000400310b861203060401370101012c5e                            (19 bytes)
2021-05-01T15:01:41.013Z CNTRLR   [Node 049] [+] [Version] libraryType: 3                           [Endpoint 0]
2021-05-01T15:01:41.013Z CNTRLR   [Node 049] [+] [Version] protocolVersion: "6.4"                   [Endpoint 0]
2021-05-01T15:01:41.015Z CNTRLR   [Node 049] [+] [Version] firmwareVersions: 1.55                   [Endpoint 0]
2021-05-01T15:01:41.015Z SERIAL » [ACK]                                                                   (0x06)
2021-05-01T15:01:41.016Z DRIVER « [Node 049] [REQ] [ApplicationCommand]
                                  └─[VersionCCReport]
                                      library type:      Enhanced Slave
                                      protocol version:  6.4
                                      firmware versions: 1.55
2021-05-01T15:01:41.017Z CNTRLR « [Node 049] received response for node versions:
                                    library type:      Enhanced Slave (0x03)
                                    protocol version:  6.4
                                    firmware versions: 1.55
```

AlCalzone, 2021-05-04:
Can you share a more complete log? I think this is due to the fact that the device claims it supports `Version CC` in version 1, but uses a V2 command.

jtronicus, 2021-05-04:
I re-enabled logs, then did a full interview of the node in question.  Interview starts at `2021-05-04T12:47:19.643Z`

```
2021-05-04T12:46:15.387Z DRIVER   ███████╗ ██╗    ██╗  █████╗  ██╗   ██╗ ███████╗             ██╗ ███████╗
                                  ╚══███╔╝ ██║    ██║ ██╔══██╗ ██║   ██║ ██╔════╝             ██║ ██╔════╝
                                    ███╔╝  ██║ █╗ ██║ ███████║ ██║   ██║ █████╗   █████╗      ██║ ███████╗
                                   ███╔╝   ██║███╗██║ ██╔══██║ ╚██╗ ██╔╝ ██╔══╝   ╚════╝ ██   ██║ ╚════██║
                                  ███████╗ ╚███╔███╔╝ ██║  ██║  ╚████╔╝  ███████╗        ╚█████╔╝ ███████║
                                  ╚══════╝  ╚══╝╚══╝  ╚═╝  ╚═╝   ╚═══╝   ╚══════╝         ╚════╝  ╚══════╝
2021-05-04T12:46:15.389Z DRIVER   version 7.2.4
2021-05-04T12:46:15.389Z DRIVER   
2021-05-04T12:46:15.389Z DRIVER   starting driver...
2021-05-04T12:46:15.391Z DRIVER   opening serial port /dev/ttyACM0
2021-05-04T12:46:15.397Z DRIVER   serial port opened
2021-05-04T12:46:15.397Z SERIAL » [NAK]                                                                   (0x15)
2021-05-04T12:46:16.902Z DRIVER   loading configuration...
2021-05-04T12:46:17.201Z DRIVER   beginning interview...
2021-05-04T12:46:17.202Z DRIVER   added request handler for AddNodeToNetwork (0x4a)...
                                  1 registered
2021-05-04T12:46:17.202Z DRIVER   added request handler for RemoveNodeFromNetwork (0x4b)...
                                  1 registered
2021-05-04T12:46:17.202Z DRIVER   added request handler for ReplaceFailedNode (0x63)...
                                  1 registered
2021-05-04T12:46:17.202Z CNTRLR   beginning interview...
2021-05-04T12:46:17.202Z CNTRLR   querying version info...
2021-05-04T12:46:17.204Z SERIAL » 0x01030015e9                                                         (5 bytes)
2021-05-04T12:46:17.204Z DRIVER » [REQ] [GetControllerVersion]
2021-05-04T12:46:17.205Z SERIAL « [ACK]                                                                   (0x06)
2021-05-04T12:46:17.206Z SERIAL « 0x011001155a2d5761766520332e3935000199                              (18 bytes)
2021-05-04T12:46:17.207Z SERIAL » [ACK]                                                                   (0x06)
2021-05-04T12:46:17.207Z DRIVER « [RES] [GetControllerVersion]
                                    payload: 0x5a2d5761766520332e39350001
2021-05-04T12:46:17.208Z CNTRLR   received version info:
                                    controller type: Static Controller
                                    library version: Z-Wave 3.95
2021-05-04T12:46:17.208Z CNTRLR   querying controller IDs...
2021-05-04T12:46:17.209Z SERIAL » 0x01030020dc                                                         (5 bytes)
2021-05-04T12:46:17.209Z DRIVER » [REQ] [GetControllerId]
2021-05-04T12:46:17.210Z SERIAL « [ACK]                                                                   (0x06)
2021-05-04T12:46:17.211Z SERIAL « 0x01080120e8fa50a60133                                              (10 bytes)
2021-05-04T12:46:17.211Z SERIAL » [ACK]                                                                   (0x06)
2021-05-04T12:46:17.212Z DRIVER « [RES] [GetControllerId]
                                    payload: 0xe8fa50a601
2021-05-04T12:46:17.212Z CNTRLR   received controller IDs:
                                    home ID:     0xe8fa50a6
                                    own node ID: 1
2021-05-04T12:46:17.212Z CNTRLR   querying controller capabilities...
2021-05-04T12:46:17.213Z SERIAL » 0x01030005f9                                                         (5 bytes)
2021-05-04T12:46:17.213Z DRIVER » [REQ] [GetControllerCapabilities]
2021-05-04T12:46:17.214Z SERIAL « [ACK]                                                                   (0x06)
2021-05-04T12:46:17.215Z SERIAL « 0x010401051ce3                                                       (6 bytes)
2021-05-04T12:46:17.215Z SERIAL » [ACK]                                                                   (0x06)
2021-05-04T12:46:17.215Z DRIVER « [RES] [GetControllerCapabilities]
                                    payload: 0x1c
2021-05-04T12:46:17.216Z CNTRLR   received controller capabilities:
                                    controller role:     primary
                                    is in other network: false
                                    is SIS present:      true
                                    was real primary:    true
                                    is a SUC:            true
2021-05-04T12:46:17.216Z CNTRLR   querying API capabilities...
2021-05-04T12:46:17.218Z SERIAL » 0x01030007fb                                                         (5 bytes)
2021-05-04T12:46:17.218Z DRIVER » [REQ] [GetSerialApiCapabilities]
2021-05-04T12:46:17.219Z SERIAL « [ACK]                                                                   (0x06)
2021-05-04T12:46:17.222Z SERIAL « 0x012b0107010000860101005afe81ff884f1f0000fb9f7da067000080008086000 (45 bytes)
                                  000e87300000e0000600000fa
2021-05-04T12:46:17.222Z SERIAL » [ACK]                                                                   (0x06)
2021-05-04T12:46:17.222Z DRIVER « [RES] [GetSerialApiCapabilities]
                                    payload: 0x010000860101005afe81ff884f1f0000fb9f7da067000080008086000000e8730
                                  0000e0000600000
[... 2480 lines cut ...]
[... cut ...]
                                      groups:
                                      · Group #4
                                        mode:       0
                                        profile:    8195
                                        event code: 0
2021-05-04T12:47:22.824Z CNTRLR « [Node 049] Received info for association group #4:
                                  info is dynamic: false
                                  profile:         Control: Key 03
2021-05-04T12:47:22.824Z CNTRLR   [Node 049] [

AlCalzone, 2021-05-04:
Ahh - the CC version is requested **after** the query that needs it.

PR #2556 (2021-05-05): fix(cc): query Version CC version before relying on it
fixes: #2516
closed 2021-05-05 by AlCalzone