zwave-js/zwave-js #2518: Missing device configuration 0x0441-0x2400-0x1000

yuvalmesika, 2021-05-02:
**Which device is missing?**

Nice IBT4 BusT4-Z-Wave interface

**What are the IDs?**

Manufacturer ID: 0x1234  
Product type: 0x4321  
Product ID: 0x0101  

**Do you have a link to the PDF manual?**

link to the PDF manual:
https://products.z-wavealliance.org/ProductManual/File?folder=&filename=product_documents/3837/IBT4ZWAVE-T-v0.7.pdf

link to z-wavealliance:
https://products.z-wavealliance.org/products/3837?selectedFrequencyId=-1

export of my node:
[node_dump.txt](https://github.com/zwave-js/node-zwave-js/files/6410839/node_dump.txt)

cdnninja, 2021-05-03:
@zwave-js-bot import config 3837 from zwa

cdnninja, 2021-05-03:
I believe Ids are 0x0441.0x2400.0x1000.   @AlCalzone  could you do the import for me and I can cleanup?  If not I can start from scratch too.

AlCalzone, 2021-05-03:
@zwave-js-bot import config 3837 from zwa

@cdnninja you can just make a PR onto the PR the bot will create.

PR #2525 (2021-05-10): feat(config): add config file for Nice IBT4 BusT4
fixes: #2518

**TODO:**
- [x]	Change PR title to be more specific
closed 2021-05-10 by AlCalzone