zwave-js/zwave-js #2524: Zooz Zen77 Inclusion

jesse-schein, 2021-05-02:
Cant seem to add a zooz zen77 700 series dimmer with my aeotec ZW090 controller. This is controller is the new version that works with the raspbery pi 4.

Here are the steps I did:

1. Put controller into inclusion
2. Put device in inclusion
3. dongle is plugged into pi which is 12 inches away from switch
4. inclusion fails

I then tried the same steps but with secured inclusion which also fails.

I then tried this with the switch next to it and that also failed.

Any idea?

2021-05-02 15:08:30.306 DEBUG SOCKET: User disconnected 4BRXe5y6SBKqu3CbAAAL
2021-05-02 15:08:30.328 INFO APP: GET /control-panel 304 8.053 ms - -
2021-05-02 15:08:30.400 INFO APP: GET /static/css/app.ef50254efc76b722abda.css 304 1.875 ms - -
2021-05-02 15:08:30.406 INFO APP: GET /static/js/app.a3320bb046b712ca125a.js 304 1.478 ms - -
2021-05-02 15:08:31.162 INFO APP: GET /api/auth-enabled 304 4.126 ms - -
2021-05-02 15:08:31.399 INFO APP: GET /static/logo.png 304 2.067 ms - -
2021-05-02 15:08:31.753 INFO APP: GET /static/fonts/MaterialIcons-Regular.fa3334f.woff2 304 2.200 ms - -
2021-05-02 15:08:31.844 DEBUG SOCKET: New connection 1SHzXl6Absn6zAfEAAAN
2021-05-02 15:08:31.929 INFO APP: GET /api/settings 200 121.835 ms - 11272
2021-05-02 15:08:31.950 DEBUG SOCKET: Event INITED emitted to 1SHzXl6Absn6zAfEAAAN
2021-05-02 15:08:32.105 INFO APP: GET /static/favicons/favicon-32x32.png 304 1.217 ms - -
2021-05-02 15:08:32.144 INFO APP: GET /api/auth-enabled 304 2.157 ms - -
2021-05-02 15:08:35.138 DEBUG SOCKET: Event ZWAVE_API emitted to 1SHzXl6Absn6zAfEAAAN
2021-05-02 15:08:35.141 INFO ZWAVE: Calling api startInclusion with args: [ false, [length]: 1 ]
15:08:35.143 CNTRLR starting inclusion process...
15:08:35.160 SERIAL » 0x0105004ac10475 (7 bytes)
15:08:35.162 DRIVER » [REQ] [AddNodeToNetwork]
payload: 0xc104
15:08:35.171 SERIAL « [ACK] (0x06)
15:08:35.179 SERIAL « 0x0107004a04010000b7 (9 bytes)
15:08:35.181 SERIAL » [ACK] (0x06)
15:08:35.183 DRIVER « [REQ] [AddNodeToNetwork]
payload: 0x04010000
15:08:35.186 DRIVER handling request AddNodeToNetwork (74)
15:08:35.187 DRIVER 1 handler registered!
15:08:35.187 DRIVER invoking handler #0
15:08:35.188 CNTRLR handling add node request (status = Ready)
15:08:35.188 CNTRLR the controller is now ready to add nodes
2021-05-02 15:08:35.189 INFO ZWAVE: Controller status: Non-secure inclusion started
15:08:35.191 DRIVER the message was handled
2021-05-02 15:08:35.192 INFO ZWAVE: Success zwave api call startInclusion { success: true, message: 'Success zwave api call', result: true }
15:08:44.226 SERIAL « 0x0107004a04020000b4 (9 bytes)
15:08:44.229 SERIAL » [ACK] (0x06)
15:08:44.233 DRIVER « [REQ] [AddNodeToNetwork]
payload: 0x04020000
15:08:44.239 DRIVER handling request AddNodeToNetwork (74)
15:08:44.241 DRIVER 1 handler registered!
15:08:44.242 DRIVER invoking handler #0
15:08:44.243 CNTRLR handling add node request (status = NodeFound)
15:08:51.932 SERIAL « 0x0107004a04070000b1 (9 bytes)
15:08:51.936 SERIAL » [ACK] (0x06)
15:08:51.942 DRIVER « [REQ] [AddNodeToNetwork]
payload: 0x04070000
15:08:51.949 DRIVER handling request AddNodeToNetwork (74)
15:08:51.951 DRIVER 1 handler registered!
15:08:51.952 DRIVER invoking handler #0
15:08:51.953 CNTRLR handling add node request (status = Failed)
15:08:51.954 CNTRLR starting the inclusion failed
15:08:51.956 CNTRLR stopping inclusion process...
15:08:51.972 SERIAL » 0x0105004ac50570 (7 bytes)
15:08:51.974 DRIVER » [REQ] [AddNodeToNetwork]
payload: 0xc505
15:08:51.989 SERIAL « [ACK] (0x06)
15:08:52.001 DRIVER the message was handled
15:08:52.032 SERIAL « 0x0107004a05060000b1 (9 bytes)
[... 94 lines cut ...]
15:09:49.904 DRIVER « [REQ] [AddNodeToNetwork]
payload: 0x08070000
15:09:49.907 DRIVER handling request AddNodeToNetwork (74)
15:09:49.908 DRIVER 1 handler registered!
15:09:49.909 DRIVER invoking handler #0
15:09:49.909 CNTRLR handling add node request (status = Failed)
15:09:49.910 CNTRLR starting the inclusion failed
15:09:49.911 CNTRLR stopping inclusion process...
15:09:49.920 SERIAL » 0x0105004ac5097c (7 bytes)
15:09:49.921 DRIVER » [REQ] [AddNodeToNetwork]
payload: 0xc509
15:09:49.926 SERIAL « [ACK] (0x06)
15:09:49.937 DRIVER the message was handled
15:09:49.976 SERIAL « 0x0107004a09060000bd (9 bytes)
15:09:49.978 SERIAL » [ACK] (0x06)
15:09:49.980 DRIVER « [REQ] [AddNodeToNetwork]
payload: 0x09060000
15:09:49.983 DRIVER handling request AddNodeToNetwork (74)
15:09:49.984 DRIVER 1 handler registered!
15:09:49.984 DRIVER invoking handler #0
15:09:49.985 CNTRLR handling add node request (status = Done)
15:09:49.986 CNTRLR done called for 0
15:09:49.986 CNTRLR the inclusion process was stopped
2021-05-02 15:09:49.987 INFO ZWAVE: Controller status: Inclusion stopped
15:09:49.988 DRIVER the message was handled

AlCalzone, 2021-05-02:
Since the stick only tells us that the inclusion failed, we cannot be sure what's happening without a Zniffer capture.

You could try to power-cycle the stick, meaning: shut down, pull it, put it back in, start back up.

jesse-schein, 2021-05-02:
I found out all my issues... Using a usb stick with my raspberry pi having the usb plugged directly in must have been causing interference. 

I put it on a cheapo usb hub and all is well now.

closed 2021-05-02 by jesse-schein