zwave-js/zwave-js #2526: Fibaro unknown manufacturer/product

baleksx, 2021-05-03:
Hi!
I have Fibaro FGWPF-102 (device ID is 0x010f-0x0602-0x4003) - this is the same that Product Identifiers: 0x0602:0x1001, 0x0602:0x1003, but RU version. Also i have FGKF601 (device ID is 0x010f-0x1001-0x4000) - RU version. There is only PID 0x1001:0x1000, 0x1001:0x2000, 0x1001:0x3000 in z-wave js database.
Both fibaro devices sowed as Unknown manufacturer 0x010f/Unknown product 0x4003 for FGWPF-102 and Unknown manufacturer 0x010f/Unknown product 0x4000 for FGKF601.
Also there aren't "Configuration" section to manage devise parameters. As they the same versions of devices but Ru versions, how i can make to detect they correctly?
[image]

AlCalzone, 2021-05-03:
@baleksx you can create a PR here to add the missing device identifier to the existing file.

baleksx, 2021-05-04:
how can I do this? unfortunately I have little experience for this

marcus-j-davies, 2021-05-05:
@baleksx,

Have taken the liberty of tasing the PRs.
Although what Firmware version is your  FGWPF-102?

closed 2021-05-05 by AlCalzone

baleksx, 2021-05-05:
@marcus-j-davies, firmvare version is 3.2

marcus-j-davies, 2021-05-05:
Thats fine, the change is now merged, so should become available on the next release.