zwave-js/zwave-js #2533: Increase network heal timeouts

blhoward2, 2021-05-03:
Current timeouts are calculated too aggressively, which can cause multiple heals to be requested before the device can respond on large networks.

AlCalzone, 2021-05-04:
Was this still an issue when you healed nodes 1-by-1 with the new stick firmware?

blhoward2, 2021-05-04:
Nope. None failed going one by one I don't think. I'll look at the log later. I healed all individually working out from the stick and then tried a network heal to see what would happen.

blhoward2, 2021-05-05:
Logs:

[one-by-one.log](https://github.com/zwave-js/node-zwave-js/files/6424861/one-by-one.log)
[network_heal.log](https://github.com/zwave-js/node-zwave-js/files/6424860/network_heal.log)