zwave-js/zwave-js #2534: Node values not correctly identified

mcozzo, 2021-05-03:
-   [ ] a developer has told me to come here.

**Describe the bug**

The notification value metadata is missing the proper type (number) and states (Idle + monitoring)

Debug: https://pastebin.com/Cgs180A6
Debug: https://pastebin.com/6J5pVQMs
Debug: https://pastebin.com/1aEeS9C1 

States: Motion, tamper are provided to MQTT but do not show up in HomeAssistant. 

Device: http://www.getzooz.com/zooz-zse40-4-in-1-sensor.html
Node 23,24

Has never worked. 
zwavejs2mqtt: 4.0.0
zwave-js: 7.3.0

All instances are deployed through docker. 
  image: zwavejs/zwavejs2mqtt:latest
  image: homeassistant/home-assistant:stable

**To Reproduce**

Steps to reproduce the behavior:

1. Re-interview node
2. Restart / update Home Assistant
3. components are missing

AlCalzone, 2021-05-03:
This is the metadata I'm getting when zwave-js parses the received command for supported events I found in your log:
```
// value ID {"commandClass":113,"endpoint":0,"property":"Home Security","propertyKey":"Cover status"}

{
  type: 'number',
  readable: true,
  writeable: false,
  min: 0,
  max: 255,
  label: 'Cover status',
  states: { '0': 'idle', '3': 'Tampering, product cover removed' },
  ccSpecific: { notificationType: 7 }
}
```
and
```
// value ID {"commandClass":113,"endpoint":0,"property":"Home Security","propertyKey":"Motion sensor status"}

{
  type: 'number',
  readable: true,
  writeable: false,
  min: 0,
  max: 255,
  label: 'Motion sensor status',
  states: { '0': 'idle', '8': 'Motion detection' },
  ccSpecific: { notificationType: 7 }
}
```
This looks good and I also have a sensor with the same notifications which works fine in another software. So I'd say this works correctly in zwave-js, but I'm a bit at a loss where this breaks... zwavejs2mqtt, zwave-js-server or HA.

mcozzo, 2021-05-03:
I had gotten the sensors installed over the weekend. The issue I was running into was from zwave-js to HA. HA wasn't seeing the additional properties, motion, tamper sensor. It appeared that the additional sensors were showing up as "unknown" when I would interview the node. I tried restarting both HA & zwave-js-server with no change. I figured leaving it for a couple days would help. 

HA could see all the sensors through MQTT.
Through zwave-js-server, HA would only see battery, illuminence, humidity. 

I did a number of RE-Interview tasks while we were talking this morning before I opened the ticket. Everything got a restart as well. Nothing seemed to update HA. Now after leaving it sit for the day and working on other stuff it's now reporting all of the device states. 

This was where we got before I opened the ticket: 
```
2021-05-03T17:52:30.540Z DRIVER « [Node 023] [REQ] [ApplicationCommand]
                                  └─[NotificationCCEventSupportedReport]
                                      notification type: Home Security
                                      supported events:
                                      · Unknown (0x03)
                                      · Unknown (0x08)
```
"it discovers the supported events (cover tamper and motion), but thinks they're "unknown""

At this point it's working. If there's nothing obvious to touch I think it's fine to close. My apologies for not opening the ticket correctly.

AlCalzone, 2021-05-04:
> "it discovers the supported events (cover tamper and motion), but thinks they're "unknown""

That's a bug in the code for logging. The events are correctly supported but the logging doesn't look up the correct name.

I think this **might** be a bug in zwave-js-server, but I'm not 100% sure. Working theory:
- Notification comes in before the device is interviewed
- zwave-js-server requests the metadata, gets the fallback (type any), because there isn't any yet
- Interview completes, correct metadata gets saved
- zwave-js-server doesn't update the default metadata it has