zwave-js/zwave-js #2535: color CC for warm color not available for ZDB5100

Roagert, 2021-05-03:
I have found that my device "ZDB5100" from "Logic Group" has this feature that it can enabled an "enhanced LED" function.
This function allows the device to use the color CC "warm" to set a blinking feature.

See definition below:
`When the parameter for extended LED control is switched on (parameter number 10 is set to the value '1'), there is
possibility to change lights and colors for all the different LED indications using the Command Class Switch
Color Set commands. Furthermore, it is possible to make the light indicator flash at different frequencies.
To control the color in the LED groups, standard Command Class Switch Color Set commands are used
values ​​for red, green and blue, respectively. But to select which status type the command is valid for,
the field for warm white (ID: 0) is used to indicate different values ​​according to the table below:
Warm white field in Command Class Switch Color Set
Bit # 7 6 5 4 3 2 1 0
Blink Off
status
Switch on
status
Direct
control Time for flashing frequency for 0.1 secondMATRIX ZDB5100 User Manual GB
Logic Group A / S Page 13/39
Descriptions of the various functions are shown in the table below.
Flash When this bit is set, the LEDs will flash with the color levels indicated in the rest of
command, the time in bits 0 - 3 indicate the speed of the flash.
Off status The command applies to lights in the off / passive LEDs (key not
activated).
Switch on status The command is valid for lights in the LEDs on / active mode (key activated).
Direct
control
The command is valid for a temporary override of the LEDs that will be valid
until the next key activation (or a Z-Wave command).
Time
These 4 bits indicate the time of the blink frequency (valid only if the Blink bit is set), where the time
is specified for 0.1 second and the time is for the half period (time for light on or off).
For example, the value 5 indicates 0.5 seconds, i.e. the LEDs are on for 0.5 seconds, off for 0.5
seconds, ie a period time of 1 second, corresponding to a 1 Hz flashing frequency.`

Log Attached.
Node 3 is one of the devices that has this feature.
The parameter for enabling this is parameter #10

[zwavejs.log](https://github.com/zwave-js/node-zwave-js/files/6417068/zwavejs.log)

AlCalzone, 2021-05-03:
I found it in the log you posted on Slack:
```
[Node 003] [REQ] [ApplicationCommand]
└─[MultiChannelCCCommandEncapsulation]
  │ source:      1
  │ destination: 0
  └─[ColorSwitchCCSupportedReport]
      supported color components:
      · Red
      · Green
      · Blue
```
(same for the other endpoints). This is **after** you configured the device, so it just doesn't tell us that it has a warm white channel.

Long story short, you'll probably have to go through the `commandClasses` API to control the white channel.
To do so, call the `set` method https://zwave-js.github.io/node-zwave-js/#/api/CCs/ColorSwitch?id=set with `{"warmWhite": 2}` (replace the 2 with the value you want to set). This API is exposed in zwavejs2mqtt too.

Roagert, 2021-05-03:
If this is exposed in zwavejs2mqtt Can this then be controller through HA ?

kpine, 2021-05-03:
No, HA does not currently have a way to use the `commandClasses` API.

Roagert, 2021-05-04:
> No, HA does not currently have a way to use the `commandClasses` API.

Is it on the roadmap ?

kpine, 2021-05-04:
According to [this](https://github.com/home-assistant-libs/zwave-js-server-python/issues), no.

Roagert, 2021-05-04:
> According to [this](https://github.com/home-assistant-libs/zwave-js-server-python/issues), no.

Could there be a potential “advanced” feature set that could be added in the future to enable this ? 
Maybe it’s not on the roadmap now, but i Can become?

closed 2021-05-07 by zwave-js-assistant[bot]