zwave-js/zwave-js #2538: Supervision Report not being sent by device: ZL-PD-100

tseven, 2021-05-04:
**Checklist:**

-   [x] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
-   [x] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
-   [x] I have checked the troubleshooting section and my problem is **not** described there.
-   [x] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

Home Assistant's UI does not update to reflect the changes in state of the device when the state is changed in Node Red or via the physical button on the device.

Also when toggling the state from the Device UI the switch will revert back previous position after a few seconds.

It appears to not be sending a Supervision Report back.

**Device information**

Which device(s) is/are affected (make/model)?  
ZL-PD-100
by zwaveproducts.com
Firmware: 50.3
https://products.z-wavealliance.org/products/2809?selectedFrequencyId=-1
https://cdn.shopify.com/s/files/1/0067/9814/7669/files/ZL-PD-100_Users_Guide.pdf?953

What are the node IDs?
Node ID: 7

**Did you change anything?**

-   [ ] **Yes**: _(please describe)_
-   [x] **No**

**Did this use to work before?**

-   [ ] **Don't know**, this is a new device
-   [ ] **No**, it never worked anywhere
-   [x] **Yes**, in: OpenZWave 0.9.1

**How are you using `node-zwave-js`**

-   [ ] `zwavejs2mqtt` (latest) docker image
-   [ ] `zwavejs2mqtt` (dev) docker image
-   [ ] `zwavejs2mqtt` Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] ioBroker.zwave2 adapter
-   [x] `HomeAssistant` version 2021.4.6
-   [ ] Pkg
-   [ ] Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] `node-red-contrib-zwave-js`
    -   Module version (double click node): 
-   [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github))
-   [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:

1. Go to 'Configruation' --> Devices
2. Click on 'Dimmer device'
3. Trigger the device to be on or off from Node Red or from physical button
4. UI in HA is not updated.
5. When toggling the state from the Device UI the switch will revert back previous position after a few seconds.

**Additional context**

Add any other context about the problem here.

**Logfile:**
[z-wave-silly.log](https://github.com/zwave-js/node-zwave-js/files/6424446/z-wave-silly.log)

blhoward2, 2021-05-05:
From the log, this happens each time:

```
2021-05-04T22:53:26.298Z DRIVER » [Node 007] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      23
                                  └─[SupervisionCCGet]
                                    │ session id:      22
                                    │ request updates: true
                                    └─[MultilevelSwitchCCSet]
                                        target value: 0
2021-05-04T22:53:26.303Z SERIAL « [ACK]                                                                   (0x06)
2021-05-04T22:53:26.306Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2021-05-04T22:53:26.307Z SERIAL » [ACK]                                                                   (0x06)
2021-05-04T22:53:26.308Z DRIVER « [RES] [SendData]
                                    was sent: true
2021-05-04T22:53:26.321Z SERIAL « 0x010500131700fe                                                     (7 bytes)
2021-05-04T22:53:26.322Z SERIAL » [ACK]                                                                   (0x06)
2021-05-04T22:53:26.324Z DRIVER « [REQ] [SendData]
                                    callback id:     23
                                    transmit status: OK
2021-05-04T22:53:26.336Z SERIAL « 0x010b00040007056c021600008a                                        (13 bytes)
2021-05-04T22:53:26.337Z SERIAL » [ACK]                                                                   (0x06)
2021-05-04T22:53:26.339Z DRIVER « [Node 007] [REQ] [ApplicationCommand]
                                  └─[SupervisionCCReport]
                                      session id:          22
                                      more updates follow: false
                                      status:              NoSupport
                                      duration:            [Duration: 0 seconds]
```

AlCalzone, 2021-05-05:
> status:              NoSupport

Very likely. It obviously supports Supervision, but does not support supervision-encapsulated Multilevel Switch CC, which is a no-no.

PR #2539 (2021-05-05): fix: disable supervision for ZL-PD-100
Closes #2538
closed 2021-05-05 by AlCalzone