zwave-js/zwave-js #2545: Ability to perform OTW firmware update Z-Wave 700 series USB sticks like the SLUSB001A

Hedda, 2021-05-05:
Please consider adding support to perform OTW firmware update the SLUSB001A (SLUSB7000A) Z-Wave 700 series UZB-7 USB stick.

https://www.silabs.com/documents/public/data-sheets/DSH14483.pdf

This same dongle is the official Z-Wave 700 reference design from Silicon Labs Z-Wave 700 Wireless Starter Kit (SLWSTK6050A):

https://www.silabs.com/development-tools/wireless/z-wave/z-wave-700-starter-kit

https://www.silabs.com/development-tools/wireless/z-wave/z-wave-long-range-700-starter-kit

Surprisingly the SLUSB001A / SLUSB7000A UZB-7 looks like the most affordable Z-Wave 700 USB adapters readily available today:

https://www.google.com/search?q=SLUSB001A

PS: This is related to the questions about firmware posted in the discussion section -> https://github.com/zwave-js/node-zwave-js/discussions/2544

PPS: This is a follow-up feature request to https://github.com/zwave-js/node-zwave-js/issues/1357 which was about adding support for the SLUSB001A (SLUSB7000A) UZB-7 USB stick.

Hedda, 2021-05-05:
As long as the standard Silabs Gecko bootloader is working it should be also able to upgrade firmware via XMODEM binary transfer

https://silabs-prod.adobecqms.net/content/usergenerated/asi/cloud/content/siliconlabs/en/community/wireless/z-wave/knowledge-base/jcr:content/content/primary/blog/z-wave_700_programm-4Hx0.social.0.10.html

As such a workaround could be to utilize Elelabs EZSP Firmware Update Utility and/or walthowd firmware updater image for it:

https://github.com/Elelabs/elelabs-zigbee-ezsp-utility

https://github.com/walthowd/husbzb-firmware

PS: Like other Z-Wave 700 sticks it's based on EFR32ZG14 Zen Gecko Z-Wave and require CP210x USB to UART Bridge VCP Drivers.

https://www.silabs.com/documents/public/schematic-files/PHD14501.zip

blhoward2, 2021-05-06:
One issue is that the firmware is bundled with Simplicity Studio and it is a copyright violation to distribute it on our own. If OTW was supported, from where would you get the file?

stephenl03, 2021-05-07:
Being able to download the software and then just point to a bin file (or whatever the extension is) to update is much easier than installing additional software and having to run it. This is especially true if you are running zwavejs in a headless environment on a non-Ubuntu host.

blhoward2, 2021-05-07:
It doesn't work that way though. You have to run Simplicity Studio to tell it to install the additional tools for zwave to have it install the next 2-3 GB including the zwave controller. At least I think that is the point at which it is installing the file. The whole thing is a series of enormous downloads.

blhoward2, 2021-05-07:
FTR, I just emailed Zooz and Aeotec to see if they intend to distribute firmware files directly for their sticks either now or int he future.

Hedda, 2021-05-07:
> FTR, I just emailed Zooz and Aeotec to see if they intend to distribute firmware files directly for their sticks either now or in the future.

I guess the question here is also about if the Z-Wave JS project/community should even bother to ask for permission before hosting and/or distributing commercial companies self-contained copyrighted OTW firmware images in their unmodified binary blob format. 

Distributing unmodified copyrighted firmware binary blobs is usually fine as long as you simply do not ask for permission. You would be using these OTW firmware image files for their intended purpose and that should not hurt the company in any way.

It is probably better not to ask and just do, wait and hope that they will not send your cease and desist letter. I am guessing that the chances are much higher than you are more likely to get a no as the answer if you ask in advance than them stopping you later if you do not ask at all in the first place. 

Not sure if fully legal or not but from what I can tell hardware manufacturers normally do not actively chase or sue open-source projects for simply hosting and/or distributing their companies self-contained binary blobs such as firmware image files or driver binaries as long as they are not modified in any way or used for anything other then their intended purposes.

For reference, see examples https://github.com/Koenkk/Z-Stack-firmware/ and https://github.com/Koenkk/zigbee-OTA

Maybe not quite covered by "Fair use" or "Fair dealing" 

https://en.wikipedia.org/wiki/Fair_use

https://en.wikipedia.org/wiki/Fair_dealing

Hedda, 2021-05-07:
> Rather than try to use one of the Python implementations linked, there is at least one XMODEM module for Node.js. https://github.com/exsilium/xmodem.js/

FYI, one of the benefits of the firmware updater Docker image by walthowd is that it contains scripts to probe/scan to detect which exact adapter it is and if the adapter supports it can automatically restart it into bootloader-mode so it can upgrade "automagically"

https://github.com/walthowd/husbzb-firmware

But yes, for Z-Wave JS purpose it would probably be better to just borrow inspiration from that concept for a new implementaion.

blhoward2, 2021-05-07:
We've already decided we are not interested in hosting firmware images. It's a copyright violation and we want to work with manufacturers. SiLabs has been supportive, as has Aeotec and Zooz.

blhoward2, 2021-05-07:
I've confirmed that Zooz doesn't intend to issue firmware updates for its sticks. Aeotec has no present plans to do so.

jspanitz, 2021-05-10:
> I've confirmed that Zooz doesn't intend to issue firmware updates for their sticks.

Well that sucks - I hope if enough of us ask they will change their mind.

Hedda, 2021-09-22:
https://siliconlabs.force.com/s/article/z-wave-700-programming-uzb7-controller-stick?s1oid=00DA0000000L2

_The SerialAPI command 0x27 FUNC_ID_AUTO_PROGRAMMING will drop the 700 series into the bootloader._

_Send the binary string below._

_01h, 03h, 00h, **27h**, DBh_ _will drop into the bootloader as described in the 500 series INS11681 section 5.2_

https://cn.silabs.com/documents/public/user-guides/INS11681-Instruction-500-Series-Z-Wave-Chip-Programming-Mode.pdf

Aeotec Z-Stick 7 now has a 'how-to' upgrade firmware guide with download links for the GBL firmware files for OTW:

https://help.aeotec.com/support/solutions/articles/6000252296-update-z-stick-7-sdk-7-15

jspanitz, 2021-09-22:
> Aeotec now has a 'how-to' upgrade firmware guide with download links for the Aeotec Z-Stick 7 GBL firmware files for OTW:
> 
> https://help.aeotec.com/support/solutions/articles/6000252296-update-z-stick-7-sdk-7-15

Anyone want to brick their Zooz?

Hedda, 2021-09-22:
> > Aeotec now has a 'how-to' upgrade firmware guide with download links for the Aeotec Z-Stick 7 GBL firmware files for OTW:
> > https://help.aeotec.com/support/solutions/articles/6000252296-update-z-stick-7-sdk-7-15
> 
> Anyone want to brick their Zooz?

I do not have a Zooz USB 700 Series Stick, but know that flashing that EFR32ZG14 firmware for US Frequency should not brick it.

PS: Check out additional flashing instructions for OTW Firmware Upgrade via Simplicity Studio v5 Z-Wave Tool in https://github.com/zwave-js/node-zwave-js/discussions/2544 and https://github.com/zwave-js/node-zwave-js/discussions/2549

blhoward2, 2021-09-25:
Confirmed upgrade working in Linux. Credit to cpw:

1. installed minicom
2. figured out where device was (/dev/ttyUSB0 for me)
3. unplugged/plugged in device
4. in a shell: stty 115200 cs8 -cstopb -F /dev/ttyUSB0
5. then in same shell: printf '\x01\x03\x00\x27\xDB' >/dev/ttyUSB0
6. configure minicom: minicom -s important: turn hw flow control OFF here, set baud etc
7. run minicom: minicom -D /dev/ttyUSB0
8. press enter a bunch of  times, you'll see menu as demoed. (may need to replug stick to get this to happen)
9. select 1 in menu to flash. in minicom, use ctrl shift S, select xmodem, and specify path to file. Upload should proceed (it'll take a minute)
10. press 2 once upload is complete in minicom to "run" the new firmware.

kpine, 2021-12-03:
FYI, just picked up a Zooz ZST10 700. It came installed with SDK version 7.15.01. Either the stick I received does **not** support controller OTW firmware updates, or something is different than the others. I attempted to install multiple newer versions of FW and all fail.

[image]

From the [OTW guide](https://siliconlabs.force.com/s/article/z-wave-700-otw-of-controller):

>  If you may get error 0x50 or 0x18 during ZGM130S or EFR32ZG14 controller firmware update, please make sure the bootloader and encryption keys have already been flashed into the device via SWD interface before you perform the OTW/OTA update. UZB-7 does not expose the SWD interface, it is required to open the plastic box and connect its SWD interface to programmer in this situation. Besides, ensure the gbl file being load contains an increased firmware version comparing the one that the device is currently running.

PC-Controller, minicom, and zw_programmer all fail. The bootloader is installed, and the fw version is newer. I'm wondering if they are using different encryption keys than are used for the Silicon Labs GBL files, or the keys are not installed?

```
root@dev:/app/bin# ./zw_programmer -s /dev/ttyUSB0 -p ZW_SerialAPI_Controller_7.16.3_297_EFR32ZG14_REGION_US.gbl
Using serial device /dev/ttyUSB0
Connected to Serial device: OK
Serial version: 9, Chip type: 7, Chip version: 0, SDK: 7.15.01, SDK Build no: 178 SDK git hash: 30313233343536373839414243444546
Chip library: 7, ZW protocol: 7.15, NVM: bridge7.15
nvm_id: bridge7.15
Auto Programming enabled

Gecko Bootloader v1.12.01
1. upload gbl
2. run
3. ebl info
BL >1

begin uploadnumbytes: 159488, oldnumbytes: 159452
Enter C(67) ACK(1) NAK (2)
Received: d
unknown char received d
Enter C(67) ACK(1) NAK (2)
Received: a

unknown char received a
Enter C(67) ACK(1) NAK (2)
Received: 0
unknown char received 0
Enter C(67) ACK(1) NAK (2)
Received: 43 C
Received C
2106708837 Sending Firmware file to Gecko module
|send_data 0 159488 1                                |
Sent 0 bytes, remaining:159488 bytes
>sent offset :0
new offset :128
sending 133 bytes: 0x01 0x01 0xfe
Enter C(67) ACK(1) NAK (2)
Received: 18

2106708869 Received CANError in transmission
Closing Serial connection
```

Hedda, 2021-12-03:
> FYI, just picked up a Zooz ZST10 700. It came installed with SDK version 7.15.01. Either the stick I received does **not** support controller OTW firmware updates, or something is different than the others. I attempted to install multiple newer versions of FW and all fail.

@kpine If you have a Windows OS machine then maybe try the instructions I posted in https://github.com/zwave-js/node-zwave-js/discussions/2544 for Silicon Labs UZB-7 USB Stick:

_Originally posted by @Hedda in https://github.com/zwave-js/node-zwave-js/discussions/2544#discussioncomment-1323267_

Note! Users should first update firmware to a later GA (General Availability) version of firmware before using UZB-7 as a commercial product.

Tip! You can perform local firmware upgrade using OTW (Over The Wire) on UZB7 USB stick using Z-Wave 700 Serial API firmware GBL XModem file via the "OTW Firmware Upgrade" option in Silabs "Z-Wave PC Controller" tool that gets installed with Simplicity Studio.

### Quick Start-Up Guide to flash UZB-7 USB stick with latest firmware

Here is a quick start-up guide on how to perform OTW Firmware Upgrade on UZB-7 using "Z-Wave PC Controller" tool in Simplicity Studio.

Prerequisites: 

To actually get/download the latest firmware file for Z-Wave 700 USB stick you need to register for a free Silicon Labs Community account at https://community.silabs.com/SL_CommunitiesSelfReg and install Simplicity Studio v5 for your operating system, start it and login to install + update the Z-Wave technology type SDK package inside Simplicity Studio. See https://www.silabs.com/developers/simplicity-studio

1. Connect/plug-in the Z-Wave 700 Controller USB stick to your computer.
2. Z-Wave 700 Controller USB sticks will normally use a Silabs CP2102N USB-to-serial converter chip for the MCU SoC so if needed by your operating system then install the Virtual COM Port (VCP) drivers and CP210x VCP Drivers from Silicon Labs. The latest drivers are available at https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers (for Windows 10 download and install the "CP210x Universal Windows Driver" package).
3. If using Windows 10 then be sure to disable Energy Saving Mode on the virtual com port so the dongle does not go into sleep: https://silabs-prod.adobecqms.net/community/wireless/z-wave/knowledge-base.entry.html/2020/06/04/uzb7_not_accessibleonwindows10-0Dw1
4. Start and run the "Z-Wave PC Controller" from Simplicity Studio from under the Tools menu.
5. Click on the “Setting-wheel” in the Right-Top corner of the Z-Wave PC Controller tool and select “Silicon Labs CP210x USB
to UART Bridge”. Click OK.
6. Click on "NVM Backup/Restore" select a folder and file path to backup files to in order to keep your a backup network stored. Make sure to name it with something like "ZWave_USB_Stick_Backup_date_and_time". Click on Backup.
7. Click the "OTW Firmware Upgrade" box and select the correct GBL file, (taking note that if you do not select a firmware with a version that is newer than your existing version you will get an error an upgrade will fail as downgrade or reflashing same version is not possible here). Firmware upgrade should be started and completed automatically.

FYI, if use default directories for Simplicity Studio + the Z-Wave technology type package GBL firmware binary files should after install be in:

C:\SiliconLabs\SimplicityStudio\v5\developer\sdks\gecko_sdk_suite\v3.2\protocol\z-wave\Apps\bin\gbl

Silabs UZB-7 USB Stick uses an EFR32ZG14 SoC so you need to select the latest ZW SerialAPI Controller firmware for it and your RF region.

So for example, currently, currently the latest firmware file (Version 7.16.2.0 / Version 7.16 Build 267) for the UZB7 USB sticks for Europe (RF region) is:

"ZW_SerialAPI_Controller_7.16.2_267_EFR32ZG14_REGION_EU.gbl"

Respectively, currently, the latest firmware file (Version 7.16.2.0 / Version 7.16 Build 267) for the UZB7 USB sticks for the United States / North America (RF region) is:

"ZW_SerialAPI_Controller_7.16.2_267_EFR32ZG14_REGION_US.gbl"

PS: Move information is available in INS13114 Z-Wave PC based Controller v5 User Guide from Silicon Labs:

https://www.silabs.com/documents/public/user-guides/INS13114-Z-Wave-PC-Based-Controller-v5-User-Guide.pdf

kpine, 2021-12-03:
I don't see anything in your reply that isn't something I've already tried. Is there some **new information** there?

blhoward2, 2021-12-03:
I'll just ask Zooz.

kpine, 2021-12-03:
And yes, the region matches the firmware files I tried, US. For good measure, I applied the region setting even though it was already listed as US.

[image]

jspanitz, 2021-12-03:
> I'll just ask Zooz.

Hope you get a better answer than I did - I asked when the stick was still pre-release and they stated then they won't support upgrades to it.

blhoward2, 2021-12-03:
They won’t support upgrades to it but that’s not what I asked them. Supporting upgrades is different than affirmatively blocking upgrades. In any event, we’ll see what our contact has to say.

kars85, 2021-12-04:
Plugged my Zooz ZST10 into my Win10 laptop and gave Simplicity Studio OTW a shot. Same error as @kpine - this was at ZW_SerialAPI_Controller_7.16.3_297_EFR32ZG23_REGION_US.gbl

Nothing of substance to add here, really. I have had my stick for maybe 6 weeks preinstalled with 7.15.

blhoward2, 2021-12-04:
Can you all try a different version? I'm curious if it's just an issue with 7.16.3.

kpine, 2021-12-04:
I tried several, with the same failure.

- ZW_SerialAPI_Controller_7.15.4_256_EFR32ZG14_REGION_US.gbl
- ZW_SerialAPI_Controller_7.16.2_267_EFR32ZG14_REGION_US.gbl
- ZW_SerialAPI_Controller_7.16.3_297_EFR32ZG14_REGION_US.gbl

kars85, 2021-12-04:
Here is a picture of my PCB. [image]

kpine, 2021-12-04:
Wow, thanks for opening it. It looks like it's using a different Z-Wave SOC than the UZB7? [ZGM130S](https://www.silabs.com/documents/public/data-sheets/zgm130s-datasheet.pdf) vs. [EFR32ZG14](https://www.silabs.com/documents/public/data-sheets/efr32zg14-datasheet.pdf). Maybe the file we want is `ZW_SerialAPI_Controller_7.16.3_297_ZGM130S_REGION_US.gbl`?

blhoward2, 2021-12-04:
Looks like it. Wonder why they swapped the chip.

kpine, 2021-12-04:
`ZW_SerialAPI_Controller_7.16.3_297_ZGM130S_REGION_US.gbl` worked for me. Nice work @kars85 !

```
root@dev:/app/bin# ./zw_programmer -s /dev/ttyUSB0 -t
Using serial device /dev/ttyUSB0
Connected to Serial device: OK
Serial version: 9, Chip type: 7, Chip version: 0, SDK: 7.16.03, SDK Build no: 41 SDK git hash: 30303030303030303030303030303030
Chip library: 7, ZW protocol: 7.16, NVM: bridge7.16
nvm_id: bridge7.16
Closing Serial connection
```

blhoward2, 2021-12-04:
What program did you use for this? It looks easier than our method with minicom.

kpine, 2021-12-04:
It's the `zw_programmer` utility included with the Z/IP Gateway. It's easy to use, but it's difficult to build. It was listed as in option in the OTW guide I linked above.

blhoward2, 2021-12-04:
What license is the source under?

kpine, 2021-12-04:
I believe it's proprietary.
```
❯ cat LICENSE
/* © 2014 Silicon Laboratories Inc. */
```

kars85, 2021-12-04:
No problem @kpine - here's to good luck moving forward with the project.

jcam, 2021-12-05:
Just built zw_programmer on mac, and used it to update a zooz 700 stick with ZW_SerialAPI_Controller_7.16.3_297_ZGM130S_REGION_US.gbl, it worked the same as @kpine. Thanks!

I wonder whether it'd be worth it to ask SiLabs if they'd be willing to MIT license the zw_programmer source. It is included in the Z/IP SDK, but they wouldn't have to open the whole SDK...

AlCalzone, 2021-12-05:
AFAIK there is a way to do it via the serial API that zwave-js uses - as long as there is a bootloader already installed on the device. I've not yet figured out which commands though.

kpine, 2021-12-05:
For 700-series, the `zw_programmer` program update is exactly the same as using minicom. It just enables auto-programming mode and interacts with the bootloader and sends the file with xmodem. The 500-series flashes the NVR (NVM?) using functionality provided by the Z/IP gateway library. That might be tough to get right.

blhoward2, 2021-12-05:
That Python script you sent me the other day does the 500 though, so those commands are know.

kpine, 2021-12-05:
Maybe... that method (OTW) only works for certain devices (according to the comments).

blhoward2, 2021-12-05:
I think that's because they're distributing the FW as part of the script. The method in zw_programmer only differentiates 500 vs 700. And it doesn't look that complicated.

blhoward2, 2021-12-05:
At @AlCalzone request, adding this reference for later:

https://github.com/drzwave/ZWaveOTW

k3vmcd, 2022-01-01:
It may now be possible to implement this feature since the firmware is recently published to GitHub for direct download? As of Gecko SDK (GSDK) Suite 4.0 the firmware files are now at: https://github.com/SiliconLabs/gecko_sdk/tree/gsdk_4.0/protocol/z-wave/Apps/bin/gbl

The changes are described in this Knowledge Article: https://community.silabs.com/s/article/Notes-on-the-recent-Simplicity-Studio-version-5-3-0-0-release?language=en_US

blhoward2, 2022-01-01:
The 700 series firmware has been available from Aeotec for some time. And you could always download Simplicity Studio to get it. The OTW Upgrade feature itself is on our roadmap.

Hedda, 2022-01-07:
> The 700 series firmware has been available from Aeotec for some time, though this will be more up to date. 

For reference, Aeotec looks to only provide direct download links for their latest tested GBL firmware files for OTW in how-to guide:

https://help.aeotec.com/support/solutions/articles/6000252296-update-z-stick-7-with-windows

Aeotec as such does not look to have a repository with revision control so could download older firmware if need to downgrade.

blhoward2, 2022-01-07:
The stick won't let you downgrade at all. The bootloader will only accept a new fw that is a version bump. You can't even redo the current version.

alexruffell, 2022-01-24:
> The stick won't let you downgrade at all. The bootloader will only accept a new fw that is a version bump. You can't even redo the current version.

I've read somewhere something about the _255 version to enable getting around version limitations. What I read seemed to be written backwards as it said the version number was coded to 255 to enable a downgrade... maybe I read incorrectly.

Anyhow I just updated my Aeotec and Zooz 700 series sticks to 7.17 using the appropriate files in the Simplicity Studio SDK folder. To upgrade I used OTW in Zwave PC Controller.

blhoward2, 2022-01-24:
Could you provide a link to what you’re talking about? I haven’t heard that and it would be helpful to fake it into a downgrade.

kpine, 2022-01-24:
@blhoward2 https://community.silabs.com/s/question/0D58Y00008C5A5PSAV/difference-between-firmware-files?language=en_US

> The _v255 just has the firmware version set to 255, this is intended to be used for OTA/OTW testing only, as it is required to load a gbl file having a greater firmware version when performing firmware upgrade for a ZGM130S or EFR32ZG14 device.

blhoward2, 2022-01-24:
It’s not going off the file name though. I think that means they made the version .255 when they built the file so it was a higher version.

alexruffell, 2022-01-24:
@blhoward2 Yes, there is a v255 file. See here:

https://github.com/SiliconLabs/gecko_sdk/tree/gsdk_4.0/protocol/z-wave/Apps/bin/gbl

blhoward2, 2022-01-24:
Right, but that just lets you reapply the same version again (a second time) in order to test OTA upgrades. It doesn’t let you downgrade.

alexruffell, 2022-01-24:
@blhoward2 I am sure you are right, but I thought I read otherwise. If you can't downgrade, what happens when you install the v255 version? Won't that prevent you from installing the next release?

blhoward2, 2022-01-24:
I think it's probably 7.17.255. So 7.18 would be higher.

AlCalzone, 2022-01-24:
> For some reason the folder is limited to 1000 files. I thought I was able to view the files before

you'll likely have to clone the repo and view them locally

blhoward2, 2022-01-24:
I’m happy for someone to prove me wrong, I’m just not sure how it would work otherwise. Or what the point of the restriction would be if they were going to distribute a file also that broke it.

kpine, 2022-01-24:
Sorry, I didn't think the link I posted needed any extra commentary. Downgrades are not supported obviously, based on that link. The _v255 file is strictly for developers or QA.

bmcdonnell158, 2022-01-29:
@kpine - any further progress?  I am having the same issue trying to upgrade a zooz 700 stick from 7.15 to 7.17.1.344.  I can get into the bootloader - but no file transfer... xmodem times out.  bootloader v1.12.01.  All other methods (simplicity studio, pc_controller) all fail as well.

kpine, 2022-01-29:
Sorry, I don't have any more suggestions. I had no issue updating my UZB7 (using `zw_programmer`), and my Zooz using PC Controller.

vezinpi, 2022-02-08:
@kpine @bmcdonnell158 

Hi, Just updated my Zooz 700 from 7.15 to 7.17.1 using Simplicity studio OTW.

I am not sure that your are aware but the Zooz 700 stick is using a different chipset than the other stick like the Aeotec. Aeotec is using the  EFR32ZG14 and the zooz is using the ZGM130S. Therefore the file for Zooz 700 stick for the us is:

ZW_SerialAPI_Controller_7_17_1_344_ZGM130S_REGION_US.gbl

The documentation out there is good for the general upgrade process but is really vague, when it comes to selecting the proper file based on the chipset the USB Stick is using.

Hope this help

kpine, 2022-02-08:
@vezinpi Thanks, but we figured that out over [two months ago](https://github.com/zwave-js/node-zwave-js/issues/2545#issuecomment-986104258). 😉 I also have it documented in [my upgrade guide](https://github.com/kpine/zwave-js-server-docker/wiki/700-series-Controller-Firmware-Updates-(Linux)#firmware-files).

lymanepp, 2022-02-24:
From Zooz support regarding errors when upgrading firmware on ZST10 700 in Simplicity Studio.

> We found that the previous file provided by Silicon Labs was missing some components and for some users, resulted in the error you're seeing. We've since been provided with a different file and have created new instructions for the update - however, they couldn't be applied if the original file was already attempted.

> There are two options for a solution:

> * We can initiate warranty service for your ZST10 700. Since you're in the US, you can send the stick back to us and we can reset the stick with a special tool in-house (which unfortunately can't be done remotely).
> * We can wait for the 7.18 firmware to update the stick with that (since the same firmware cannot be applied unfortunately). 

> We currently don't know of a release date for the next firmware update though.

[28 more replies]

PR #5321 (2023-01-12): feat: support OTW upgrade of 700+ series controllers
fixes: #2545
closed 2023-01-12 by AlCalzone