zwave-js/zwave-js #2557: Controller (and repeater?) nodes marked as dead

GSV3MiaC, 2021-05-05:
A developer told me to come here.

My zwave network is showing dead nodes for the backup controller (node 22) and the repeater. I was advised to open the zwave to mqtt control panel and try to ping the dead node, I thine tried to refresh the mode info (requery then node) that worked, as you can see, but ping still fails.

Heal doesn't work, since it doesn't even try (node being dead) but of course since we can get to the node a heal is pretty pointless.

2021-05-05 16:06:10.317 INFO ZWAVE: Calling api refreshInfo with args: [ 22, [length]: 1 ]
2021-05-05T17:06:10.366Z CNTRLR   [Node 022] Beginning interview - last completed stage: None
2021-05-05T17:06:10.370Z CNTRLR   [Node 022] new node, doing a full interview...
2021-05-05 16:06:10.372 INFO ZWAVE: Node 22: interview started
2021-05-05T17:06:10.377Z CNTRLR » [Node 022] querying protocol info...
2021-05-05 16:06:10.400 INFO ZWAVE: Success zwave api call refreshInfo { success: true, message: 'Success zwave api call', result: undefined }
2021-05-05T17:06:10.432Z CNTRLR « [Node 022] received response for protocol info:
                                  basic device class:    Static Controller
                                  generic device class:  Static Controller
                                  specific device class: PC Controller
                                  node type:             Controller
                                  is always listening:   true
                                  is frequent listening: false
                                  can route messages:    false
                                  supports security:     false
                                  supports beaming:      true
                                  maximum data rate:     100000 kbps
                                  protocol version:      3
2021-05-05 16:06:10.437 INFO ZWAVE: Node 22: interview stage PROTOCOLINFO completed
2021-05-05T17:06:10.441Z CNTRLR   [Node 022] Interview stage completed: ProtocolInfo
2021-05-05T17:06:10.443Z CNTRLR » [Node 022] pinging the node...
2021-05-05 16:06:16.294 INFO ZWAVE: Node 19: metadata updated: 49-0-Air temperature
2021-05-05 16:06:16.305 INFO ZWAVE: Node 19: value updated: 49-0-Air temperature 17.6 => 17.5
2021-05-05 16:06:17.947 INFO ZWAVE: Node 25: metadata updated: 49-0-Air temperature
2021-05-05 16:06:17.951 INFO ZWAVE: Node 25: value updated: 49-0-Air temperature 23.5 => 23.4
2021-05-05 16:06:18.976 INFO ZWAVE: Node 25: metadata updated: 49-0-Humidity
2021-05-05 16:06:18.982 INFO ZWAVE: Node 25: value updated: 49-0-Humidity 58 => 58
2021-05-05 16:06:19.463 INFO ZWAVE: Node 19: metadata updated: 49-0-Air temperature
2021-05-05 16:06:19.469 INFO ZWAVE: Node 19: value updated: 49-0-Air temperature 17.5 => 17.5
2021-05-05 16:06:19.638 INFO ZWAVE: Node 19: metadata updated: 49-0-Humidity
2021-05-05 16:06:19.645 INFO ZWAVE: Node 19: value updated: 49-0-Humidity 47 => 47
2021-05-05T17:06:20.564Z CNTRLR   [Node 022] Node 22 did not respond after 1 attempts, it is presumed dead
2021-05-05T17:06:20.570Z CNTRLR   [Node 022] ping failed: Failed to send the command after 1 attempts (Status No
                                  Ack)
2021-05-05T17:06:20.572Z CNTRLR » [Node 022] querying node info...
2021-05-05T17:06:20.574Z CNTRLR » [Node 022] pinging the node...
2021-05-05 16:06:25.910 INFO ZWAVE: Node 19: metadata updated: 49-0-Illuminance
2021-05-05 16:06:25.914 INFO ZWAVE: Node 19: value updated: 49-0-Illuminance 1707 => 1770
2021-05-05T17:06:26.508Z CNTRLR   [Node 022] Node 22 did not respond after 1 attempts, it is presumed dead
2021-05-05T17:06:26.515Z CNTRLR   [Node 022] ping failed: Failed to send the command after 1 attempts (Status No
                                  Ack)
2021-05-05T17:06:26.517Z CNTRLR   [Node 022] Interview attempt (1/5) failed, node is dead.
2021-05-05 16:06:26.519 ERROR ZWAVE: Interview of node 22 has failed: The node is dead

Here is the heal attempt:
Tried but it doesn;t get far:
2021-05-05 16:18:48.646 INFO ZWAVE: Calling api healNode with args: [ 22, [length]: 1 ]
2021-05-05T17:18:48.650Z CNTRLR   [Node 022] Skipping heal because the node is not responding.
2021-05-05 16:18:48.654 INFO ZWAVE: Success zwave api call healNode { success: true, message: 'Success zwave api call', result: false }

I have an aeon extender which exhibits same problem, but I have not logged that yet. Seems to me these devices are not responding to ping at all.

AlCalzone, 2021-05-05:
The message you're seeing is a response from the stick which reads the info from its memory. Can you bring the node closer to the stick?

GSV3MiaC, 2021-05-05:
Which message is from the stick? The config info, or the ping failure? 

The node is part of the extended network, so moving it is an issue. My extender (zw117) has the same issue. Shows as dead but seems to be working fine. 

Dead node is not reported by ozw 1.4 or 1.6 under domoticz, although 1.6 seems to have trouble deciding what sort of devices they are. If they were not working I think half my network would be unreachable. 

⁣Rgds, Dave Storey​

AlCalzone, 2021-05-05:
> My extender (zw117) has the same issue. Shows as dead but seems to be working fine.

What makes you think the node is working fine? The log above certainly doesn't.

> 2021-05-05T17:06:10.432Z CNTRLR « [Node 022] received response for protocol info:

Is a response from the controller. This is the very first step of a node interview and does not communicate with the node. I don't know what you mean by config info.

> Dead node is not reported by ozw 1.4 or 1.6 under domoticz

I'm not sure how those drivers treat missing ACKs from the node. I keep reading "it worked in XYZ" in several issues,  but changing the software on the host can't really change anything about the RF connection between the stick and the device. It's likely that you didn't notice because the software didn't tell you.
zwave-js considers a node as dead if it did not acknowledge the receipt of a command 3 times (once for pings) until a message from it is received again. A dumbed down version of a typical communication where a node is dead looks like this:
```
zwave-js: Hey stick, send this 
Stick: 🔊
Stick: I did
...
Stick: hey zwave-js, the node didn't acknowledge the message
```

blhoward2, 2021-05-06:
Someone else said their Aeotec Gen 5 stick was also dead when included as a secondary controller. Those do respond to pings, right?

GSV3MiaC, 2021-05-06:
Apparently not, since i took the controller back to the PC and tried the aeotec software on it, and it works fine. The repeater was dead as far as i can tell though. Presumably aeotec could tell us? 

⁣Rgds, Dave Storey​

AlCalzone, 2021-05-06:
> i took the controller back to the PC and tried the aeotec software on it, and it works fine

Let me get this straight: Is Node 22 portable / battery powered? Because your log above says it is known to the current primary as "always listening", i.e. permanently powered.

GSV3MiaC, 2021-05-06:
It is another zstick5 controller, which was plugged into a USB power adapter stuck in the wall halfway along the house. So it was always listening, or should have been. 

⁣Rgds, Dave Storey​

AlCalzone, 2021-05-24:
Not sure if there's anything we can do here. If the device does not ACK commands, we can't really communicate with it.

closed 2021-05-28 by zwave-js-assistant[bot]