zwave-js/zwave-js #2558: [bug] Investigate sensor mapping issue with zooz zse11

blhoward2, 2021-05-05:
Reported from Zooz and @paulcusick3: Seeing that motion is being received as the tamper sensor, not the motion sensor.  Am I alone?

blhoward2, 2021-05-05:
@paulcusick3 Please post a Silly level log of the interview and motion being tripped, and an export of the node.

paulcusick3, 2021-05-05:
Log here - didn't see u opened issue.  node 46. log set to debug.  let me know if you need verbose log.

[zwavejs_log.txt](https://github.com/zwave-js/node-zwave-js/files/6430633/zwavejs_log.txt)

paulcusick3, 2021-05-05:
silly log.  node 046

[silly_log_zwavejs.txt](https://github.com/zwave-js/node-zwave-js/files/6430652/silly_log_zwavejs.txt)

paulcusick3, 2021-05-05:
Interview log
[zwavejs_interview.txt](https://github.com/zwave-js/node-zwave-js/files/6430669/zwavejs_interview.txt)

paulcusick3, 2021-05-05:
not sure how to export the node.....

blhoward2, 2021-05-05:
That’s it, thanks I’ll look later tonight.

blhoward2, 2021-05-06:
It looks like they send the wrong notification type:

```
2021-05-05T22:36:46.030Z DRIVER « [Node 046] [REQ] [ApplicationCommand]
                                  └─[NotificationCCReport]
                                      notification type:   7
                                      notification status: 255
                                      notification event:  0x00
                                      event parameters:    
2021-05-05T22:36:46.035Z CNTRLR   [Node 046] [~] [Notification] Home Security[Cover status]: 3 => 0 [Endpoint 0]
2021-05-05T22:36:46.299Z SERIAL « 0x01100004002e0a7105000000ff0700010341                              (18 bytes)
2021-05-05T22:36:46.300Z SERIAL » [ACK]                                                                   (0x06)
2021-05-05T22:36:46.301Z DRIVER « [Node 046] [REQ] [ApplicationCommand]
                                  └─[NotificationCCReport]
                                      notification type:   7
                                      notification status: 255
                                      notification event:  0x00
                                      event parameters:    
2021-05-05T22:36:46.303Z CNTRLR   [Node 046] [~] [Notification] Home Security[Cover status]: 0 => 0 [Endpoint 0]
2021-05-05T22:36:47.924Z SERIAL « 0x010500131a01f2                                                     (7 bytes)
2021-05-05T22:36:47.924Z SERIAL » [ACK]                                                                   (0x06)
2021-05-05T22:36:47.925Z DRIVER « [REQ] [SendData]
                                    callback id:     26
                                    transmit status: NoAck
```

blhoward2, 2021-05-06:
The one they send me appears to report correctly. Can you confirm your parameter settings?

```
21:35:11.406 DRIVER « [Node 006] [REQ] [BridgeApplicationCommand]
└─[NotificationCCReport]
notification type: Home Security
notification status: 255
notification state: Motion detection
21:35:11.415 CNTRLR [Node 006] [~] [Notification] Home Security[Motion sensor status] [Endpoint 0]
: 0 => 8
2021-05-05 21:35:11.417 INFO ZWAVE: Node 6: value updated: 113-0-Home Security-Motion sensor status 0 => 8
```

blhoward2, 2021-05-06:
Never mind, if you set Parameter 16 to enable the binary sensor I get both:

```
21:46:38.463 SERIAL » [ACK] (0x06)
21:46:38.477 DRIVER « [Node 006] [REQ] [BridgeApplicationCommand]
└─[NotificationCCReport]
notification type: 7
notification status: 255
notification event: 0x00
event parameters: 
21:46:38.483 CNTRLR [Node 006] [~] [Notification] Home Security[Cover status]: 3 => 0 [Endpoint 0]
2021-05-05 21:46:38.485 INFO ZWAVE: Node 6: value updated: 113-0-Home Security-Cover status 3 => 0
21:47:09.202 SERIAL « 0x010d00a8000106043003000c00e385 (15 bytes)
21:47:09.205 CNTRLR [Node 006] [Binary Sensor] Motion: metadata updated [Endpoint 0]
2021-05-05 21:47:09.213 INFO ZWAVE: Node 6: metadata updated: 48-0-Motion
21:47:09.217 CNTRLR [Node 006] [~] [Binary Sensor] Motion: true => false [Endpoint 0]
2021-05-05 21:47:09.219 INFO ZWAVE: Node 6: value updated: 48-0-Motion true => false
21:47:09.223 SERIAL » [ACK] (0x06)
21:47:09.229 DRIVER « [Node 006] [REQ] [BridgeApplicationCommand]
└─[BinarySensorCCReport]
type: Motion
value: false
```

blhoward2, 2021-05-06:
For now, setting parameter 16 to 0 fixes this and causes it to work fine using the notification sensor in HA. Until the device file comes across you can set that manually at the bottom of zwavejs2mqtt. Parameter 16, size 1, value 0.

paulcusick3, 2021-05-06:
Thank you.  Can I set the parameter via the service call zwave_js.set_config_parameter

blhoward2, 2021-05-06:
Not until the new device file is being run. Zjs2mqtt can set raw configuration parameters but the service is restricted to what is defined in the file.