zwave-js/zwave-js #2567: Zwave.JS - Everspring HAC01/Eminent EM6514 Wall Remotes not working

Flight777, 2021-05-07:
**Checklist:**

-   [ ] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
-   [ ] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
-   [ ] I have checked the troubleshooting section and my problem is **not** described there.
-   [ ] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

As per Home Assistant issue: https://github.com/home-assistant/core/issues/49061 we were referred to here.

Eminent EM6514 switches or better known as Everspring HAC01 are not giving out proper events/commands and only return a BasicCC log entry, despite the configuration showing available in the database: https://devices.zwave-js.io/

`2021-04-11T13:13:31.938Z CNTRLR [Node 019] treating BasicCC::Set as a report 2021-04-11T13:13:32.367Z CNTRLR [Node 019] treating BasicCC::Set as a report 2021-04-11T13:13:32.414Z CNTRLR [Node 019] treating BasicCC::Set as a report 2021-04-11T13:13:32.483Z CNTRLR [Node 019] treating BasicCC::Set as a report 2021-04-11T13:14:14.003Z CNTRLR [Node 019] treating BasicCC::Set as a report 2021-04-11T13:14:15.653Z CNTRLR [Node 019] treating BasicCC::Set as a report 2021-04-11T13:14:38.378Z CNTRLR [Node 019] treating BasicCC::Set as a report 2021-04-11T13:14:39.656Z CNTRLR [Node 019] treating BasicCC::Set as a report 2021-04-11T13:14:40.222Z CNTRLR [Node 019] treating BasicCC::Set as a report 2021-04-11T13:14:40.248Z CNTRLR [Node 019] treating BasicCC::Set as a report 2021-04-11T13:14:40.273Z CNTRLR [Node 019] treating BasicCC::Set as a report 2021-04-11T13:14:40.299Z CNTRLR [Node 019] treating BasicCC::Set as a report 2021-04-11T13:14:40.364Z CNTRLR [Node 019] treating BasicCC::Set as a report 2021-04-11T13:14:40.405Z CNTRLR [Node 019] treating BasicCC::Set as a report 2021-04-11T13:14:40.794Z CNTRLR [Node 019] treating BasicCC::Set as a report 2021-04-11T13:14:40.840Z CNTRLR [Node 019] treating BasicCC::Set as a report`

**Device information**

Which device(s) is/are affected (make/model)?  

Eminent EM6514 switches or better known as Everspring HAC01

What are the node IDs?

Node 009
Node 019

**Did you change anything?**

-   [ ] **Yes**: _(please describe)_
-   [**X**] **No**

**Did this use to work before?**

-   [ ] **Don't know**, this is a new device
-   [ ] **No**, it never worked anywhere
-   [**X**] **Yes**, in: Home Assistants old (deprecated) Zwave integration as well as in Homey before.

**How are you using `node-zwave-js`**

-   [**X**] `HomeAssistant` version 2021.5.0

**To Reproduce**

Add device and try the switching

**Additional context**

N/A

**Logfile:**

See attached
[zwave_js_dump (2).zip](https://github.com/zwave-js/node-zwave-js/files/6440845/zwave_js_dump.2.zip)

AlCalzone, 2021-05-07:
Please make a zwave-js log, loglevel `debug` or `silly` and attach it here as a file.

What do you mean by this?

> are not giving out proper events/commands and only return a BasicCC log entry

Flight777, 2021-05-07:
Thanks for the quick reply and apologies for confusion I might cause in my limited knowledge around this!

I've attached an excerpt of the log, search for Node 009 and Node 019

[zwavejs.log](https://github.com/zwave-js/node-zwave-js/files/6441136/zwavejs.log)

I have tried multiple time to re-interview/remove and add the device, always the same result.

What I mean by that, is that this should be a relatively simple on/off switch but is not recognised as such by Home Assistant and I'm not able to tell any scenes or value changes coming from Zwave.JS other than that line.

AlCalzone, 2021-05-07:
There are changes in the `Basic CC` `currentValue`, but I'm not sure if that is all the device offers:

> 2021-05-07T11:17:38.124Z CNTRLR   [Node 009] [~] [Basic] currentValue: 0 => 99                      [Endpoint 0]

Can you please re-interview the node and capture another log of that?

On a side note: Your node 14 is reporting its current power every second. That seems a bit excessive and causes a bunch of RF traffic. Might be worth to configure this differently

ehindriks, 2021-05-21:
Hi, I have the same issue with this device. Below a log of a re-interview (if this is what you mean with the log at least).

2021-05-21 10:53:44.235 INFO ZWAVE: Calling api refreshInfo with args: [ 4, [length]: 1 ]
2021-05-21 10:53:44.286 INFO ZWAVE: Node 4: interview started
2021-05-21 10:53:44.347 INFO ZWAVE: Success zwave api call refreshInfo { success: true, message: 'Success zwave api call', result: undefined }
2021-05-21 10:53:44.377 INFO ZWAVE: Node 4: interview stage PROTOCOLINFO completed
2021-05-21 10:53:44.457 INFO ZWAVE: Node 4: interview stage NODEINFO completed
2021-05-21 10:53:44.574 INFO ZWAVE: Node 4: value added: 114-0-manufacturerId => 96
2021-05-21 10:53:44.577 INFO ZWAVE: Node 4: value added: 114-0-productType => 16
2021-05-21 10:53:44.579 INFO ZWAVE: Node 4: value added: 114-0-productId => 1
2021-05-21 10:53:44.806 INFO ZWAVE: Node 4: value added: 134-0-libraryType => 6
2021-05-21 10:53:44.809 INFO ZWAVE: Node 4: value added: 134-0-protocolVersion => 3.42
2021-05-21 10:53:44.812 INFO ZWAVE: Node 4: value added: 134-0-firmwareVersions => 2.0
2021-05-21 10:53:45.380 INFO ZWAVE: Node 4: metadata updated: 112-0-1
2021-05-21 10:53:45.384 INFO ZWAVE: Node 4: metadata updated: 112-0-2
2021-05-21 10:53:45.535 INFO ZWAVE: Node 4: value added: 112-0-1 => 99
2021-05-21 10:53:45.590 INFO ZWAVE: Node 4: value added: 112-0-2 => 1
2021-05-21 10:53:45.640 INFO ZWAVE: Node 4: interview stage COMMANDCLASSES completed
2021-05-21 10:53:45.645 INFO ZWAVE: Node 4: interview stage OVERWRITECONFIG completed
2021-05-21 10:53:45.676 INFO ZWAVE: Node 4: interview stage NEIGHBORS completed
2021-05-21 10:53:45.690 INFO ZWAVE: Node 4: interview stage COMPLETE completed
2021-05-21 10:53:45.732 INFO ZWAVE: Node 4: interview COMPLETED, all values are updated
2021-05-21 10:53:45.768 INFO ZWAVE: Node 4: value added 4-134-0-libraryType => 6
2021-05-21 10:53:45.769 INFO ZWAVE: Node 4: value added 4-134-0-protocolVersion => 3.42
2021-05-21 10:53:45.770 INFO ZWAVE: Node 4: value added 4-134-0-firmwareVersions => 2.0
2021-05-21 10:53:45.772 INFO ZWAVE: Node 4: value added 4-114-0-manufacturerId => 96
2021-05-21 10:53:45.773 INFO ZWAVE: Node 4: value added 4-114-0-productType => 16
2021-05-21 10:53:45.774 INFO ZWAVE: Node 4: value added 4-114-0-productId => 1
2021-05-21 10:53:45.776 INFO ZWAVE: Node 4: value added 4-112-0-1 => 99
2021-05-21 10:53:45.777 INFO ZWAVE: Node 4: value added 4-112-0-2 => 1
2021-05-21 10:53:45.782 INFO ZWAVE: Node 4 ready: Everspring - HAC01 (In-Wall Remote Insert)

AlCalzone, 2021-05-21:
> Below a log of a re-interview (if this is what you mean with the log at least).

Yes, that is what I meant, but I need the driver log, not the one from zwavejs2mqtt.

ehindriks, 2021-05-21:
Can you please give me a hint on how to get it?

AlCalzone, 2021-05-21:
https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report?id=export-logs
The first one (zwave-js)

ehindriks, 2021-05-21:
Thanks! Found it.
[zwavejs2mqtt-store.zip](https://github.com/zwave-js/node-zwave-js/files/6521492/zwavejs2mqtt-store.zip)

AlCalzone, 2021-05-24:
@zwave-js-bot add compat flag to 0x0060:0x0010:0x0001
```json
enableBasicSetMapping: true
```

Flight777, 2021-05-31:
Apologies for my extremely late reply... I've been out of the running for a bit unfortunately.

I was happy to see progress, however, I'm not yet sure what to do with it. Updated ZwaveJS in Home Assistant, reinterviewed the devices but there still seems no way how I can capture the switch. It does not show up as entity, nor as any Home Assistant recognised ZwaveJS event. Any pointers what I'm doing wrong? :)

AlCalzone, 2021-05-31:
This should be coming in as a `"value event"` - I'm not sure how that can be accessed in HA. A driver log on loglevel "debug" would be good to verify.

ehindriks, 2021-06-10:
Could be me, but I also didn't notice any difference. I even excluded and re-included the device. I also checked the events in the developer tools of HA, but nothing pops up when I press the button, but the basic value changes in the zwave-js control page. I added a driver log below. Can you please have a look if you can find something in it @AlCalzone?

It is node 23. The log contains an interview and a button press.

[zwavejs_2021-06-10.log](https://github.com/zwave-js/node-zwave-js/files/6634166/zwavejs_2021-06-10.log)

AlCalzone, 2021-06-10:
> 2021-06-10T20:06:54.728Z CNTRLR   [Node 023] treating BasicCC::Set as a report

Nothing is being mapped, but I know why. The device doesn't support any CC we could map to. The NIF includes `Binary Switch`
```
2021-06-10T20:06:43.497Z CNTRLR « [Node 023] node info received
                                  supported CCs:
                                  · Binary Switch
                                  · Basic
                                  · Version
                                  · Manufacturer Specific
                                  · Association
                                  · Configuration
```
but then later tells us it doesn't support that CC:
```
2021-06-10T20:06:43.895Z DRIVER « [Node 023] [REQ] [ApplicationCommand]
                                  └─[VersionCCCommandClassReport]
                                      CC:      Binary Switch
                                      version: 0
2021-06-10T20:06:43.898Z CNTRLR   [Node 023]   does NOT support CC Binary Switch (0x25)
```
The manual also doesn't mention it:
[image]

If it actually supports `Binary Switch CC`, we can override what it tells us. This is described in https://zwave-js.github.io/zwavejs2mqtt/#/development/intro?id=testing-custom-devices-config, you can use this file to test:
[hac01.json.txt](https://github.com/zwave-js/node-zwave-js/files/6634375/hac01.json.txt) - make sure to rename it back to `.json`

ehindriks, 2021-06-11:
Thanks for looking into it. I will try your config file once I found out how to update it on HA,

ehindriks, 2021-06-23:
@AlCalzone I tried the attached hac01.json file and with this config I now have an entity in HA that I can use. Thanks for the update!

AlCalzone, 2021-06-23:
Thanks for the feedback --> https://github.com/zwave-js/node-zwave-js/pull/2903

Flight777, 2021-07-10:
Sorry to jump back in, was trying the new version just now. I indeed see an entity created in HA, however, it does not do anything (change state) when I press the switch on the wall.

This is what I see in the log, its noude 019 and 09 (also there are some reinterviews in between to see if that helped):

`2021-07-10T09:23:36.497Z SERIAL « 0x010b0004001305700602010192                                        (13 bytes)
2021-07-10T09:23:36.498Z CNTRLR   [Node 019] [+] [Configuration] 2: 1
2021-07-10T09:23:36.498Z SERIAL » [ACK]                                                                   (0x06)
2021-07-10T09:23:36.499Z DRIVER « [Node 019] [REQ] [ApplicationCommand]
                                  └─[ConfigurationCCReport]
                                      parameter #: 2
                                      value size:  1
                                      value:       1
2021-07-10T09:23:36.499Z CNTRLR « [Node 019] parameter #2 has value: 1
2021-07-10T09:23:36.500Z CNTRLR   [Node 019] [+] [Configuration] interviewComplete: true [Endpoint 0] [internal]
2021-07-10T09:23:36.507Z CNTRLR   [Node 019] Interviewing Binary Switch...
2021-07-10T09:23:36.507Z CNTRLR » [Node 019] querying Binary Switch state...
2021-07-10T09:23:36.510Z SERIAL » 0x0109001313022502254bbd                                            (11 bytes)
2021-07-10T09:23:36.510Z DRIVER » [Node 019] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      75
                                  └─[BinarySwitchCCGet]
2021-07-10T09:23:36.512Z SERIAL « [ACK]                                                                   (0x06)
2021-07-10T09:23:36.518Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2021-07-10T09:23:36.518Z SERIAL » [ACK]                                                                   (0x06)
2021-07-10T09:23:36.519Z DRIVER « [RES] [SendData]
                                    was sent: true
2021-07-10T09:23:36.534Z SERIAL « 0x010700134b000002a2                                                 (9 bytes)
2021-07-10T09:23:36.534Z SERIAL » [ACK]                                                                   (0x06)
2021-07-10T09:23:36.534Z DRIVER « [REQ] [SendData]
                                    callback id:     75
                                    transmit status: OK
2021-07-10T09:23:36.543Z SERIAL « 0x01090004001303250300c4                                            (11 bytes)
2021-07-10T09:23:36.543Z CNTRLR   [Node 019] [+] [Binary Switch] currentValue: false                [Endpoint 0]
2021-07-10T09:23:36.544Z CNTRLR   [Node 019] [+] [Binary Switch] targetValue: undefined             [Endpoint 0]
2021-07-10T09:23:36.544Z SERIAL » [ACK]                                                                   (0x06)
2021-07-10T09:23:36.544Z DRIVER « [Node 019] [REQ] [ApplicationCommand]
                                  └─[BinarySwitchCCReport]
                                      current value: false
2021-07-10T09:23:36.545Z CNTRLR « [Node 019] received Binary Switch state:
                                  current value:      false
2021-07-10T09:23:36.546Z CNTRLR   [Node 019] [+] [Binary Switch] interviewComplete: true [Endpoint 0] [internal]
2021-07-10T09:23:36.546Z CNTRLR   [Node 019] Interview stage completed: CommandClasses
2021-07-10T09:23:36.546Z CNTRLR   [Node 019] Interview stage completed: OverwriteConfig
2021-07-10T09:23:36.546Z CNTRLR » [Node 019] requesting node neighbors...
2021-07-10T09:23:36.548Z SERIAL » 0x01070080130000006b                                                 (9 bytes)
2021-07-10T09:23:36.549Z DRIVER » [REQ] [GetRoutingInfo]
                                    remove non-repeaters: false
                                    remove bad links:     false
2021-07-10T09:23:36.551Z SERIAL « [ACK]                                                                   (0x06)
2021-07-10T09:23:36.553Z SERIAL « 0x012001807f6103000000000000000000000000000000000000000000000000000 (34 bytes)
                                  043
2021-07-10T09:23:36.553Z SERIAL » [ACK]                                                                   (0x06)
2021-07-10T09:23:36.554Z DRIVER « [RES] [GetRoutingInfo]
                                    node ids: 1, 2, 3, 4, 5, 6, 7, 9, 14, 15, 17, 18
2021-07-10T09:23:36.555Z CNTRLR « [Node 019] node neighbors received: 1, 2, 3, 4, 5, 6, 7, 9, 14, 15, 17, 18
2021-07-10T09:23:36.555Z CNTRLR   [Node 019] Interview stage completed: Neighbors
2021-07-10T09:23:36.555Z CNTRLR   [Node 019] Interview completed
2021-07-10T09:23:36.556Z CNTRLR   [Node 019] The node is ready to be used
2021-07-10T09:30:25.813Z SERIAL » 0x010a001313032501ff254c44                                          (12 bytes)
2021-07-10T09:30:25.813Z DRIVER » [Node 019] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      76
                                  └─[BinarySwitchCCSet]
                                      target value: true
2021-07-10T09:30:25.816Z SERIAL « [ACK]                                                                   (0x06)
2021-07-10T09:30:25.822Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2021-07-10T09:30:25.822Z SERIAL » [ACK]                                                                   (0x06)
2021-07-10T09:30:25.823Z DRIVER « [RES] [SendData]
[... cut ...]
2021-07-10T09:30:47.941Z SERIAL « 0x01090004001303200163a0                                            (11 bytes)
2021-07-10T09:30:47.942Z SERIAL » [ACK]                                                                   (0x06)
2021-07-10T09:30:47.942Z DRIVER « [Node 019] [REQ] [ApplicationCommand]
                                  └─[BasicCCSet]
                                      target value: 99
2021-07-10T09:30:47.943Z CNTRLR   [Node 019] treating BasicCC::Set as a report
2021-07-10T09:30:47.944Z CNTRLR   [Node 019] [+] [Basic] currentValue: 99                           [Endpoint 0]`

AlCalzone, 2021-07-10:
@Flight777 doesn't look like it's being mapped to another CC. Can you share a log of a re-interview?

Flight777, 2021-07-10:
Here you go:

`Subscribed to Z-Wave JS Log Messages...
2021-07-10T09:57:40.077Z CNTRLR   [Node 019] Beginning interview - last completed stage: None
2021-07-10T09:57:40.079Z CNTRLR   [Node 019] new node, doing a full interview...
2021-07-10T09:57:40.080Z CNTRLR » [Node 019] querying protocol info...
2021-07-10T09:57:40.082Z SERIAL » 0x0104004113a9                                                       (6 bytes)
2021-07-10T09:57:40.083Z DRIVER » [REQ] [GetNodeProtocolInfo]
                                    payload: 0x13
2021-07-10T09:57:40.087Z SERIAL « [ACK]                                                                   (0x06)
2021-07-10T09:57:40.088Z SERIAL « 0x01090141d39c00041201ee                                            (11 bytes)
2021-07-10T09:57:40.089Z SERIAL » [ACK]                                                                   (0x06)
2021-07-10T09:57:40.089Z DRIVER « [RES] [GetNodeProtocolInfo]
                                    payload: 0xd39c00041201
2021-07-10T09:57:40.091Z CNTRLR « [Node 019] received response for protocol info:
                                  basic device class:    Routing Slave
                                  generic device class:  Remote Switch
                                  specific device class: Binary Remote Switch
                                  node type:             Routing End Node
                                  is always listening:   true
                                  is frequent listening: false
                                  can route messages:    true
                                  supports security:     false
                                  supports beaming:      true
                                  maximum data rate:     40000 kbps
                                  protocol version:      3
2021-07-10T09:57:40.091Z CNTRLR   [Node 019] Interview stage completed: ProtocolInfo
2021-07-10T09:57:40.091Z CNTRLR » [Node 019] querying node info...
2021-07-10T09:57:40.093Z SERIAL » 0x010400601388                                                       (6 bytes)
2021-07-10T09:57:40.093Z DRIVER » [Node 019] [REQ] [RequestNodeInfo]
                                    payload: 0x13
2021-07-10T09:57:40.096Z SERIAL « [ACK]                                                                   (0x06)
2021-07-10T09:57:40.102Z SERIAL « 0x01040160019b                                                       (6 bytes)
2021-07-10T09:57:40.103Z SERIAL » [ACK]                                                                   (0x06)
2021-07-10T09:57:40.103Z DRIVER « [RES] [RequestNodeInfo]
                                    payload: 0x01
2021-07-10T09:57:40.129Z SERIAL « 0x010f004984130904120125208672857034                                (17 bytes)
2021-07-10T09:57:40.130Z SERIAL » [ACK]                                                                   (0x06)
2021-07-10T09:57:40.131Z DRIVER « [Node 019] [REQ] [ApplicationUpdateRequest]
                                    payload: 0x841309041201252086728570
2021-07-10T09:57:40.133Z CNTRLR « [Node 019] node info received
                                  supported CCs:
                                  · Binary Switch
                                  · Basic
                                  · Version
                                  · Manufacturer Specific
                                  · Association
                                  · Configuration
                                  controlled CCs:
2021-07-10T09:57:40.133Z CNTRLR   [Node 019] Interview stage completed: NodeInfo
2021-07-10T09:57:40.134Z CNTRLR   [Node 019] Interviewing Manufacturer Specific...
2021-07-10T09:57:40.134Z CNTRLR » [Node 019] querying manufacturer information...
2021-07-10T09:57:40.137Z SERIAL » 0x0109001313027204255dfa                                            (11 bytes)
2021-07-10T09:57:40.137Z DRIVER » [Node 019] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      93
                                  └─[ManufacturerSpecificCCGet]
2021-07-10T09:57:40.139Z SERIAL « [ACK]                                                                   (0x06)
2021-07-10T09:57:40.146Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2021-07-10T09:57:40.146Z SERIAL » [ACK]                                                                   (0x06)
2021-07-10T09:57:40.147Z DRIVER « [RES] [SendData]
                                    was sent: true
2021-07-10T09:57:40.161Z SERIAL « 0x010700135d000002b4                                                 (9 bytes)
2021-07-10T09:57:40.162Z SERIAL » [ACK]                                                                   (0x06)
2021-07-10T09:57:40.163Z DRIVER « [REQ] [SendData]
                                    callback id:     93
                                    transmit status: OK
2021-07-10T09:57:40.171Z SERIAL « 0x010e00040013087205006000100001e8                                  (16 bytes)
2021-07-10T09:57:40.172Z CNTRLR   [Node 019] [+] [Manufacturer Specific] manufacturerId: 96         [Endpoint 0]
2021-07-10T09:57:40.172Z CNTRLR   [Node 019] [+] [Manufacturer Specific] productType: 16            [Endpoint 0]
[... 328 lines cut ...]
2021-07-10T09:57:40.680Z SERIAL » [ACK]                                                                   (0x06)
2021-07-10T09:57:40.680Z DRIVER « [Node 019] [REQ] [ApplicationCommand]
[... cut ...]
2021-07-10T09:57:40.691Z SERIAL » [ACK]                                                                   (0x06)
2021-07-10T09:57:40.692Z DRIVER « [RES] [GetRoutingInfo]
                                    node ids: 1, 2, 3, 4, 5, 6, 7, 9, 14, 15, 17, 18
2021-07-10T09:57:40.695Z CNTRLR « [Node 019] node neighbors received: 1, 2, 3, 4, 5, 6, 7, 9, 14, 15, 17, 18
2021-07-10T09:57:40.695Z CNTRLR   [Node 019] Interview stage completed: Neighbors
2021-07-10T09:57:40.695Z CNTRLR   [Node 019] Interview completed
2021-07-10T09:57:40.696Z CNTRLR   [Node 019] The node is ready to be used
`

AlCalzone, 2021-07-11:
@Flight777 that's truncated. Please try again (wait for the "interview complete" message).
Seeing the cache files (the ones starting with the homeid) wouldn't hurt either.

Flight777, 2021-07-11:
Sorry about that, somehow the download log function in the HA Integration seems to truncate automatically, even thought I see the full log lines until completion... Manually copied now, see attached.

Let me check where to get the cache files.

[zwave_js.log](https://github.com/zwave-js/node-zwave-js/files/6796849/zwave_js.log)

Flight777, 2021-07-11:
@AlCalzone

AlCalzone, 2021-07-11:
Ok the Binary Switch CC is at least being queried during the interview, which is good. The cache files can be downloaded here:
https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report?id=download-zip

Flight777, 2021-07-12:
@AlCalzone I don't think I can reach those in Home Assistant OS, cannot get to the Addons files :(

AlCalzone, 2021-07-12:
Any chance you could get zwavejs2mqtt to run outside HA?

ehindriks, 2021-07-13:
> Sorry to jump back in, was trying the new version just now. I indeed see an entity created in HA, however, it does not do anything (change state) when I press the switch on the wall.
> 
> This is what I see in the log, its node 019 (I did do some reinterviews in between to see if that helped):
> 
> [image]
> 
> Don't know if @ehindriks had more luck?

Mine is still working with the latest update, but I must admit that I still have the config file from AlCalzone installed as well. Strangly I noticed that I now have two entities in HA. One is updating the other is not. The one that is updating is named: sensor.in_wall_remote_insert_basic
The one that isn't working is: sensor.in_wall_remote_insert_node_status

AlCalzone, 2021-07-14:
@Flight777 maybe I can work with HA's node dump, but no guarantee.

PR #2688 (2021-05-24): fix(config): enable Basic Set mapping for Everspring HAC01
fixes: #2567
closed 2021-07-25 by zwave-js-assistant[bot]