zwave-js/zwave-js #2568: Aeotec Indoor Siren 6 and Doorbell 6 (ZW164) Undefined values under multiple command classes

BrandonKarl, 2021-05-08:
**Checklist:**

-   [x ] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
-   [ ] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
-   [ ] I have checked the troubleshooting section and my problem is **not** described there.
-   [ ] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

After upgrading my zwavejs2mqtt docker container to zwavejs2mqtt version 4.0.1 and zwave-js version to 7.3.0, my Aeotec Siren had undefined values for nearly every value in the notification, version, battery and manufacturer specific sections. My main issue is that I used to have at least 8 notification values that represented the state of each siren (pl;aying or not). Now they are all undefined and I have a new value just called Siren Status which is idle always (even when siren is playing). 

Interestingly, the sound switch part of it all works still. 

**Device information**

Which device(s) is/are affected (make/model)?  Aeotec Indoor Siren 6 and Doorbell 6 (ZW164
What are the node IDs? 10

**Did you change anything?**

-   [ x] **Yes**: Happened after upgrading
-   [ ] **No**

**Did this use to work before?**

-   [ ] **Don't know**, this is a new device
-   [ ] **No**, it never worked anywhere
-   [ x] **Yes**, in: Sorry do know know exact version, it was not very recent, a few months at least

**How are you using `node-zwave-js`**

-   [x ] `zwavejs2mqtt` (latest) docker image
-   [ ] `zwavejs2mqtt` (dev) docker image
-   [ ] `zwavejs2mqtt` Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] ioBroker.zwave2 adapter
-   [ ] `HomeAssistant` version XYZ
-   [ ] Pkg
-   [ ] Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] `node-red-contrib-zwave-js`
    -   Module version (double click node): 
-   [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github))
-   [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:

Pair a Aeotec Indoor Siren 6 and Doorbell 6 (ZW164) is all I can think of

**Additional context**

Tried re-interviewing many times to no avail. Seems to be the same behavior as https://github.com/home-assistant/core/issues/49630

**Logfile:**
[zwave_js_dump.zip](https://github.com/zwave-js/node-zwave-js/files/6445198/zwave_js_dump.zip)

Please let me know if more info is needed and possibly how to get it. Thank you in advance

raman325, 2021-05-08:
It would help to switch to debug level for logs, reinterview the node, and post the logs here

BrandonKarl, 2021-05-08:
> It would help to switch to debug level for logs, reinterview the node, and post the logs here

[logs.zip](https://github.com/zwave-js/node-zwave-js/files/6445639/logs.zip)

Please let me know if this is the wrong thing or if you need something else. Thanks

raman325, 2021-05-08:
thanks, for now can we close this issue? based on the last comment in the other issue, this may be on the HA side since the other user is reporting that zwavejs2mqtt sees all of the endpoints

BrandonKarl, 2021-05-08:
Hi, I may be wrong here, but I'm having undefined values in the zwjs2mqtt console itself, is that within the scope of the home assistant ticket? Seems to be home assistant not working is just a side effect of the values being weird within zwjs? I asked in the zwave discord over there and they recommended I create the ticket here. If you still think we should close the ticket I will, but I would like to avoid the problem of home assistant thinking its a zwavejs problem while zwavejs thinks its a home assistant problem when (in my very uninformed opinion) it seems more like a zwavejs problem because I run zwjs2mqtt in a completely separated container from home assistant, and still seeing undefined values in there.

raman325, 2021-05-08:
Thanks for pointing that out @BrandonKarl I missed that in your description. Then yes lets keep it here for now, I don't think HA receives undefined values (they get silently dropped) so that may explain why what we see in HA is different from what we see in zwavejs2mqtt

BrandonKarl, 2021-05-08:
Thank you, I can provide any other info you need upon request

kpine, 2021-05-08:
You can attach the node debug info from the zwavejs2mqtt control panel for comparison.

BrandonKarl, 2021-05-08:
@kpine 
[node_debug.zip](https://github.com/zwave-js/node-zwave-js/files/6446147/node_debug.zip)

kpine, 2021-05-08:
The zwavejs2mqtt debug info also shows only a single Siren Status notification value on endpoint 0, so this eliminates HA as the cause.

BrandonKarl, 2021-05-09:
> > It would help to switch to debug level for logs, reinterview the node, and post the logs here
> 
> 
> 
> [logs.zip](https://github.com/zwave-js/node-zwave-js/files/6445639/logs.zip)

@AlCalzone do you need something different than this? Not sure exactly

AlCalzone, 2021-05-09:
🙈 that's the second time I didn't see a logfile today. Must be blind...
I'll let you know.

AlCalzone, 2021-05-09:
Ok it looks like the Endpoint CCs are interviewed in the wrong version for some reason. That's why your endpoints only have `alarmType` and `alarmLevel` values instead of the notifications.

BrandonKarl, 2021-05-10:
> 🙈 that's the second time I didn't see a logfile today. Must be blind...

Haha, no worries, been there.

> Ok it looks like the Endpoint CCs are interviewed in the wrong version for some reason

Sorry not sure what this means really, kind of out of the scope of my knowledge, but if you need anything from me, like to try anything or provide more info please let me know. 😃 I am almost certain there used to be at least 8 notification fields that would get populated while their respective alarms would play if you did not already know.

closed 2021-06-08 by AlCalzone

AlCalzone, 2023-07-18:
Please don't post in multiple old issues.

polskikrol, 2023-07-18:
> Please don't post in multiple old issues.

Apologies. I noticed after posting that this issue was closed, so wasnt sure if any action would be taken. Please feel free to delete as appropriate.

PR #2816 (2021-06-08): fix: always fall back to root to determine endpoint version
The problem was caused by the `Version CC` being mandatory for the endpoints' device classes, which caused us to no longer fall back to the root endpoint version in some cases. Ultimately a case of contracting or unclear specs.

fixes: #2568
fixes: #1449