zwave-js/zwave-js #2570: Add Sunricher Knob Dimmer device

Aesculapius, 2021-05-08:
**Which device is missing?**

Sunricher Dimmer Knob. It appears to be the same to this one: https://devices.zwave-js.io/?jumpTo=0x0330:0x0200:0xd017:1.14 but just reports different product id hence isn't recognized.
I purchased them over here (rebranded): https://www.robbshop.nl/z-wave-draaidimmer-robb-smarrt 
Manual is at: https://www.robbshop.nl/media/73/6e/6e/1607613471/robbsmarrtz-wavedraaidimmermanual_561490868.pdf

**What are the IDs?**

Manufacturer ID: 0x0330  
Product type: 512 (0x0200)  
Product ID: 53271 (0xd017)

**Did you check if we really don't have it?**

- [X ] It is not in the [configuration DB](https://devices.zwave-js.io/)
- [X ] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)
- [X] It is not in the [TODO list](https://github.com/zwave-js/node-zwave-js/issues/1600)

**Do you have a link to the PDF manual?**

Yes: https://www.robbshop.nl/media/73/6e/6e/1607613471/robbsmarrtz-wavedraaidimmermanual_561490868.pdf
If so, it helps in creating the config file.

marcus-j-davies, 2021-05-09:
The DB link you have provided, is for the ID's you want adding - so it's not finding anything.
Can you provide the DB link for the device that it should match the config for?

Aesculapius, 2021-05-09:
Sure, the manual states it had type id "ZV2835RAC-NF", and looking at specs, its closest match looks like this one: https://devices.zwave-js.io/?jumpTo=0x0330:0x0200:0xd010:0.0

But is is missing some config paramters like 8 (Setting step of the knob), as can be seen in the linked manual.

closed 2021-05-25 by AlCalzone

Aesculapius, 2021-05-29:
Although it seems to link to a known item in the database, it still shows 'unknown manufacturer':
[image]

Or is this a zwavejs2Mqtt thing and should I post there?

AlCalzone, 2021-05-29:
Can you post a driver log of the startup?

Aesculapius, 2021-05-30:
I set the log level to Info, is that sufficient?

`2021-05-30 06:51:04.535 INFO ZWAVE: Connecting to tcp://10.0.15.11:3333
2021-05-30 06:51:04.526 INFO MQTT: MQTT is disabled
2021-05-30 06:51:04.548 INFO ZWAVE: Zwavejs usage statistics ENABLED
2021-05-30 06:51:04.560 INFO APP: POST /api/settings 200 52.930 ms - 753
2021-05-30 06:51:04.635 INFO APP: GET /api/auth-enabled 304 1.193 ms - -
2021-05-30 06:51:04.699 INFO ZWAVE: Next update scheduled for: Mon May 31 2021 00:00:00 GMT+0000 (Coordinated Universal Time)
2021-05-30 06:51:06.943 INFO ZWAVE: Zwave driver is ready
2021-05-30 06:51:06.944 INFO ZWAVE: Controller status: Driver ready
2021-05-30 06:51:06.949 INFO ZWAVE: Scanning network with homeid: 0xe1ed6cbf
2021-05-30 06:51:06.950 INFO ZWAVE: Node 1 is alive
2021-05-30 06:51:07.087 INFO ZWAVE: Node 1 ready: AEON Labs - ZW090 (Z‐Stick Gen5 USB Controller)
2021-05-30 06:51:07.089 INFO ZWAVE: Node 8 ready: Shenzhen Neo Electronics Co., Ltd. - PD03Z (PIR Motion Sensor with Temperature Sensor)
2021-05-30 06:51:07.090 INFO ZWAVE: Node 11 ready: Shenzhen Neo Electronics Co., Ltd. - PD03Z (PIR Motion Sensor with Temperature Sensor)
2021-05-30 06:51:07.092 INFO ZWAVE: Node 19 ready: AEON Labs - ZW122 (Water Sensor 6)
2021-05-30 06:51:07.093 INFO ZWAVE: Node 24 ready: Shenzhen Neo Electronics Co., Ltd. - NAS-DS01Z (Door/Window Sensor)
2021-05-30 06:51:07.094 INFO ZWAVE: Node 25 ready: Shenzhen Neo Electronics Co., Ltd. - NAS-DS01Z (Door/Window Sensor)
2021-05-30 06:51:07.095 INFO ZWAVE: Node 26 ready: Shenzhen Neo Electronics Co., Ltd. - PD03Z (PIR Motion Sensor with Temperature Sensor)
2021-05-30 06:51:07.096 INFO ZWAVE: Node 28 ready: Shenzhen Neo Electronics Co., Ltd. - PD03Z (PIR Motion Sensor with Temperature Sensor)
2021-05-30 06:51:07.096 INFO ZWAVE: Node 29 ready: Shenzhen Neo Electronics Co., Ltd. - PD03Z (PIR Motion Sensor with Temperature Sensor)
2021-05-30 06:51:07.160 INFO ZWAVE: Node 5 is alive
2021-05-30 06:51:07.172 INFO ZWAVE: Node 5: value added 5-38-0-targetValue => 0
2021-05-30 06:51:07.173 INFO ZWAVE: Node 5: value added 5-38-0-duration => 0s
2021-05-30 06:51:07.193 INFO ZWAVE: Node 5: value added 5-38-0-currentValue => 0
2021-05-30 06:51:07.194 INFO ZWAVE: Node 5: value added 5-38-0-Up => undefined
2021-05-30 06:51:07.194 INFO ZWAVE: Node 5: value added 5-38-0-Down => undefined
2021-05-30 06:51:07.194 INFO ZWAVE: Node 5: value added 5-43-0-sceneId => undefined
2021-05-30 06:51:07.195 INFO ZWAVE: Node 5: value added 5-43-0-dimmingDuration => undefined
2021-05-30 06:51:07.195 INFO ZWAVE: Node 5: value added 5-50-0-value-65537 => 4.616
2021-05-30 06:51:07.196 INFO ZWAVE: Node 5: value added 5-50-0-value-66049 => 0
2021-05-30 06:51:07.196 INFO ZWAVE: Node 5: value added 5-50-0-value-66561 => 1.968
2021-05-30 06:51:07.196 INFO ZWAVE: Node 5: value added 5-50-0-value-66817 => 0
2021-05-30 06:51:07.197 INFO ZWAVE: Node 5: value added 5-50-0-reset => undefined
2021-05-30 06:51:07.197 INFO ZWAVE: Node 5: value added 5-113-0-Power Management-Over-current status => 0
2021-05-30 06:51:07.197 INFO ZWAVE: Node 5: value added 5-113-0-System-Emergency shutoff status => 0
2021-05-30 06:51:07.198 INFO ZWAVE: Node 5: value added 5-114-0-manufacturerId => 816
2021-05-30 06:51:07.198 INFO ZWAVE: Node 5: value added 5-114-0-productType => 512
2021-05-30 06:51:07.198 INFO ZWAVE: Node 5: value added 5-114-0-productId => 53271
2021-05-30 06:51:07.199 INFO ZWAVE: Node 5: value added 5-134-0-libraryType => 3
2021-05-30 06:51:07.199 INFO ZWAVE: Node 5: value added 5-134-0-protocolVersion => 5.3
2021-05-30 06:51:07.199 INFO ZWAVE: Node 5: value added 5-134-0-firmwareVersions => 1.14
2021-05-30 06:51:07.199 INFO ZWAVE: Node 5: value added 5-134-0-hardwareVersion => 1
2021-05-30 06:51:07.201 INFO ZWAVE: Node 5 ready: Unknown manufacturer 0x0330 - Unknown product 0xd017 (0x0200)
2021-05-30 06:51:07.283 INFO ZWAVE: Node 6 is alive
2021-05-30 06:51:07.288 INFO ZWAVE: Node 6: value added 6-38-0-targetValue => 0
2021-05-30 06:51:07.288 INFO ZWAVE: Node 6: value added 6-38-0-duration => 0s
2021-05-30 06:51:07.290 INFO ZWAVE: Node 6: value added 6-38-0-currentValue => 0
2021-05-30 06:51:07.290 INFO ZWAVE: Node 6: value added 6-38-0-Up => undefined
2021-05-30 06:51:07.291 INFO ZWAVE: Node 6: value added 6-38-0-Down => undefined
2021-05-30 06:51:07.291 INFO ZWAVE: Node 6: value added 6-43-0-sceneId => undefined
2021-05-30 06:51:07.291 INFO ZWAVE: Node 6: value added 6-43-0-dimmingDuration => undefined
2021-05-30 06:51:07.291 INFO ZWAVE: Node 6: value added 6-50-0-value-66561 => 5.262
2021-05-30 06:51:07.292 INFO ZWAVE: Node 6: value added 6-50-0-value-66817 => 0
2021-05-30 06:51:07.295 INFO ZWAVE: Node 6: value added 6-50-0-value-65537 => 12.488
2021-05-30 06:51:07.295 INFO ZWAVE: Node 6: value added 6-50-0-value-66049 => 0
2021-05-30 06:51:07.295 INFO ZWAVE: Node 6: value added 6-50-0-reset => undefined
2021-05-30 06:51:07.296 INFO ZWAVE: Node 6: value added 6-113-0-Power Management-Over-current status => 0
2021-05-30 06:51:07.296 INFO ZWAVE: Node 6: value added 6-113-0-System-Emergency shutoff status => 0
2021-05-30 06:51:07.296 INFO ZWAVE: Node 6: value added 6-114-0-manufacturerId => 816
2021-05-30 06:51:07.297 INFO ZWAVE: Node 6: value added 6-114-0-productType => 512
2021-05-30 06:51:07.297 INFO ZWAVE: Node 6: value added 6-114-0-productId => 53271
2021-05-30 06:51:07.297 INFO ZWAVE: Node 6: value added 6-134-0-libraryType => 3
2021-05-30 06:51:07.297 INFO ZWAVE: Node 6: value added 6-134-0-protocolVersion => 5.3
[... cut ...]
2021-05-30 06:52:34.337 INFO ZWAVE: Node 17: value updated: 48-0-Any true => false
2021-05-30 06:52:35.109 INFO ZWAVE: Node 17: value updated: 113-0-Home Security-Motion sensor status 8 => 0
2021-05-30 06:52:43.924 WARN APP: Shutdown detected: closing clients...
2021-05-30 06:52:43.924 INFO GATEWAY: Closing Gateway...
2021-05-30 06:52:43.936 INFO ZWAVE: Client closed
2021-05-30 06:52:45.226 INFO ZWAVE: Connecting to tcp://10.0.15.11:3333
2021-05-30 06:52:45.219 INFO MQTT: MQTT is disabled
`

(I removed some entities to keep the log compact).

AlCalzone, 2021-05-30:
That's not a driver log. Please read https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report?id=driver-logs

And I need log level "debug", info is not enough.

Aesculapius, 2021-05-30:
Apologies, attached. 
[zwavejs_2021-05-30.log](https://github.com/zwave-js/node-zwave-js/files/6566693/zwavejs_2021-05-30.log)
Thanks.

AlCalzone, 2021-05-30:
```
2021-05-30T18:10:26.560Z DRIVER   version 7.7.0
2021-05-30T18:10:26.560Z CONFIG   version 7.3.1-20210503
```
You have an outdated version of a configuration DB loaded somehow - maybe an incorrect Docker volume mapping?

Aesculapius, 2021-05-30:
Not sure what I should do about that? I run it using docker-compose and this compose file: https://github.com/zwave-js/zwavejs2mqtt/blob/master/docker/docker-compose.yml

Updating is done by issuing `docker-compose pull` and `docker-compose up -d --remove-orphans`

PR #2657 (2021-05-25): feat(config): added support for sunricher ZV2835RAC-NF
closes #2570.