zwave-js/zwave-js #2573: Device issues after move from other driver - shows interviewed, but no info/entities

AnthiasD, 2021-05-09:
**Checklist:**

-   [x ] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
-   [x ] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
-   [x ] I have checked the troubleshooting section and my problem is **not** described there.
-   [x ] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

System: Home Assistant
Moved from Z-Wave (deprecated) integration to ZWaveJS2MQTT and the Z-Wave JS integration. Most devices transferred over with no issues. I have a few devices that are showing like this:
[image]

Log for nodes 10, 11, 12:
[zwavejs_352.log](https://github.com/zwave-js/node-zwave-js/files/6446996/zwavejs_352.log)

Node 10 device page, shows a single entity that is off - light is currently on. Previously had multiple entities:
[image]
Node 10 export file: 
[node_10.txt](https://github.com/zwave-js/node-zwave-js/files/6447001/node_10.txt)

Other nodes with same issue show the same single entity listing.

**Device information**

Using node 10 as an example going forward. Attached is the json export. 
[node_10.txt](https://github.com/zwave-js/node-zwave-js/files/6446985/node_10.txt)
The device is an Aeotec Limited ZWA002 LED Bulb 6 Multi-Color
10, 11, 12 are the same model

24 and 25 are Zipato RGBW bulbs

**Did you change anything?**

-   [x ] **Yes**: _(please describe)_
    - Moved from Z-wave (deprecated) to Zwave JS and ZwaveJS2MQTT. Has not worked since the move.
-   [ ] **No**

**Did this use to work before?**

-   [ ] **Don't know**, this is a new device
-   [ ] **No**, it never worked anywhere
-   [x ] **Yes**, in: _(specify application with versions)_
    - Worked with Zwave (Deprecated) with no issue. Has not worked since move to the new ZWave drivers. Other devices are working. 

**How are you using `node-zwave-js`**

-   [x ] `HomeAssistant` version 7.3.0

**To Reproduce**

Steps to reproduce the behavior:

1) Uninstalled old integration.
2) Restarted server
3) installed zwavejs2mqtt add-on
4) started add-on
5) configured add-on
6) added Z-Wave JS integration
7) Devices started updating
8) after identifying there was an issue with these nodes
     i) restarted
     ii) re-interviewed all of them at least once
9) give it time to shake out while not home - no change after 12 hours

**Additional context**

I have not yet, but will be attempting to re-add one or two of the devices to see if that takes care of it - location is difficult to reach. 

**Logfile:**

[zwave_js_dump.txt](https://github.com/zwave-js/node-zwave-js/files/6447018/zwave_js_dump.txt)

AlCalzone, 2021-05-09:
Re-adding will not be necessary. Unfortunately, the dump does not help debugging this. 
Please make a zwave-js log, loglevel `debug` or `silly` and attach it here as a file.

AnthiasD, 2021-05-09:
The section in my issue description with: 
   Log for nodes 10, 11, 12:
   zwavejs_352.log
is a silly level log where nodes 10,11,12 were specified to log. Does that cover what is needed, or a different log record?

AlCalzone, 2021-05-09:
Oh apologies - I only noticed the zwave_js_dump.txt under "Logfile".

AnthiasD, 2021-05-09:
As an update, now all my devices are showing as unknown manufacturer/product, but they have actual manufacturer/product codes, and entities show up as they should. This happened while everything was running fine, re-interviews and a restart did not have any impact.
[image]
[image]

AlCalzone, 2021-05-10:
Nodes 10-12 at least are included securely. However, they do not respond when asked for their securely supported CCs, so the driver assumes them to be insecure. This only leaves you with `Z-Wave Plus Info` and `Security` CCs which is basically no information to work with.

Please double-check the network key - this usually happens when that one is wrong.

AnthiasD, 2021-05-12:
Thanks for taking a look, that handled it. I had the wrong key written down. Thankfully this is virtualized, and I was able to make a copy, roll it back, grab the correction, and do a check without changing anything.

closed 2021-05-12 by AnthiasD