zwave-js/zwave-js #2600: [bug] All battery devices are not ready after upgrading to 4.2.0

kpine, 2021-05-12:
### Version

**Checklist:**

- [ ] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is **not** described there.
- [X] I have read the changelog problem was **not** mentioned there.

Build/Run method

- [X] Docker
- [ ] PKG
- [ ] Snap package
- [ ] Manually built (git clone - yarn install - yarn run build )

zwavejs2mqtt version: 4.2.0
zwave-js version: 7.4.9

### Describe the bug

After upgrading to 4.2.0, all of my battery devices are shown w/o any information in the control panel. The node debug info shows `"ready": false,`. However, HA the zwave_js integration is functioning perfectly. HA considers the nodes as ready and all values are populated.

This also appears to affect the controller node as well.

### To Reproduce

Steps to reproduce the behavior:

1. Upgrade to zwavejs2mqtt 4.2.0
2. Goto the control panel
3. Notice all battery devices are unavailable
4. When using the HA zwave_js integration, notice it is working

### Expected behavior

The nodes should be populated and values visible

### Additional context

I am using a custom `STORE_DIR` and noticed the external DB [wasn't populated](https://github.com/zwave-js/zwavejs2mqtt/issues/1181). I thought it might be related, but after fixing that and restarting, the issue persists.

[image]

Docker logs: [z2m.log](https://github.com/zwave-js/zwavejs2mqtt/files/6461111/z2m.log)
Data store, with zwave-js logs: [zwavejs2mqtt-store.zip](https://github.com/zwave-js/zwavejs2mqtt/files/6461091/zwavejs2mqtt-store.zip)
Here's a dump from HA: [zwave_js_dump.json.txt](https://github.com/zwave-js/zwavejs2mqtt/files/6461083/zwave_js_dump.json.txt)

robertsLando, 2021-05-12:
@AlCalzone

robertsLando, 2021-05-12:
> I am using a custom STORE_DIR and noticed the external DB wasn't populated

If so you should change the env var `ENV ZWAVEJS_EXTERNAL_CONFIG=/usr/src/app/store/.config-db` to match that custom directory. About nodes not ready I'm not sure if this could be related

robertsLando, 2021-05-12:
Based on your logs seems like the `ready` event is never triggered from those nodes. @AlCalzone thoughts?

kpine, 2021-05-12:
> If so you should change the env var ENV ZWAVEJS_EXTERNAL_CONFIG=/usr/src/app/store/.config-db to match that custom directory. 

Just to confirm, I did make this change prior to submitting the issue. The directory is being populated now.

> Based on your logs seems like the ready event is never triggered from those nodes

Hmm, if the ready event is indicated by log `"The node is ready to be used`", then I think all nodes are accounted for. I just did a restart to confirm, and I see:

```
$ grep 'The node is ready' zwavejs_1.log_2021-05-11.log
...<snip previous restarts>...
2021-05-12T06:40:26.867Z CNTRLR   [Node 001] The node is ready to be used
2021-05-12T06:40:26.919Z CNTRLR   [Node 003] The node is ready to be used
2021-05-12T06:40:26.925Z CNTRLR   [Node 009] The node is ready to be used
2021-05-12T06:40:26.931Z CNTRLR   [Node 010] The node is ready to be used
2021-05-12T06:40:26.936Z CNTRLR   [Node 011] The node is ready to be used
2021-05-12T06:40:26.938Z CNTRLR   [Node 012] The node is ready to be used
2021-05-12T06:40:26.940Z CNTRLR   [Node 013] The node is ready to be used
2021-05-12T06:40:26.959Z CNTRLR   [Node 016] The node is ready to be used
2021-05-12T06:40:26.962Z CNTRLR   [Node 017] The node is ready to be used
2021-05-12T06:40:26.964Z CNTRLR   [Node 018] The node is ready to be used
2021-05-12T06:40:26.966Z CNTRLR   [Node 019] The node is ready to be used
2021-05-12T06:40:26.968Z CNTRLR   [Node 020] The node is ready to be used
2021-05-12T06:40:26.976Z CNTRLR   [Node 023] The node is ready to be used
2021-05-12T06:40:27.564Z CNTRLR   [Node 002] The node is ready to be used
2021-05-12T06:40:27.630Z CNTRLR   [Node 014] The node is ready to be used
2021-05-12T06:40:27.899Z CNTRLR   [Node 015] The node is ready to be used
2021-05-12T06:40:27.969Z CNTRLR   [Node 021] The node is ready to be used
2021-05-12T06:40:28.124Z CNTRLR   [Node 022] The node is ready to be used
2021-05-12T06:40:28.164Z CNTRLR   [Node 024] The node is ready to be used
2021-05-12T06:40:28.423Z CNTRLR   [Node 025] The node is ready to be used
```
There is a log for every one of my nodes (20 of them). Also, the HA zwave_js integration sees all the nodes as ready and all the entities are functioning normally.

robertsLando, 2021-05-12:
@kpine if you check z2m logs and grep lines with `Node XX ready` you will see that not every node is listed there. I dunno if this could be caused by the new way i'm fetching node neighbors, maybe the request of getting node neighbors remains stale?

AlCalzone, 2021-05-12:
> the promise never resolves on battery devices

The communication is done directly with the controller, so this shouldn't be it - here's an example for a dead node:

```
09:30:23.347 CNTRLR   [Node 015] The node is dead.
09:30:23.347 CNTRLR   All nodes are ready to be used
09:30:23.348 CNTRLR   [Node 015] ping failed: Failed to send the command after 1 attempts (Status No
                      Ack)
09:30:29.097 CNTRLR » [Node 015] requesting node neighbors...
...
09:30:29.124 CNTRLR « [Node 015] node neighbors received: 1, 7, 17
```

bushvin, 2021-05-12:
Same behaviour here. This is post saving z-wave configuration changes:
```sh
# docker logs zwavejs2mqtt_silabs |awk '/ready/' |sort -k 4
14:15:17.460 CNTRLR   [Node 001] The node is ready to be used
14:15:17.979 CNTRLR   [Node 002] The node is ready to be used
14:15:18.011 CNTRLR   [Node 003] The node is ready to be used
14:15:18.104 CNTRLR   [Node 004] The node is ready to be used
14:15:17.469 CNTRLR   [Node 005] The node is ready to be used
14:15:17.473 CNTRLR   [Node 006] The node is ready to be used
14:15:17.475 CNTRLR   [Node 007] The node is ready to be used
14:15:18.136 CNTRLR   [Node 008] The node is ready to be used
14:15:18.822 CNTRLR   [Node 009] The node is ready to be used
14:15:18.853 CNTRLR   [Node 010] The node is ready to be used
14:15:18.963 CNTRLR   [Node 011] The node is ready to be used
14:15:19.011 CNTRLR   [Node 012] The node is ready to be used
14:15:17.488 CNTRLR   [Node 013] The node is ready to be used
14:15:17.490 CNTRLR   [Node 014] The node is ready to be used
14:15:19.643 CNTRLR   [Node 015] The node is ready to be used
14:15:19.698 CNTRLR   [Node 017] The node is ready to be used
14:15:19.812 CNTRLR   [Node 018] The node is ready to be used
14:15:19.843 CNTRLR   [Node 020] The node is ready to be used
14:15:20.039 CNTRLR   [Node 021] The node is ready to be used
14:15:20.090 CNTRLR   [Node 022] The node is ready to be used
14:15:20.222 CNTRLR   [Node 023] The node is ready to be used
14:15:21.452 CNTRLR   [Node 024] The node is ready to be used
14:15:17.509 CNTRLR   [Node 025] The node is ready to be used
14:15:17.513 CNTRLR   [Node 027] The node is ready to be used
14:15:21.592 CNTRLR   [Node 029] The node is ready to be used
14:15:22.462 CNTRLR   [Node 031] The node is ready to be used
14:15:17.519 CNTRLR   [Node 032] The node is ready to be used
14:15:17.521 CNTRLR   [Node 033] The node is ready to be used
14:15:17.523 CNTRLR   [Node 035] The node is ready to be used
14:15:17.525 CNTRLR   [Node 038] The node is ready to be used
14:15:22.660 CNTRLR   [Node 039] The node is ready to be used
14:15:22.691 CNTRLR   [Node 040] The node is ready to be used
14:15:22.692 CNTRLR   All nodes are ready to be used
```

It seems this happens every time the daemon is reloaded: I made sure all my battery-owered nodes were correctly interviewed and healed.
And then I changing my z-wave logging settings from info to warn and back. Result: all nodes were *blank* in the UI.

Running on docker
```yml
---
version: '3.7'
services:
  zwavejs2mqtt_silabs:
    container_name: zwavejs2mqtt_silabs
    image: zwavejs/zwavejs2mqtt:4.2.0
    restart: always
    tty: true
    stop_signal: SIGINT
    networks:
      - zwavejs
    devices:
      - '/dev/ttyUSB0:/dev/ttyUSB0'
    volumes:
      - zwavejs_store_silabs:/usr/src/app/store
    ports:
      - '8092:8091'
    environment:
      - TZ=Europe/Brussels
      - NETWORK_KEY=xxxxxxxxx
      - SESSION_SECRET=xxxxxxxxx

networks:
  zwavejs:
volumes:
  zwavejs_store_silabs:
```
I created the `.config-db` in `/usr/src/app/store`, no population of devices, *blank* battery-powered devices in the UI.
I added `ZWAVEJS_EXTERNAL_CONFIG=/usr/src/app/store/.config-db` to the environment section of the compose file, the devices got populated, but the UI showed *blank* entries for battery-powered devices.

Each time I made sure one node was completely re-interviewed and had the information properly populated.

robertsLando, 2021-05-12:
I have found a possible race condition on code initroduced by https://github.com/zwave-js/zwavejs2mqtt/pull/1146, could someone give a try to https://github.com/zwave-js/zwavejs2mqtt/pull/1195 and give me some feedback?

bushvin, 2021-05-12:
Let me know which docker image I should use, and I'll give you all the feedback you want ;)

robertsLando, 2021-05-12:
There is no docker image ready to test it, you should build it your own. BTW @AlCalzone is testing it so once he gives me a 👍🏼 I will merge that and release a patch :)

bushvin, 2021-05-12:
Ah, I am not that well versed in docker images, unfortunately. Trying to find the time, but reality and time constraints are always in my way.

closed 2021-05-12 by robertsLando

rroller, 2021-05-12:
The fix is merged and now under the latest tag in Docker. I verified that the fix works for me.

robertsLando, 2021-05-12:
Thanks for the feedback @rroller :)