zwave-js/zwave-js #2604: Do not direct to Home Assistant github unless using zwave_js integration

Altycoder, 2021-05-12:
Do not direct home assistant users to the home assistant page unless they are using the zwave_js integration.  Users using an mqtt broker need to file issues here.

Item 1 in the new issues template directs all home assistant users away from here so it needs to be re-worded.

Suggest changing the wording on the new issues template from:

  1. Are you using HomeAssistant?
  If yes, please open your issue at https://github.com/home-assistant/core/issues
  UNLESS a developer told you to come here.

To:

  1. Are you using HomeAssistant **with the zwave_js integration**?
  If yes, please open your issue at https://github.com/home-assistant/core/issues
  UNLESS a developer told you to come here.

AlCalzone, 2021-05-12:
Accepting PR

blhoward2, 2021-05-17:
@kreene1987 I see that you pushed a commit to your local repository referencing this issue. Did you mean to push a PR to the project pushing that change over?

kreene1987, 2021-05-17:
Yep, very new to GH, let me see if I can get it figured out.

blhoward2, 2021-05-17:
It's not a commit. Go to the GitHub website, compare your branch to the master at the project, and create a pull request.

AlCalzone, 2021-05-18:
closed in #2633

closed 2021-05-18 by AlCalzone