zwave-js/zwave-js #2605: Provide raw AlarmType and AlarmLevel in notification

poldim, 2021-05-12:
**Is your feature request related to a problem? Please describe.**
Currently, there is no way to extract the true alarmstate & alarmtype as the values were/are originally in the device rather than the newer transposed values presented in the zwave_js_notification.  The created but disabled entities for alarmstate & alarmtype are never updated and always stay in a state of Unknown once they are enabled.

I'm getting the below on an event 3 (RF lock operation) but per the Yale manual, I would expect to see 24 (RF Lock).  I did some searching but couldn't find anything.

[image]

**Describe the solution you'd like**
Fully expose alarmstate & alarmtype as entities when that lock is added.  Users can disable them if they aren't needed.

[image]

**Describe alternatives you've considered**
I've tried re-interviewing the device, but this does not help.
[image]

blhoward2, 2021-05-13:
This a multi-layered problem and we first need to figure out in which project the problem lies. This could be an issue with the integration of the driver. Let’s treat it as a bug before we consider an enhancement. 

Please post silly level logs from zwave-js showing the device interview and the alarm message being received and send. Please also post a node export. 

If you’re running zwavejs2mqtt, or could try doing so, it would help to see how this is setup in the UI there and whether the entities update on their end.

raman325, 2021-05-13:
Let me clarify - if a device config has alarm levels and alarm types mapped to the notification CC, all you get are notification events when those value update occurs. This is a feature request to update the alarm level and alarm type values even if the value is mapped (and consider that unmapped values still get sent to the alarm type and alarm level value, so the user essentially has to watch in two places). This was something that @AlCalzone and I discussed in another thread, I don't know that we decided whether we should do it or not.

blhoward2, 2021-05-13:
Got it, thanks.

poldim, 2021-05-14:
Just making sure I'm clear and not confusing myself or anyone else: 
Yale lock is paired, interview complete, the disabled entities for AlarmType and AlarmLevel have been enabled (for days).
I operated the lock via RF, any am monitoring both notifications and events. Only the notification is logged and no event. The original AlarmType and AlarmLevel are not sent.

[image]

[image]

```json
{"event_type":"zwave_js_notification","topic":"zwave_js_notification","payload":{"event_type":"zwave_js_notification","event":{"domain":"zwave_js","node_id":15,"home_id":3518803333,"device_id":"2162fe43b6cac3a902d59cea27e0d144","command_class":113,"command_class_name":"Notification","label":"Access Control","type":6,"event":4,"event_label":"RF unlock operation","parameters":{}},"origin":"LOCAL","time_fired":"2021-05-14T22:22:26.343100+00:00","context":{"id":"1cc63658f4442f11e37de996d9c34d51","parent_id":null,"user_id":null}},"_msgid":"def8c756.4b36e8"}
```

poldim, 2021-05-14:
> This is a multi-layered problem and we first need to figure out in which project the problem lies. This could be an issue with the integration or the driver. Let’s treat the entities not working as a bug before we consider an enhancement.
> 
> Please post silly level logs from zwave-js showing the device interview and the alarm message being received and sent. Please also post a node export. And please post what version of zwave-js you are running.
> 
> If you’re running zwavejs2mqtt, or could try doing so, it would help to see how this is setup in the UI there and whether the entities update on their end.
> 
> /cc @raman325

I'm running the HASSio set with ZwaveJS integration.  I couldn't find/get access to the addon's files, so I had to pull them through the zwavejs2mqtt GUI.  There's a ton of zwavejs little logs (see below), but hopefully I'm attaching the ones you actually need.

[image]

[zwavejs_355.log_2021-05-14.log](https://github.com/zwave-js/node-zwave-js/files/6481514/zwavejs_355.log_2021-05-14.log)
[zwavejs_354.log_2021-05-14.log](https://github.com/zwave-js/node-zwave-js/files/6481515/zwavejs_354.log_2021-05-14.log)
[zwavejs2mqtt.log](https://github.com/zwave-js/node-zwave-js/files/6481516/zwavejs2mqtt.log)

blhoward2, 2021-05-15:
The log is named based on the process ID. It appears that you're doing something causing it to restart a bunch?

poldim, 2021-05-15:
was setting up the pi that day so lots of restarts

AlCalzone, 2021-05-15:
> if a device config has alarm levels and alarm types mapped to the notification CC, all you get are notification events when those value update occurs. This is a feature request to update the alarm [...]

This. The reason why we're not using the raw `alarmType` and `alarmLevel` for mapped notifications is because they are duplicating functionality in a worse way. In order to interpret these two values, you need to look up their meaning somewhere, whereas the standardized notifications always mean the same. We still use `alarmType` and `alarmLevel` for those values that cannot be mapped 1:1.

> I'm getting the below on an event 3 (RF lock operation) but per the Yale manual, I would expect to see 24 (RF Lock). I did some searching but couldn't find anything.

That's correct. The unstandardized `alarmType 24` is mapped to the standardized notification event `3` (RF lock operation):
https://github.com/zwave-js/node-zwave-js/blob/9260dd1105af1d0f933822085fb53a70b56f9713/packages/config/config/notifications.json#L543

closed 2021-05-24 by AlCalzone

poldim, 2021-05-24:
> Okay I take back the part about not doing this. We actually need to do it in order to avoid breaking automations when adding a mapping
> [#2409 (comment)](https://github.com/zwave-js/node-zwave-js/issues/2409#issuecomment-823638655)

Thanks.  I see the two values are now populated.  However, they've been the same for several days and I know the lock has been used.  Do I need to do something for it to continuously send/receive/update these values?

[image]
[image]

raman325, 2021-05-25:
@poldim zwave-js-server and zwavejs2mqtt need to be updated to use the latest version of zwave-js. This will probably happen soon

mmiller7, 2022-03-20:
Is there some way to tell what something maps to for this "new way"?

For example, I have some Yale deadbolts that (among other things) the manual states have alarms for things like "Jammed" (type=9), "Front Escutcheon Removed" (type=161,level=2), or "Unlocked by master PIN" (type=19,level=251) among others?

Previous Z-Wave on Home Assistant it was trivial to reference the manufacturer list of valid values and write automation triggers easily by alarm_type and alarm_level for any zwave device, but the way zwave-js is trying to abstract these makes it extremely hard to guess what something might map to or if it is at all.

AlCalzone, 2022-03-20:
@marcus-j-davies I think it could be a good idea to document a select subset of compat flags in the device browser for lookup.
For example the mapping of alarm -> notifications

marcus-j-davies, 2022-03-20:
I can't see why not - do you have an example config?

marcus-j-davies, 2022-03-20:
Oh! I'll model it using this device 😄

mmiller7, 2022-03-20:
Yeah like I would prefer to ask for a generic "how do you do it" for any given thing...although my primary interest is in my Yale AssureLock SL Z-Wave deadbolts.

So like back when I was figuring this all out I used a combination of things...and ended up testing all I could on my table before installing them.  I ultimately built a spreadsheet of things it appeared my lock "probably supported" and "definitely supported".

The "old way" I found 2 documents which helped me understand all the things that my lock seemed to send back (because neither seemed 100% complete)

https://www.homecontrolassistant.com/download/Doc/TechNotes/TechNote_304_ZWaveLocks.pdf

https://us.yalehome.com/Yale/Yale%20US/Real%20Living/home%20control/Compatible%20Z-Wave%20Systems/Yale%20Real%20Living%20ZWave%20Plus%20System%20Integrators%20Guide%20YRL%20deadbolts.pdf

These docs seemed wonderful to me - just the sort of interface design specification docs I am used to with network things at work.  Tells me every possible value "this thing" might send and how to interpret it.  No guessing where it comes from.

Between them, I then built an Excel spreadsheet and tested as much as I could from that (some I couldn't test like battery alerts on demand, others I could like unplugging the keypad to "simulate" it being damaged/destroyed/removed).
(attached, if you care
[ZWave Codes Translation.xlsx](https://github.com/zwave-js/node-zwave-js/files/8311403/ZWave.Codes.Translation.xlsx)
)

Armed with this knowledge, I easily paired that down to just what my lock actually seemed to have on features (e.g. mine I don't think can do scheduling) and wrote my automation and YAML off that with a nice sensor in no time.

Old sensor:
`
sensor:
  - platform: template
    sensors:
      deadbolt_inside_garage_door_status:
        friendly_name: "Inside Garage Door Lock"
        value_template: >-
          {% set lock_entity_id = 'sensor.deadbolt_inside_garage_door' %}
          {% set alarm_type = states(lock_entity_id + '_alarm_type') | int(0) %}
          {% set alarm_level = states(lock_entity_id + '_alarm_level') | int(0) %}

          {% if alarm_type == 0 %}
              No Report
          {% elif alarm_type == 9 %}
              Motor Jammed
          {% elif alarm_type == 10 %}
              Handing Completed
          {% elif alarm_type == 18 %}
              {% if alarm_level == 251 %}
                  Locked by Keypad Master PIN
              {% else %}
                  Locked by Keypad Slot {{ alarm_level }} PIN
              {% endif %}
          {% elif alarm_type == 19 %}
              {% if alarm_level == 251 %}
                  Unlocked by Keypad Master PIN
              {% else %}
                  Unlocked by Keypad Slot {{ alarm_level }} PIN
              {% endif %}
          {% elif alarm_type == 21 %}
              {% if alarm_level == 1 %}
                  Locked by Manual Mechanical
              {% elif alarm_level == 2 %}
                  Locked by Keypad Touch
              {% else %}
                  Locked by Manual
              {% endif %}
          {% elif alarm_type == 22 %}
              Unlock by Manual Mechanical
          {% elif alarm_type == 24 %}
              Locked by RF Remote
          {% elif alarm_type == 25 %}
              Unlocked by RF Remote
          {% elif alarm_type == 27 %}
              Locked by Auto-Lock
          {% elif alarm_type == 33 %}
              Deleted Slot {{ alarm_level }} PIN
          {% elif alarm_type == 48 %}
              Factory Reset Defaults
          {% elif alarm_type == 112 %}
[... 29 lines cut ...]

Normal-events:
"unlocked with a valid PIN" - this means someone is entering and I can do some stuff like turn on lights, adjust HVAC, unlock inner door when outer door was unlocked by authorized PIN

Suspicious events:
Unlocked outer door by manual inside knob when nobody should be home

Non-Normal events - so I get notifications on most any other event
Motor Jammed
Lock/unlock by Master PIN (nobody should use this, that means it may be compromised)
Delete user
Factory reset
Master PIN change
Add user
Duplicate PIN added
Power to RF module cycled
Too many PIN attempts
Front Escutcheon removed
Low battery (according to the lock, not %)
Critical battery (according to the lock, not %)
Too low power to operate

When I had things briefly working on ZwaveJS in my attempts to migrate, I reproduced a "door not quite shut motor jammed" and couldn't figure out where or how it tells me.  I also dismantled one of them and unplugged the keypad and didn't see any notification events fire about tampering.  I didn't go thru everything exhaustively, but I can't figure out how I would get a list of every possible valid value my devices might send over so I can write sensors and automations on them.

Somehow there ought to be a way to "map" all the valid alarm type/levels to a new whatever event without needing trial-and-error. THAT is the process I'm asking for, and I would prefer if it wasn't specific to one device but a more general "if you find the docs on a specific device, you can translate it".

AlCalzone, 2022-03-20:
> I can't figure out how I would get a list of every possible valid value my devices might send over so I can write sensors and automations on them.

You won't find this in this project. The reason we do the mapping at all is because starting with Notification CC V2, the `notificationType` and `notificationEvent` values are **standardized** and can be queried for support. Locks frequently don't use them though and resort to unstandardized and un-queryable alarm values (`alarmType` and `alarmLevel`).
Only by mapping some of the alarm values to standardized notification values we are able to build generic behavior, like updating lock states, on top of this.

The best we can do is tell you in a more prominent way which alarm values are mapped to which notification values, which is what I'm suggesting here. With this information and some help of the manual, you can distinguish:
a) which **notifications** to look out for
a) which remaining alarm **values** there are and cannot be mapped.

Not every possible alarm value translates to a valid notification, so we have to stick with this dual approach.

marcus-j-davies, 2022-03-29:
@AlCalzone 
Alarm Mappings are now displayed on the Browser. 👍

mmiller7, 2022-03-29:
> @AlCalzone Alarm Mappings are now displayed on the Browser. +1

where do you go to look for that?

marcus-j-davies, 2022-03-29:
They now appear on the BD Browser (last table) - providing the device you're looking, has them recorded.

YRD210
https://devices.zwave-js.io/?jumpTo=0x0129:0x0004:0x0000:0.0

mmiller7, 2022-03-30:
> They now appear on the BD Browser (last table) - providing the device you're looking at, has them recorded.
> 
> YRD210 https://devices.zwave-js.io/?jumpTo=0x0129:0x0004:0x0000:0.0

Ah, that's extremely useful, at least for the ones that map over!

AlCalzone, 2022-03-30:
If you have any undocumented alarm values that could be mapped to a standardized notification, feel free to open a new issue with the information or a PR adding them.

PR #2681 (2021-05-24): fix(cc): do not delete V1 alarm values when mapping notification
Deleting the alarm values causes each new mapping to be a breaking change, so we shouldn't do this.

related to https://github.com/zwave-js/node-zwave-js/issues/2409#issuecomment-823638655

fixes: #2605