zwave-js/zwave-js #2608: [bug] Fibaro - Missing associations

matthias-matze, 2021-05-13:
Hy, all my fibaro devices are missing assocation group lifeline.

The devices are alive ans interview complete.

i cant add association manually the field "Group" is empty.

All other devices (aeotec or qubino) are ok and working

AlCalzone, 2021-05-24:
If I trust the cache files, both Node 33 and 42 **do not** support `Association CC` or `Multi Channel Association CC`. However, they do support `Association Group Information`, which makes me sceptical there's something fishy.

Can you please set the loglevel to `"debug"`, re-interview both nodes and post a log of that?

matthias-matze, 2021-06-05:
Hy, 
upgraded to latest zwav-js version. Attached is log files with debug level. Hope it helps, and sorry for the delay to give you the attached files.
[zwavejs2mqtt-store(1).zip](https://github.com/zwave-js/node-zwave-js/files/6602065/zwavejs2mqtt-store.1.zip)

AlCalzone, 2021-06-06:
There are no logfiles in the attached archive. Please read here: https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report?id=export-logs

closed 2021-06-17 by zwave-js-assistant[bot]