zwave-js/zwave-js #2609: Retry logic after ` Timeout while waiting for an ACK from the controller`

robertsLando, 2021-05-13:
If I start my application before connecting the controller `driver.start()` keeps failing to connect as there is no tty connected and this is good as I can handle this and retry the connection but after some retries the start doesn't fail but after some seconds I get the error `Failed to initialize the driver: Timeout while waiting for an ACK from the controller` and after that nothing else happens. IMO there should be a retry logic implemented to retry sending that request

AlCalzone, 2021-05-13:
I've never needed to retry the connection - so I'm sure this has to do with the serial port not being ready before starting the driver. How would I reproduce this? Just start z2m without the stick plugged in and then plug in?

robertsLando, 2021-05-13:
> Just start z2m without the stick plugged in and then plug in?

Yep

AlCalzone, 2021-05-19:
Okay, I can't reproduce this 😅
Can you make me a log?

```
22:10:41.062 DRIVER   ███████╗ ██╗    ██╗  █████╗  ██╗   ██╗ ███████╗             ██╗ ███████╗
                      ╚══███╔╝ ██║    ██║ ██╔══██╗ ██║   ██║ ██╔════╝             ██║ ██╔════╝
                        ███╔╝  ██║ █╗ ██║ ███████║ ██║   ██║ █████╗   █████╗      ██║ ███████╗
                       ███╔╝   ██║███╗██║ ██╔══██║ ╚██╗ ██╔╝ ██╔══╝   ╚════╝ ██   ██║ ╚════██║
                      ███████╗ ╚███╔███╔╝ ██║  ██║  ╚████╔╝  ███████╗        ╚█████╔╝ ███████║
                      ╚══════╝  ╚══╝╚══╝  ╚═╝  ╚═╝   ╚═══╝   ╚══════╝         ╚════╝  ╚══════╝
22:10:41.063 DRIVER   version 7.5.0
22:10:41.063 CONFIG   version 7.5.0
22:10:41.063 DRIVER
22:10:41.064 DRIVER   starting driver...
22:10:41.066 DRIVER   opening serial port COM5
22:10:41.066 DRIVER   Serial port errored: Opening COM5: File not found
22:10:41.067 DRIVER   destroying driver instance...
2021-05-19 22:10:41.067 INFO ZWAVE: Controller status: Driver: Opening COM5: File not found
2021-05-19 22:10:41.067 WARN ZWAVE: Retry connection in 3 seconds...
2021-05-19 22:10:44.075 INFO ZWAVE: Connecting to COM5
22:10:44.076 DRIVER   ███████╗ ██╗    ██╗  █████╗  ██╗   ██╗ ███████╗             ██╗ ███████╗
                      ╚══███╔╝ ██║    ██║ ██╔══██╗ ██║   ██║ ██╔════╝             ██║ ██╔════╝
                        ███╔╝  ██║ █╗ ██║ ███████║ ██║   ██║ █████╗   █████╗      ██║ ███████╗
                       ███╔╝   ██║███╗██║ ██╔══██║ ╚██╗ ██╔╝ ██╔══╝   ╚════╝ ██   ██║ ╚════██║
                      ███████╗ ╚███╔███╔╝ ██║  ██║  ╚████╔╝  ███████╗        ╚█████╔╝ ███████║
                      ╚══════╝  ╚══╝╚══╝  ╚═╝  ╚═╝   ╚═══╝   ╚══════╝         ╚════╝  ╚══════╝
22:10:44.077 DRIVER   version 7.5.0
22:10:44.077 CONFIG   version 7.5.0
22:10:44.077 DRIVER
22:10:44.077 DRIVER   starting driver...
22:10:44.079 DRIVER   opening serial port COM5
22:10:44.079 DRIVER   Serial port errored: Opening COM5: File not found
22:10:44.080 DRIVER   destroying driver instance...
2021-05-19 22:10:44.080 INFO ZWAVE: Controller status: Driver: Opening COM5: File not found
2021-05-19 22:10:44.080 WARN ZWAVE: Retry connection in 3 seconds...
2021-05-19 22:10:47.091 INFO ZWAVE: Connecting to COM5
22:10:47.092 DRIVER   ███████╗ ██╗    ██╗  █████╗  ██╗   ██╗ ███████╗             ██╗ ███████╗
                      ╚══███╔╝ ██║    ██║ ██╔══██╗ ██║   ██║ ██╔════╝             ██║ ██╔════╝
                        ███╔╝  ██║ █╗ ██║ ███████║ ██║   ██║ █████╗   █████╗      ██║ ███████╗
                       ███╔╝   ██║███╗██║ ██╔══██║ ╚██╗ ██╔╝ ██╔══╝   ╚════╝ ██   ██║ ╚════██║
                      ███████╗ ╚███╔███╔╝ ██║  ██║  ╚████╔╝  ███████╗        ╚█████╔╝ ███████║
                      ╚══════╝  ╚══╝╚══╝  ╚═╝  ╚═╝   ╚═══╝   ╚══════╝         ╚════╝  ╚══════╝
22:10:47.093 DRIVER   version 7.5.0
22:10:47.093 CONFIG   version 7.5.0
22:10:47.093 DRIVER
22:10:47.094 DRIVER   starting driver...
22:10:47.095 DRIVER   opening serial port COM5
22:10:47.200 DRIVER   serial port opened
22:10:47.201 SERIAL » [NAK]                                                                   (0x15)
22:10:47.202 DRIVER   Checking for configuration updates...
22:10:47.218 SERIAL « 0x0118000a05000102010f5e86727322855970565a989f6c5574f4              (26 bytes)
22:10:47.219 SERIAL » [ACK]                                                                   (0x06)
22:10:47.220 DRIVER « [REQ] [UNKNOWN_FUNC_UNKNOWN_0x0a]
                        payload: 0x05000102010f5e86727322855970565a989f6c5574
22:10:47.223 DRIVER   handling request UNKNOWN_FUNC_UNKNOWN_0x0a (10)
22:10:47.223 DRIVER     no handlers registered!
22:10:47.583 DRIVER   No configuration update available...
2021-05-19 22:10:47.584 INFO ZWAVE: Next update scheduled for: Thu May 20 2021 00:00:00 GMT+0200 (GMT+02:00)
22:10:48.719 DRIVER   loading configuration...
22:10:49.061 DRIVER   beginning interview...
22:10:49.062 DRIVER   added request handler for AddNodeToNetwork (0x4a)...
                      1 registered
22:10:49.062 DRIVER   added request handler for RemoveNodeFromNetwork (0x4b)...
                      1 registered
22:10:49.062 DRIVER   added request handler for ReplaceFailedNode (0x63)...
                      1 registered
22:10:49.063 CNTRLR   beginning interview...
22:10:49.063 CNTRLR   querying version info...
22:10:49.126 SERIAL » 0x01030015e9                                                         (5 bytes)
22:10:49.127 DRIVER » [REQ] [GetControllerVersion]
22:10:49.133 SERIAL « [ACK]                                                                   (0x06)
22:10:49.134 SERIAL « 0x011001155a2d5761766520372e3131000797                              (18 bytes)
22:10:49.134 SERIAL » [ACK]                                                                   (0x06)
22:10:49.135 DRIVER « [RES] [GetControllerVersion]
                        payload: 0x5a2d5761766520372e31310007
22:10:49.138 CNTRLR   received version info:
                        controller type: Bridge Controller
                        library version: Z-Wave 7.11
```

robertsLando, 2021-05-20:
Cannot reproduce now, closing and reopening if I will get this again

robertsLando, 2021-05-21:
Finally able to reproduce it: 

[zwavejs_13188.log](https://github.com/zwave-js/node-zwave-js/files/6520629/zwavejs_13188.log)

AlCalzone, 2021-05-21:
Like we discussed in Slack: you can detect the startup failure using the `"error"` event and check for `error.code === ZWaveErrorCodes.Driver_Failed`.

closed 2021-05-21 by robertsLando