zwave-js/zwave-js #2634: Add Aeotec Smart Switch 7

AlCalzone, 2021-05-17:
This seems premature until the config from @blhoward2 is available publicly as it seems a few people are tracking using this PR.

_Originally posted by @Nixon506E in https://github.com/zwave-js/node-zwave-js/issues/2027#issuecomment-842668658_

tommyjlong, 2021-06-11:
HI, Wanted to get a status on this.....should I go ahead and try to make my own device file for this device until the official one comes out?

blhoward2, 2021-06-11:
You’re welcome to do so. I only have a few left to do but I’ve had a busy few weeks at work.

tommyjlong, 2021-06-14:
I'm working on a config device file and have a question.....  For the ZWA023 there is a config parameter: "Night Light On Time". It is a 4 byte value and the format is: `0xRRHHMMSS`  with the default of `0x00180000` which the manual says `means 6pm` (i.e 18 based on a 24 hour clock).

Focusing on the hour HH, since 0x18 means 18 decimal, this indicates to me that they are using Binary Encoded Decimal, whereby a nibble is used for each digit.  Regarding the device's config json file for this particular parameter, I understand that numbers are represented only as decimal (correct me if wrong), then I believe for HH, the values should be: `"defaultValue": 24` (i.e. x18), and the range is `"minValue": 0` to `"maxValue": 35` (x23). Is this correct?

blhoward2, 2021-06-14:
While that may be right if you take the hex literally, I believe they use 18 as in the 18th hour. So the range is 0-23 for the HH.

blhoward2, 2021-06-14:
I confirmed with Aeotec. 18 means 6 pm. 19 means 7 pm, etc. It's not a true hex number.

AlCalzone, 2021-06-14:
What the hell? So they actually have 0x00, 0x01, ..., 0x09, 0x10, 0x11, ..., 0x19, 0x20, 0x21, ..., 0x23?
Then we need to use `"options"` to model this and a range of 0..35.

blhoward2, 2021-06-14:
Which means 59 options for the minute field. Unless we dumb it down instead of bitmasking it and give people one field with options every 30 minutes from 0000 to 2400 or something.

AlCalzone, 2021-06-14:
That could be an option, but I'm sure that someone will want to set 19:57 or something :)

blhoward2, 2021-06-14:
I'm not sure how we handled this elsewhere. I sort of thing we left it manual entry and said see manual.

AlCalzone, 2021-06-14:
honestly I'd be fine with that

blhoward2, 2021-06-14:
We’ll, since its human readable in hex I vote that we let it manual entry and just put the format in the description. 235900 = 23:59 and 0 seconds.

blhoward2, 2021-06-14:
See my edit, does that work? Aren’t we going to convert it to hex before we send it?

blhoward2, 2021-06-14:
Maybe we need an option that sends the value through as hex instead of converting it?

AlCalzone, 2021-06-14:
Users would have to convert `0x235900` to decimal before entering it. Maybe @robertsLando could add something to the UI that does that when users enter a `0x` prefixed hex number.

tommyjlong, 2021-06-15:
As an FYI.....
 after running a few tests on my Smart Switch 7 I can confirm that the device does as Aeotec says and as the manual shows (i.e 0x18 is 6pm).

blhoward2, 2021-06-15:
@robertsLando Can you do something similar to HA so we can make the description reference the hex value? Essentially if a hex value is put into a manual entry it should be treated as a hex and not a decimal.

tommyjlong, 2021-06-16:
As a reference/FYI, here is a link to the device file I came up with:
https://github.com/tommyjlong/temp/blob/main/zwa023.json

blhoward2, 2021-06-16:
Thanks. I made good progress on these last night so hope to push the official one tonight or tomorrow.

PR #2896 (2021-06-23): feat(config): template more aeotec devices
Closes #2893
Closes #2796
Closes #2637
Closes #2634
Closed #2664 

After this, I have just three more Aeotec files to go. Bulb 6 Multi-White, Bulb 6 Multi-Color, and Multisensor 7.
closed 2021-06-23 by AlCalzone