zwave-js/zwave-js #2637: Add Aeotec ZWA019

wabbastang, 2021-05-18:
**Which device is missing?**

Aeotec Water Sensor 7 Pro
Exists in openzwave database already 0371:0013:0102

**What are the IDs?**

Manufacturer ID: 0x0371
Product type: 0x0102  
Product ID: 0x0013  

**Did you check if we really don't have it?**

- [ ] It is not in the [configuration DB](https://devices.zwave-js.io/)
- [ ] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)
- [ ] It is not in the [TODO list](https://github.com/zwave-js/node-zwave-js/issues/1600)

**Do you have a link to the PDF manual?**

https://products.z-wavealliance.org/ProductManual/File?folder=&filename=product_documents/3951/ZWA019.pdf

AlCalzone, 2021-05-18:
@blhoward2 that one is on your TODO list, right?

blhoward2, 2021-05-18:
Yes. I have a file for this with templating underway.

PR #2896 (2021-06-23): feat(config): template more aeotec devices
Closes #2893
Closes #2796
Closes #2637
Closes #2634
Closed #2664 

After this, I have just three more Aeotec files to go. Bulb 6 Multi-White, Bulb 6 Multi-Color, and Multisensor 7.
closed 2021-06-23 by AlCalzone