zwave-js/zwave-js #2643: Battery device stuck in Awake status after AssignSUCReturnRoute

kpine, 2021-05-19:
**Checklist:**

-   [X] My problem is **not** within Home Assistant or the ZWave JS integration. **Or:** a Home Assistant developer has told me to come here.
-   [X] My problem is **not** within ZWaveJS2MQTT. **Or:** a ZWaveJS2MQTT developer has told me to come here.
-   [X] I have checked the troubleshooting section and my problem is **not** described there.
-   [X] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

I manually woke up a battery node that doesn't auto-wake. Upon waking up, zwave-js issued a AssignSUCReturnRoute command. After that, the device continues to be considered Awake. I manually woke it up a second time, the wake up notification event was received, but the device continues to be considered Awake. I'm pretty certain the device is no longer awake.

**Device information**

Which device(s) is/are affected (manufacturer/model)?  Hank HKZW-DWS01 (Monoprice version)
What is/are the node IDs? 3

**Did you change anything?**

-   [ ] **Yes**: _(please describe)_
-   [X] **No**

**Did this use to work before?**

-   [X] **Don't know**
-   [ ] **No**, it never worked anywhere
-   [X] **Yes**, in: _a previous version_, can't say when.

**How are you using `node-zwave-js`**

-   [X] `zwavejs2mqtt` (4.2.1) docker image
-   [ ] `zwavejs2mqtt` (dev) docker image
-   [ ] `zwavejs2mqtt` Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] ioBroker.zwave2 adapter
-   [ ] `HomeAssistant` version XYZ
-   [ ] Pkg
-   [ ] Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] `node-red-contrib-zwave-js`
    -   Module version (double click node): 
-   [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github))
-   [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:

Not sure I can reproduce it again. These are the steps I took.

1. Manually wake up the HKZW-DWS01 device
3. Watch it wake up and
4. See the node never sleep

**Additional context**

I have not tried restarting zwavejs2mqtt yet.

**Logfile:**

Logs for this are in the 05-19 file.

For some reason I am missing logs for the entire day of 5-18. I was using the log viewer in HA yesterday, I wonder if messing with the log settings there did something to screw up the logger, there is some weird behavior with it. Anyways, there was no activity for this node on that day, so it shouldn't be missing anything. The sensors is sitting in a drawer.

AlCalzone, 2021-05-19:
Try sending the device a command, you should see it go to sleep then. Some time ago I stopped automatically marking devices as asleep after a fixed time because that just caused problems. Now a device is only marked asleep if it didn't ack a message or if it was actively sent to sleep.

kpine, 2021-05-19:
A refresh of a CC failed after 3 attempts and it went to sleep. Feel free to close if you think this behavior is fine. It's not something I've seen with any of my other devices, so seems like an edge case.

kpine, 2021-05-19:
Although, it now seems to be stuck in this scenario. I did another wake up, the battery refresh was successful, but it's still awake. Should the command to go back to sleep have been sent?

[zwavejs_1.log_2021-05-19.log](https://github.com/zwave-js/node-zwave-js/files/6510709/zwavejs_1.log_2021-05-19.log)

kpine, 2021-05-19:
Ok, here's another node where I wake it up and then it gets sent back to sleep. I'm not sure what the difference is, the ApplicationUpdateRequest?

[zwavejs_1.log_2021-05-19.log](https://github.com/zwave-js/node-zwave-js/files/6510734/zwavejs_1.log_2021-05-19.log)

AlCalzone, 2021-05-19:
Thanks for the logs - I'll check it out tomorrow if we can do better.

AlCalzone, 2021-05-20:
The node is configured with a wakeup interval of 0 (which means it only wakes up manually).
In this case, the node doesn't get sent to sleep by the controller because the user might want to interact with it.

kpine, 2021-05-20:
Sorry, I don't follow the comment, could I ask some clarifying questions? Why does the node requiring a manual wakeup mean that you will never try to put it back to sleep? How is it different from a non-zero wakeup interval?

AlCalzone, 2021-05-20:
A node with a non-zero wakeup interval wakes up on its own after the configured interval. If the interval is 0, it **never** wakes up, unless you manually wake it up.

This is also true for some remotes/keyfobs, which aren't meant to be interacted with, unless when they are put in an extended wakeup state. We don't want to put those back to sleep quickly, because this would make them tedious to interact with. They will still sleep on their own after a while, but we will only notice after trying to interact with them. We could ping in regular intervals to figure this out quicker, but that could cause other commands to be delayed.

kpine, 2021-05-20:
Ok, that's understandable. This device is hard-coded with all zeroes for the wake up interval capabilities, so I don't have a choice to change it. Not like I wake it up often anyways.

I can live with just sending another command to force it back to sleep. Do you provide the ability to ping (No Op) nodes via the API? I don't see it available in zwavejs2mqtt.

AlCalzone, 2021-05-20:
> Do you provide the ability to ping (No Op) nodes via the API? I don't see it available in zwavejs2mqtt

@robertsLando?

> I see a public node ping method, but not sure if it's meant to be public

It is - probably just missing from the docs.