zwave-js/zwave-js #2655: Multilevel Switch current level value is never refreshed after a Set - Heatit ZDim

henrikeri, 2021-05-20:
When toggeling the dimmer, the multilevel Switch current level value is never refreshed after a set. This makes the dimmer unusable with the HomeAssistant Lovelace UI. 

See https://pastebin.com/N4D7yV98 for details

kpine, 2021-05-20:
Logs from pastebin: [logs.txt](https://github.com/zwave-js/node-zwave-js/files/6519060/logs.txt)

I told @henrikeri to submit this issue, because it looks like the Multilevel Switch Set is supervised, but the Supervision Report doesn't seem to provide anything from what I can tell, so the current value is never refreshed. On the other hand, the Binary Switch Set is not supervised, and the node responds on its own with Binary and Multilevel Switch reports.

blhoward2, 2021-05-21:
@kpine So you think we need to disable Supervision?

blhoward2, 2021-05-21:
@zwave-js-bot add compat flag to 0x019b:0x0003:0x2200:0.0

```
"commandClasses": {
			"remove": {
				"0x6c": {
					// The device does not support supervision although it is mandatory for the device type
					"endpoints": "*"
				}
			}
```

blhoward2, 2021-05-21:
@zwave-js-bot add compat flag to 0x019b:0x0003:0x2200:0.0

```
"commandClasses": {
			"remove": {
				"0x6c": {
					"endpoints": "*"
				}
			}
```

blhoward2, 2021-05-21:
@zwave-js-bot add compat flag to 0x019b:0x0003:0x2200:0.0

```
"commandClasses": {
	"remove": {
		"0x6c": {
			"endpoints": "*"
		}
	}
}
```

blhoward2, 2021-05-21:
@AlCalzone See the error message on the bot. It looks like there is a bug in the code.

kpine, 2021-05-21:
> 
> 
> @kpine So you think we need to disable Supervision?

It came to mind, but I couldn't remember what was supposed to actually happen with Supervision and setting the target value to 0xFF (last level).

blhoward2, 2021-05-21:
I staged the PR but I'm not actually sure this is right:

```
2021-05-20T21:31:41.487Z DRIVER » [Node 031] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      128
                                  └─[SupervisionCCGet]
                                    │ session id:      56
                                    │ request updates: true
                                    └─[MultilevelSwitchCCSet]
                                        target value: 255
2021-05-20T21:31:41.489Z SERIAL « [ACK]                                                                   (0x06)
2021-05-20T21:31:41.496Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2021-05-20T21:31:41.496Z SERIAL » [ACK]                                                                   (0x06)
2021-05-20T21:31:41.497Z DRIVER « [RES] [SendData]
                                    was sent: true
2021-05-20T21:31:41.513Z SERIAL « 0x010700138000000368                                                 (9 bytes)
2021-05-20T21:31:41.513Z SERIAL » [ACK]                                                                   (0x06)
2021-05-20T21:31:41.514Z DRIVER « [REQ] [SendData]
                                    callback id:     128
                                    transmit status: OK
2021-05-20T21:31:41.528Z SERIAL « 0x010b0004001f056c0238ff0043                                        (13 bytes)
2021-05-20T21:31:41.528Z SERIAL » [ACK]                                                                   (0x06)
2021-05-20T21:31:41.529Z DRIVER « [Node 031] [REQ] [ApplicationCommand]
                                  └─[SupervisionCCReport]
                                      session id:          56
                                      more updates follow: false
                                      status:              Success
                                      duration:            [Duration: 0 seconds]
```

It does see a Supervision report back. @AlCalzone ?

AlCalzone, 2021-05-21:
Supervision is not the problem - its the fact that we don't refresh the value after setting it to 255.

henrikeri, 2021-05-21:
I got in contact with Thermofloor today, they were aware of the issue and supplied an OTA firmware update that resolves the issue.

AlCalzone, 2021-05-21:
> they were aware of the issue and supplied an OTA firmware update that resolves the issue.

So the expected behavior is that the device reports the actual status on its own?
That's good to know but we still need to handle those device that don't.

henrikeri, 2021-05-21:
> 
> 
> > they were aware of the issue and supplied an OTA firmware update that resolves the issue.
> 
> So the expected behavior is that the device reports the actual status on its own?
> That's good to know but we still need to handle those device that don't.

I would assume that's the expected behaviour given the fact that they did have an OTA update ready. However, they did not specifically elaborate as to wheter this expected or not.

PR #2673 (2021-05-24): fix: poll after supervised Multilevel Switch Set if targetValue is 255
fixes: #2655
closed 2021-05-24 by AlCalzone