zwave-js/zwave-js #2659: Coolcam Light switch 2 channel (EU-2) "does not support the CC assosiation" error

davidwaze, 2021-05-21:
Build/Run method

[ X ] Manually built (git clone - npm install - npm run build )
zwavejs2mqtt version: 4.2.1
zwave-js version: 7.4.0

Describe the bug
Interview fails

Device : 600-4235-3 (0x0258-0x0003-0x108b)

Tried : multiple interviews without any success / remove device / factory reset device

`2021-05-21T08:48:16.712Z CNTRLR » [Node 196] querying the CC version for Manufacturer Specific...
2021-05-21T08:48:16.714Z SERIAL » 0x010a0013c403861372251af9 (12 bytes)
2021-05-21T08:48:16.714Z DRIVER » [Node 196] [REQ] [SendData]
│ transmit options: 0x25
│ callback id: 26
└─[VersionCCCommandClassGet]
CC: Manufacturer Specific
2021-05-21T08:48:16.716Z SERIAL « [ACK] (0x06)
2021-05-21T08:48:16.725Z SERIAL « 0x0104011301e8 (6 bytes)
2021-05-21T08:48:16.725Z SERIAL » [ACK] (0x06)
2021-05-21T08:48:16.726Z DRIVER « [RES] [SendData]
was sent: true
2021-05-21T08:48:16.741Z SERIAL « 0x010700131a000003f2 (9 bytes)
2021-05-21T08:48:16.741Z SERIAL » [ACK] (0x06)
2021-05-21T08:48:16.742Z DRIVER « [REQ] [SendData]
callback id: 26
transmit status: OK
2021-05-21T08:48:16.753Z SERIAL « 0x010a000400c40486147202d3 (12 bytes)
2021-05-21T08:48:16.754Z SERIAL » [ACK] (0x06)
2021-05-21T08:48:16.754Z DRIVER « [Node 196] [REQ] [ApplicationCommand]
└─[VersionCCCommandClassReport]
CC: Manufacturer Specific
version: 2
2021-05-21T08:48:16.755Z CNTRLR [Node 196] supports CC Manufacturer Specific (0x72) in version 2
2021-05-21T08:48:16.755Z CNTRLR [Node 196] skipping query for Device Reset Locally (0x5a) because max implem
ented version is 1
2021-05-21T08:48:16.755Z CNTRLR [Node 196] skipping query for Powerlevel (0x73) because max implemented vers
ion is 0
2021-05-21T08:48:16.755Z CNTRLR » [Node 196] querying the CC version for Association...
2021-05-21T08:48:16.757Z SERIAL » 0x010a0013c403861385251b0f (12 bytes)
2021-05-21T08:48:16.757Z DRIVER » [Node 196] [REQ] [SendData]
│ transmit options: 0x25
│ callback id: 27
└─[VersionCCCommandClassGet]
CC: Association
2021-05-21T08:48:16.759Z SERIAL « [ACK] (0x06)
2021-05-21T08:48:16.768Z SERIAL « 0x0104011301e8 (6 bytes)
2021-05-21T08:48:16.769Z SERIAL » [ACK] (0x06)
2021-05-21T08:48:16.769Z DRIVER « [RES] [SendData]
was sent: true
2021-05-21T08:48:16.785Z SERIAL « 0x010700131b000002f2 (9 bytes)
2021-05-21T08:48:16.785Z SERIAL » [ACK] (0x06)
2021-05-21T08:48:16.786Z DRIVER « [REQ] [SendData]
callback id: 27
transmit status: OK
2021-05-21T08:48:16.796Z SERIAL « 0x010a000400c4048614850224 (12 bytes)
2021-05-21T08:48:16.797Z SERIAL » [ACK] (0x06)
2021-05-21T08:48:16.797Z DRIVER « [Node 196] [REQ] [ApplicationCommand]
└─[VersionCCCommandClassReport]
CC: Association
version: 2
2021-05-21T08:48:16.798Z CNTRLR [Node 196] supports CC Association (0x85) in version 2
2021-05-21T08:48:16.798Z CNTRLR » [Node 196] querying the CC version for Association Group Information...
2021-05-21T08:48:16.799Z SERIAL » 0x010a0013c403861359251cd4 (12 bytes)
2021-05-21T08:48:16.800Z DRIVER » [Node 196] [REQ] [SendData]
│ transmit options: 0x25
│ callback id: 28
[... 334 lines cut ...]
2021-05-21T08:48:17.309Z DRIVER « [RES] [SendData]
was sent: true
2021-05-21T08:48:17.325Z SERIAL « 0x0107001326000002cf (9 bytes)
2021-05-21T08:48:17.326Z SERIAL » [ACK] (0x06)
2021-05-21T08:48:17.326Z DRIVER « [REQ] [SendData]
callback id: 38
transmit status: OK
2021-05-21T08:48:17.337Z SERIAL « 0x010d000400c407600d01008e0602d3 (15 bytes)
2021-05-21T08:48:17.338Z CNTRLR [Node 196] [+] [Multi Channel Association] groupCount: [Endpoint 1] [internal]
2
2021-05-21T08:48:17.338Z SERIAL » [ACK] (0x06)
2021-05-21T08:48:17.339Z DRIVER « [Node 196] [REQ] [ApplicationCommand]
└─[MultiChannelCCCommandEncapsulation]
│ source: 1
│ destination: 0
└─[MultiChannelAssociationCCSupportedGroupingsReport]
group count: 2
2021-05-21T08:48:17.339Z CNTRLR « [Node 196] supports 2 multi channel association groups [Endpoint 1]
2021-05-21T08:48:17.340Z CNTRLR [Node 196] Error during node interview: Node 196 (endpoint 1) does not support
the Command Class Association!
2021-05-21T08:49:07.287Z SERIAL « 0x01080004005002840722 (10 bytes)
2021-05-21T08:49:07.288Z SERIAL » [ACK] (0x06)`

Thanks for your support.
[image]

AlCalzone, 2021-05-21:
This should already be fixed on master -> #2646
I'll probably release a patch tonight.

closed 2021-05-21 by AlCalzone

davidwaze, 2021-05-21:
Thnaks my friend and for your faaaaast answer :-)

davidwaze, 2021-05-22:
Hello,

Just one small question, it tried to find the answer without success (probably wrong search ...)

I'm running with zwavejs2mqtt 4.3.0 (with included zwavejs 7.5.0).

How to upgrade only zwavejs to 7.5.1 without waiting next release of zwavejs2mqtt ?

Thanks for your help.