zwave-js/zwave-js #2674: Philio - PST02A - Notifications do not update and motion detection keep triggered

Antoine1-byte, 2021-05-24:
**Describe the bug**

Since I switched from former Zwave integration, my 4 in 1 sensor is no more reporting the states into HA.
The binary sensors are triggering properly within ZwaveJS2MQTT but the notification related to these sensors aren't ok.

[image]

I did several tests by changing parameter 7 to either, 6 (former setting used in former Zwave integration,) 20, 22 without any improvement.
I also tried to remove / reintegrate the device: no luck

**Device information**

Which device(s) is/are affected (manufacturer/model)?  PHILIO PST02
What is/are the node IDs?  0x013c

**Did you change anything?**

-   [ ] **Yes**: _(please describe)_
-   [x] **No**

**Did this use to work before?**

-   [ ] **Don't know**, this is a new device
-   [ ] **No**, it never worked anywhere
-   [x] **Yes**, in: When using former Zwave integration into HA

AlCalzone, 2021-05-24:
Don't just delete part of the template. It is all there for a reason.
Please make a zwave-js log, loglevel `debug` or `silly` and attach it here as a file.

Antoine1-byte, 2021-05-24:
Sorry, I tried to update the first post to provide all the data.

Here is the logfile
[zwave_js_dump.zip](https://github.com/zwave-js/node-zwave-js/files/6530718/zwave_js_dump.zip)

AlCalzone, 2021-05-24:
That's not the correct one - I need the driver (zwave-js) logfile:
https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report?id=export-logs

Antoine1-byte, 2021-05-24:
Sorry,

Is it better now?

[zwavejs2mqtt-store.zip](https://github.com/zwave-js/node-zwave-js/files/6530981/zwavejs2mqtt-store.zip)

AlCalzone, 2021-05-24:
Yes. Node 33 is sending Binary Sensor reports, but I also saw you set param 7 to `6`, which **seems** correct. After that, I saw a couple of Notification Reports in the log:
```
2021-05-24T06:50:58.828Z DRIVER « [Node 033] [REQ] [ApplicationCommand]
                                  └─[NotificationCCReport]
                                      notification type:   Home Security
                                      notification status: 255
                                      notification state:  Motion detection
```
for example.

For Node 34 I saw you set it to 20, which should cause it to send Binary Sensor reports.

The device files are terrible and need to be corrected btw.

AlCalzone, 2021-05-24:
Ohh, I just took a look into the manual:
[image]

That is a terrible choice, since `0xfe` is the "unknown" event. For "no motion", the device should be using `0x00` (idle).

We would need to implement a compat flag that treats `0xfe` as `idle`, but there are more pressing issues right now. IMO you're better off by configuring the sensor to send Binary Sensor Reports and rely on that.

Antoine1-byte, 2021-05-24:
Thanks for the feedback.

So you mean I should go for this setting as an example? 00010110

AlCalzone, 2021-05-24:
> But at the end, as the Binary Sensor Motion is never going to false

You're right, I'm not seeing any reports with "no motion" while the Binary Sensor setting is active. Try setting param 7 to 18 (00010010) or 22 (00010110). If I read the manual correctly, this should cause the device to send the correct reports.

Antoine1-byte, 2021-05-24:
With param 7 to 22 or 18: the sensor goes from true to false and false to true, so seems to be ok on this one but the notification isn't updated and remains on "motion detection"
[image]

AlCalzone, 2021-05-24:
> but the notification isn't updated and remains on "motion detection"

If the device only sends one or the other, we can't keep both values up to date. Just ignore the notification part.

Antoine1-byte, 2021-05-24:
Not really better in fact, the device keeps triggering from false to true without any movement...

Here is how it looks like into HA:
[image]

And the logs:
[zwavejs2mqtt-store.zip](https://github.com/zwave-js/node-zwave-js/files/6534058/zwavejs2mqtt-store.zip)

If you tell me that it is a compatibility issue, I can live with it for some times (I am facing the issues for some months now but, to be honnest, I am unsure if it is due to my setup or to the ZwaveJS component).
I only know that I have see few guys mentionning more or less the same issue on the HA community so...

If you need me to proceed further tests or to provide more logs, feel free to ask.

Thanks for your help and the great job

AlCalzone, 2021-05-24:
> the device keeps triggering from false to true without any movement
> I am unsure if it is due to my setup or to the ZwaveJS component

The driver just interprets what the device sends it. If it triggers too often, the sensitivity on the device needs to be reduced somehow.

Antoine1-byte, 2021-07-31:
Hello,

Just a quick update: unfortunately, adjusting the sensitivity of the PIR sensor didn't changed anything.

The issue I am facing is linked to this one which has been raised by another user couple of weeks ago: [https://github.com/zwave-js/node-zwave-js/issues/2930](url)

Several post on the HA forum are also relating the same problem.

Do you have an idea on how we can fix it by our own or shall we wait until a incoming update to get it fix?

All my installation works perfectly since the witch to ZwaveJS2MQTT except this sensor :(

Thanks in advance,

AlCalzone, 2021-08-02:
@Antoine1-byte in the issue you linked, I requested a log which was not provided. So I'm not sure if that is a different issue than here.
Please make a driver log, loglevel debug of a cycle where the sensor detects motion and then no motion. Upload it here.

Antoine1-byte, 2021-08-02:
Hello,

Here is the last log I have just downloaded.

Is it helpfull or do you need anything else?
I seems that the detection is working properly but it does not translate it to the notification.

[zwavejs2mqtt-store.zip](https://github.com/zwave-js/node-zwave-js/files/6916262/zwavejs2mqtt-store.zip)

AlCalzone, 2021-08-02:
> it does not translate it to the notification.

Your sensor is reporting the motion via the `Binary Sensor CC` subtype `Motion`, not with notifications. I read something that HA by default disables Binary Sensor entities if the device also uses notifications. Maybe that's the case for you?

Antoine1-byte, 2021-08-02:
@AlCalzone Thanks for looking (again) at the issue!

You are right that all the entities are not automatically activated in HA.
This is something I saw when trying to understand what was wrong.

Fact is the binary sensor keeps trigerring even when there is no movement (I tried to put the sensor in a box to be sure...).
Of course I tried to reduce the sensitivity of the sensor but it doesn't change anything...
[image]

Here is another log where I changed parameter 7 from 22 to 6.

[zwavejs_2021-08-02.log](https://github.com/zwave-js/node-zwave-js/files/6918700/zwavejs_2021-08-02.log)

It seems that the "notification status" remains permanently on "Motion detection" whathever the state of the sensor?

AlCalzone, 2021-08-02:
> Fact is the binary sensor keeps trigerring even when there is no movement (I tried to put the sensor in a box to be sure...).

I'm afraid that's nothing we can change if you can't configure the device differently.

> It seems that the "notification status" remains permanently on "Motion detection" whathever the state of the sensor?

That's the "beauty" of Z-Wave. Many possibilities to do things and devices often use multiple solutions for the same problem, but only do them half-assed. I'm guessing there is a config option to switch between Binary Sensor and Notifications?
Blindly correlating different CCs isn't something we're going to do, unless it is 100% certain that it is meant to be that way - see the Lock status notifications that many locks use instead of the Lock CC reports.

Antoine1-byte, 2021-08-03:
Hi,

Do not misunderstand what I'm going to say but sometimes it is a bit technical for me to understand.
I'm trying to gather the info I have and your feedback to look after a solution.

`I'm afraid that's nothing we can change if you can't configure the device differently.`
So that means that your understanding is that if it keeps triggering it might be a config issue.
I'll try to look at it but what I do not understand is the fact that I do not know if it was better with former Z-wave intergration as I was config option with Notifications

`That's the "beauty" of Z-Wave. Many possibilities to do things and devices often use multiple solutions for the same problem, but only do them half-assed. I'm guessing there is a config option to switch between Binary Sensor and Notifications?
Blindly correlating different CCs isn't something we're going to do, unless it is 100% certain that it is meant to be that way - see the Lock status notifications that many locks use instead of the Lock CC reports.`

I think that I know and can switch between Binary Sensor (param 7 at 22 or 20) and Notifications (param 7 at 6)
Here:
- if I go to Binary Sensor: the motion detection is trigerring every 15sec, despite the PIR sensitivity I am defining
- if I go to Notifications: the Notification status is stuck on "[8] Motion detection"
What is unclear for me is why it was working with former intergration and it is not with this one (it was also working under Jeedom)
Is it something which requires some work from your side or is it as it is and I should go for another sensor?
Maybe your conclusion is that I am not able to configure properly the device, which is a possibility and which I can hear...

AlCalzone, 2021-08-04:
> once a notification has trigerred, it doesn't go back to its initial state

Yeah that is the issue you linked to before. The device does not use the standard "idle" notification to say it is back to idle, but rather the "unknown" notification. I don't know why but it is a stupid decision from the manufacturer.
Other zwave drivers likely already have a workaround for that, which is why it works there.

Antoine1-byte, 2021-11-07:
Hi,

Just in case someone is facing a similar issue than me...

I sent back the device to my retailer and he sent me a new one.
With this new one, the binary sensor is working normally and do not triggers every 30 sec so the device was faulty at the end.

To sumup:
notifications aren't working (yet?)
binary sensor is working properly, just need to activate it in HA

Regards,

Extreme79, 2023-12-17:
I just included that sensor in my configuration.
At this days, notifications stays unworking (zwavejs).
I had to switch to binary report to get working  motion and door notifications (and obviously use the right entities in HA)