zwave-js/zwave-js #2675: [question] Fibaro Thermostat FGT001 - too many batteries

kvaster, 2021-05-24:
Hello, I'm migration from openzwave to zwavejs now.

I have fibaro thermostats FGT001. Such thermostat have external temperature sensors with it's separate battery. On openzwave I was able to see only two batteries (one for thermostat and one for sensor). With zwavejs2mqtt I can see three batteries ([5-128-x-level] Battery level, where x=0,1,2). And first two have same values - thermostats battery state.

As I understand the first one value is from single channel and the second one is from multi channel number one. I am developer and already made small commits to openzwave, but I have no good knowledge of z-wave protocol yet. I will appreciate if somebody could tell me if this problem can be fixed somehow - is there something in openzwave code which helps to avoid duplicating of such values.

I'm attaching node interview logs.

[zwavejs_14580.log_2021-05-23.log](https://github.com/zwave-js/zwavejs2mqtt/files/6528716/zwavejs_14580.log_2021-05-23.log)

p.s.: also it seems that clock is not supported on this device while openzwave was able to show it.

AlCalzone, 2021-05-24:
> With zwavejs2mqtt I can see three batteries ([5-128-x-level] Battery level, where x=0,1,2). And first two have same values - thermostats battery state.

That's correct. Root endpoint 0 mirrors the functionality of endpoint 1 (the thermostat). Endpoint 2 is the external sensor. It seems you can ignore endpoint 0 for this device, but unfortunately, this is not easy to figure out programmatically.

> also it seems that clock is not supported on this device

It is, it's just not exposed to users because it is managed behind the scenes.

kvaster, 2021-05-24:
Is there any way to ignore endpoint 0 by adding some metainfo ?

And about clock - I was confused cause of such message:
```
skipping query for Clock (0x81) because max implemented version is 1
```

Am I righ that clock is automatically queried and adjusted from time to time ? Cause I was not able to find this in docs...

kvaster, 2021-05-24:
Seems that the only way is to exclude duplicate command classes from endpoint 0 or 1 one by one in device files.

AlCalzone, 2021-05-24:
> skipping query for Clock (0x81) because max implemented version is 1

That is part of the **Version** query where we find out which CC versions the device implements. If zwave-js only implements V1, there's no point in actually doing that query.

> Am I righ that clock is automatically queried and adjusted from time to time ? 

Almost. According with the specifications
> A Z-Wave Plus node SHOULD issue a Clock Report Command via the Lifeline Association Group if they suspect to have inaccurate time and/or weekdays

zwave-js checks if the reported time is off and corrects it. We don't currently force this by querying the clock setting.

kvaster, 2021-05-24:
Thank you very much for clarification!

About duplicate command classes: maybe it is a good idea to have option in device file to automatically remove diplicates between endpoint 0 and x :)

I think this question may be closed. Will try to make some PRs very soon at least for devices I own.

AlCalzone, 2021-05-24:
> maybe it is a good idea to have option in device file to automatically remove diplicates between endpoint 0 and x

Short answer: no.
Long answer: This is a huge fucking mess, but I'm trying to solve it more generically. https://github.com/zwave-js/node-zwave-js/issues/2286 is the canonical issue for this.

closed 2021-05-24 by AlCalzone